Hamlib/Hamlib #32: NET rigctl: Protocol error

kukabu, 2018-05-25:
Hello
The NET rigctl doesn't work on last git version (783ba3edd5236c748a91e3ba6c4fb199f3e46fa9).

Steps to reproduce the behavior:
1. run rigctld -vvvv -m1
2. run rigctl -vvvvv -m2 -r127.0.0.1:4532

N0NB, 2018-05-25:
* On 2018 25 May 06:48 -0500, Nate Bargmann wrote:
> Hello
> The NET rigctl doesn't work on last git version (783ba3edd5236c748a91e3ba6c4fb199f3e46fa9).
> 
> Steps to reproduce the behavior:
> 1. run rigctld -vvvv -m1
> 2. run rigctl -vvvvv -m2 -r127.0.0.1:4532

I can confirm:

$ rigctld -vvvv -m1    
rigctld, Hamlib 3.3~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
dummy: _init called
rig_register called
rig_register: rig_register (1)
rig_register called
rig_register: rig_register (2)
rig_register called
rig_register: rig_register (4)
rig_register called
rig_register: rig_register (5)
dummy_init called
rig_strvfo called
rig_passband_normal called
rig_strvfo called
rig_passband_normal called
rig_open called
port_open called
dummy_open called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
Opened rig model 1, 'Dummy'
rig_strstatus called
Backend version: 0.5, Status: Beta
rig_close called
dummy_close called
port_close called
Closed rig model 1, 'Dummy - will reopen for clients'
Connection opened from localhost:50970
client lock engaged
rig_open called
port_open called
dummy_open called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
Opened rig model 1, 'Dummy'
client lock disengaged
client lock engaged
rig_strvfo called
client lock disengaged
fscanf: Connection reset by peer
fscanf: parsing '�' with '%c'
rig_open called
client lock engaged
rig_close called
dummy_close called
port_close called
Closed rig model 1, 'Dummy - no clients, will reopen for new clients'
client lock disengaged
Connection closed from localhost:50970

and:

$ rigctl -vvvvv -m2 -r127.0.0.1:4532
rigctl, Hamlib 3.3~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
dummy: _init called
rig_register called
rig_register: rig_register (1)
rig_register called
rig_register: rig_register (2)
rig_register called
rig_register: rig_register (4)
rig_register called
rig_register: rig_register (5)
rig_open called
port_open called
network_open called
netrigctl_open called
netrigctl_transaction: called len=12
network_flush called
write_block called
write_block(): TX 12 bytes
0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a                 \dump_state.
read_string called
read_string(): RX 2 characters
0000    30 0a                                               0.
rig_open: error = Protocol error

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: http://www.n0nb.us  GPG key: D55A8819  GitHub: N0NB

N0NB, 2018-05-25:
It's the netrigctl.c commit from May 9.  Don't quite understand why it's causing this to fail now but I'll chase it down.
 de Mike W9MDB
    On Friday, May 25, 2018, 8:00:00 AM CDT, Nate Bargmann <n0nb@n0nb.us> wrote:  
 
 * On 2018 25 May 06:48 -0500, Nate Bargmann wrote:
> Hello
> The NET rigctl doesn't work on last git version (783ba3edd5236c748a91e3ba6c4fb199f3e46fa9).
> 
> Steps to reproduce the behavior:
> 1. run rigctld -vvvv -m1
> 2. run rigctl -vvvvv -m2 -r127.0.0.1:4532

I can confirm:

$ rigctld -vvvv -m1    
rigctld, Hamlib 3.3~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
dummy: _init called
rig_register called
rig_register: rig_register (1)
rig_register called
rig_register: rig_register (2)
rig_register called
rig_register: rig_register (4)
rig_register called
rig_register: rig_register (5)
dummy_init called
rig_strvfo called
rig_passband_normal called
rig_strvfo called
rig_passband_normal called
rig_open called
port_open called
dummy_open called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
Opened rig model 1, 'Dummy'
rig_strstatus called
Backend version: 0.5, Status: Beta
rig_close called
dummy_close called
port_close called
Closed rig model 1, 'Dummy - will reopen for clients'
Connection opened from localhost:50970
client lock engaged
rig_open called
port_open called
dummy_open called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
Opened rig model 1, 'Dummy'
client lock disengaged
client lock engaged
rig_strvfo called
client lock disengaged
fscanf: Connection reset by peer
fscanf: parsing '�' with '%c'
rig_open called
client lock engaged
rig_close called
dummy_close called
port_close called
Closed rig model 1, 'Dummy - no clients, will reopen for new clients'
client lock disengaged
Connection closed from localhost:50970

and:

$ rigctl -vvvvv -m2 -r127.0.0.1:4532
rigctl, Hamlib 3.3~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
dummy: _init called
rig_register called
rig_register: rig_register (1)
rig_register called
rig_register: rig_register (2)
rig_register called
rig_register: rig_register (4)
rig_register called
rig_register: rig_register (5)
rig_open called
port_open called
network_open called
netrigctl_open called
netrigctl_transaction: called len=12
network_flush called
write_block called
write_block(): TX 12 bytes
0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a                \dump_state.
read_string called
read_string(): RX 2 characters
0000    30 0a                                              0.
rig_open: error = Protocol error

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: http://www.n0nb.us  GPG key: D55A8819  GitHub: N0NB------------------------------------------------------------------------------
Check out the vibrant tech community on one of the world's most
engaging tech sites, Slashdot.org! http://sdm.link/slashdot_______________________________________________
Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2018-05-25:
Resubmitted my outstanding pull request plus the fix for this one.I had misunderstood the logic so reverting back and changing memcmp to strncmp which is more apropos.
de Mike W9MDB

 
 
 

I can confirm:

$ rigctld -vvvv -m1    
rigctld, Hamlib 3.3~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
dummy: _init called
rig_register called
rig_register: rig_register (1)
rig_register called
rig_register: rig_register (2)
rig_register called
rig_register: rig_register (4)
rig_register called
rig_register: rig_register (5)
dummy_init called
rig_strvfo called
rig_passband_normal called
rig_strvfo called
rig_passband_normal called
rig_open called
port_open called
dummy_open called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
Opened rig model 1, 'Dummy'
rig_strstatus called
Backend version: 0.5, Status: Beta
rig_close called
dummy_close called
port_close called
Closed rig model 1, 'Dummy - will reopen for clients'
Connection opened from localhost:50970
client lock engaged
rig_open called
port_open called
dummy_open called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
Opened rig model 1, 'Dummy'
client lock disengaged
client lock engaged
rig_strvfo called
client lock disengaged
fscanf: Connection reset by peer
fscanf: parsing '�' with '%c'
rig_open called
client lock engaged
rig_close called
dummy_close called
port_close called
Closed rig model 1, 'Dummy - no clients, will reopen for new clients'
client lock disengaged
Connection closed from localhost:50970

and:

$ rigctl -vvvvv -m2 -r127.0.0.1:4532
rigctl, Hamlib 3.3~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
dummy: _init called
rig_register called
rig_register: rig_register (1)
rig_register called
rig_register: rig_register (2)
rig_register called
rig_register: rig_register (4)
rig_register called
rig_register: rig_register (5)
rig_open called
port_open called
network_open called
netrigctl_open called
netrigctl_transaction: called len=12
network_flush called
write_block called
write_block(): TX 12 bytes
0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a                \dump_state.
read_string called
read_string(): RX 2 characters
0000    30 0a                                              0.
rig_open: error = Protocol error

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: http://www.n0nb.us GPG key: D55A8819  GitHub: N0NB------------------------------------------------------------------------------
Check out the vibrant tech community on one of the world's most
engaging tech sites, Slashdot.org! http://sdm.link/slashdot_______________________________________________
Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2018-05-25:
* On 2018 25 May 09:48 -0500, Black Michael via Hamlib-developer wrote:
> 
> It's the netrigctl.c commit from May 9.  Don't quite understand why
> it's causing this to fail now but I'll chase it down.

Murphy's Law?

;-)

Thanks, Mike.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: http://www.n0nb.us  GPG key: D55A8819  GitHub: N0NB

N0NB, 2018-05-25:
* On 2018 24 May 23:39 -0500, Michael Tatarinov wrote:
> Hello
> The NET rigctl doesn't work on last git version (783ba3edd5236c748a91e3ba6c4fb199f3e46fa9).
> 
> Steps to reproduce the behavior:
> 1. run rigctld -vvvv -m1
> 2. run rigctl -vvvvv -m2 -r127.0.0.1:4532

This should be corrected with commit d1b545b from Mike, W9MDB.  I have
tested here and it is working.  Please test after pulling or tomorrow's
daily snapshot.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: http://www.n0nb.us  GPG key: D55A8819  GitHub: N0NB

closed 2018-05-29 by kukabu