Hamlib/Hamlib #33: No S-Meter readings for TS590SG

caluml, 2018-06-02:
$ rigctl-wsjtx -vvvvv -m 237 -r /dev/ttyUSB0 
rigctl, Hamlib 3.1~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig:rig_init called 
initrigs3_kenwood called
rig_register (213)
rig_register (201)
rig_register (225)
rig_register (203)
rig_register (204)
rig_register (216)
rig_register (224)
rig_register (205)
rig_register (207)
rig_register (209)
rig_register (210)
rig_register (222)
rig_register (214)
rig_register (230)
rig_register (221)
rig_register (229)
rig_register (238)
rig_register (202)
rig_register (211)
rig_register (206)
rig_register (208)
rig_register (215)
rig_register (226)
rig_register (217)
rig_register (233)
rig_register (220)
rig_register (223)
rig_register (227)
rig_register (234)
rig_register (231)
rig_register (239)
rig_register (237)
rig_register (228)
rig_register (219)
rig_register (232)
rig_register (236)
kenwood_init called
kenwood_init: if_len = 37
rig:rig_open called 
kenwood_open called
kenwood_get_id called
kenwood_transaction called
kenwood_transaction: cmdstr = ID
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string(): RX 6 characters
0000    49 44 30 32 33 3b                                   ID023;          
kenwood_open: found match 023
kenwood_get_trn called
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = AI
write_block(): TX 3 bytes
0000    41 49 3b                                            AI;             
read_string(): RX 4 characters
0000    41 49 30 3b                                         AI0;            
kenwood_set_trn called
kenwood_transaction called
kenwood_transaction: cmdstr = AI0
write_block(): TX 4 bytes
0000    41 49 30 3b                                         AI0;            
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string(): RX 6 characters
0000    49 44 30 32 33 3b                                   ID023;          
kenwood_get_vfo_if called
kenwood_get_if called
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = IF
write_block(): TX 3 bytes
0000    49 46 3b                                            IF;             
read_string(): RX 38 characters
0000    49 46 30 30 30 30 30 31 39 38 30 30 30 20 20 20     IF00000198000   
0010    20 20 20 30 30 30 30 30 30 30 31 30 30 31 30 30        0000000100100
0020    30 30 30 38 30 3b                                   00080;          
Opened rig model 237, 'TS-590SG'
Backend version: 0.9.1, Status: Beta

Rig command: l ?       
Level: AF RF SQL CWPITCH RFPOWER AGC METER SWR ALC STRENGTH 

Rig command: l STRENGTH
Level: kenwood_get_level called
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = SM
write_block(): TX 3 bytes
0000    53 4d 3b                                            SM;             
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
kenwood_transaction: Unknown command or rig busy 'SM'
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = SM
write_block(): TX 3 bytes
0000    53 4d 3b                                            SM;             
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
kenwood_transaction: Unknown command or rig busy 'SM'
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = SM
write_block(): TX 3 bytes
0000    53 4d 3b                                            SM;             
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
kenwood_transaction: Unknown command or rig busy 'SM'
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = SM
write_block(): TX 3 bytes
0000    53 4d 3b                                            SM;             
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
kenwood_transaction: Unknown command or rig busy 'SM'
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = SM
write_block(): TX 3 bytes
0000    53 4d 3b                                            SM;             
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
kenwood_transaction: Unknown command or rig busy 'SM'
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = SM
write_block(): TX 3 bytes
0000    53 4d 3b                                            SM;             
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
kenwood_transaction: Unknown command or rig busy 'SM'
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = SM
write_block(): TX 3 bytes
0000    53 4d 3b                                            SM;             
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
kenwood_transaction: Unknown command or rig busy 'SM'
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = SM
write_block(): TX 3 bytes
0000    53 4d 3b                                            SM;             
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
kenwood_transaction: Unknown command or rig busy 'SM'
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = SM
write_block(): TX 3 bytes
0000    53 4d 3b                                            SM;             
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
kenwood_transaction: Unknown command or rig busy 'SM'
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = SM
write_block(): TX 3 bytes
0000    53 4d 3b                                            SM;             
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
kenwood_transaction: Unknown command or rig busy 'SM'
kenwood_transaction: Retrying shortly
kenwood_transaction: cmdstr = SM
write_block(): TX 3 bytes
0000    53 4d 3b                                            SM;             
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
kenwood_transaction: Unknown command or rig busy 'SM'
get_level: error = Command rejected by the rig

Rig command: l RAWSTR
Level: get_level: error = Invalid parameter

Rig command: m
kenwood_get_mode called
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = MD
write_block(): TX 3 bytes
0000    4d 44 3b                                            MD;             
read_string(): RX 4 characters
0000    4d 44 31 3b                                         MD1;            
kenwood2rmode called
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = DA
write_block(): TX 3 bytes
0000    44 41 3b                                            DA;             
read_string(): RX 4 characters
0000    44 41 30 3b                                         DA0;            
Mode: LSB
Passband: 2200

Rig command: f
kenwood_get_freq called
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = FA
write_block(): TX 3 bytes
0000    46 41 3b                                            FA;             
read_string(): RX 14 characters
0000    46 41 30 30 30 30 30 31 39 38 30 30 30 3b           FA00000198000;  
Frequency: 198000

Rig command: q
rig:rig_close called 
kenwood_close called
kenwood_set_trn called
kenwood_transaction called
kenwood_transaction: cmdstr = AI0
write_block(): TX 4 bytes
0000    41 49 30 3b                                         AI0;            
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string(): RX 6 characters
0000    49 44 30 32 33 3b                                   ID023;          
rig:rig_cleanup called 
kenwood_cleanup called

N0NB, 2018-06-02:
* On 2018 02 Jun 09:58 -0500, caluml wrote:
> $ rigctl-wsjtx -vvvvv -m 237 -r /dev/ttyUSB0 
> rigctl, Hamlib 3.1~git

While I forwarded your email to the mailing list, could you please try
either the 3.2 release or a snapshot of 3.3~git?  There have been two
public releases since 3.1~git was around.

3.3~git snapshots can be found at:

          http://n0nb.users.sourceforge.net

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: http://www.n0nb.us  GPG key: D55A8819  GitHub: N0NB

caluml, 2018-06-06:
I've just tried whatever is the latest in the git repo, and it doesn't even run for me now:

/usr/local/bin/rigctl -m 237 -r /dev/ttyUSB0
rig_open: error = Invalid configuration

Running it with vvvvv gives:
serial_setup: unsupported rate specified: 0
rig_open: error = Invalid configuration 

I tried specifying 115200 as well, but that's just the same.

The one I used previously still works fine (although I can't get the S-Meter readings still)

I'm away for 5 days now, so won't be able to try anything, but will when I get back.

caluml, 2018-06-06:
Hmm, just noticed that whatever I'm currently running is still Hamlib 3.1.
I didn't read the rest of your comment telling me where to get 3.3.
Apologies. I'll try that later or when I get back.

caluml, 2018-06-06:
I compiled from a 3.3 snapshot, and I get this:

/usr/local/bin/rigctl -vvvvv -m 237 -r /dev/ttyUSB0 
rigctl, Hamlib 3.1
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig:rig_init called 
initrigs3_kenwood called
rig_register (213)
rig_register (201)
rig_register (225)
rig_register (203)
rig_register (204)
rig_register (216)
rig_register (224)
rig_register (205)
rig_register (207)
rig_register (209)
rig_register (210)
rig_register (222)
rig_register (214)
rig_register (230)
rig_register (221)
rig_register (229)
rig_register (238)
rig_register (202)
rig_register (211)
rig_register (206)
rig_register (208)
rig_register (215)
rig_register (226)
rig_register (217)
rig_register (233)
rig_register (220)
rig_register (223)
rig_register (227)
rig_register (234)
rig_register (231)
rig_register (239)
rig_register (237)
rig_register (228)
rig_register (219)
rig_register (232)
rig_register (236)
kenwood_init called
kenwood_init: if_len = 37
rig:rig_open called 
serial_setup: unsupported rate specified: 0
rig_open: error = Invalid configuration 

It still says it's 3.1 - is it picking up some old libraries from somewhere?

N0NB, 2018-06-06:
* On 2018 06 Jun 07:40 -0500, caluml wrote:
> I compiled from a 3.3 snapshot, and I get this:
> 
> /usr/local/bin/rigctl -vvvvv -m 237 -r /dev/ttyUSB0 
> rigctl, Hamlib 3.1
> Report bugs to <hamlib-developer@lists.sourceforge.net>

> kenwood_init called
> kenwood_init: if_len = 37
> rig:rig_open called 
> serial_setup: unsupported rate specified: 0
> rig_open: error = Invalid configuration 
> 
> It still says it's 3.1 - is it picking up some old libraries from somewhere?

Did you run 'ldconfig' as root after running 'make install' as root?

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: http://www.n0nb.us  GPG key: D55A8819  GitHub: N0NB

mdblack98, 2019-05-22:
I take it this was never addressed.  I have some time to implement this.  I see 3 values the TS590 can return for SWR, ALC, and COMP.  Main question is what does RM0 return...I imagine it would return the current meter reading.

de Mike W9MDB

mdblack98, 2020-01-31:
SWR ALC and COMP added

closed 2020-01-31 by mdblack98