Hamlib/Hamlib #42: Machine-readable data format / capabilities

mikaelnousiainen, 2018-08-09:
The current data format of rigctld/rotctld responses is quite hard to parse -- especially the capabilities list is a bit of a nightmare. I've been thinking of introducing an alternate data format, e.g. JSON, for this purpose.

There are also some other issues with rig capabilities listings: currently they do not contain all of the information specified in the rig backends. At least TX/RX ranges and ext level definitions (numeric ranges / combo choices) are missing.

Now that the project is heading towards a new major release with possibility of breaking API/ABI, it could be a good time to introduce this kind of change. Obviously we would need a flag/option to specify which data format to use, so that the old one could be preserved for backwards-compatibility.

I'd just like to hear maintaners' opinions/ideas on this to decide whether it makes sense to proceed with implementation. Depending on the data format, we would need to add new dependencies, e.g. jansson for JSON.

N0NB, 2018-08-09:
It sounds useful to me, so go ahead!

mikaelnousiainen, 2019-01-01:
Just updating the status: I'm planning on spending some time on Hamlib code soon and this would be on my todo-list first after a couple of bugfixes to some backends.

mdblack98, 2020-01-18:
Feel free to make a JSON contribution.  Closing this for now.

closed 2020-01-18 by mdblack98