Hamlib/Hamlib #48: Create new rig for Raspberry Pi GPIO pins

dranch, 2018-09-13:
A user on the Raspberry-Pi-for-HAMs list is trying to use the Qsstv program on a Raspberry Pi.  He'd like to use one of the GPIO pins for PTT instead of say the RTS line on a serial port which seems like a good idea.  Qsstv doesn't support GPIO pins natively but it DOES support hamlib.  My question or request to the group is if GPIO pin control could be added into Hamlib.  I think this could be pretty valuable to do various controls of both:

   - INPUT: Get status of things like DCD, position, whatever

   - OUTPUT:  PTT, automate rotators turning

Seems like a good automation interface to me.  Possible?

--David

nouse4anick, 2020-01-15:
whats the current progress on this? I would love to be able to do wspr type modes on my pi but i need GPIO support which the developers for wsjt-x don't want to implement on their end.

73
Art
kc7sda

mdblack98, 2020-01-18:
Capability already in hamlib...duplicate issue here

closed 2020-01-18 by mdblack98

bence98, 2020-01-18:
Our fork works fine, however as it turns out, this feature was already in Hamlib, but not documented in the manpage!

Basically, what you want to do is figure out the physical GPIO pin number of the wanted PTT pin. You can do it by data-mining datasheets, or you can install wiringPi and use the `gpio readall` command which will print a graphic representation of the pin header on your Pi (you'll need the 'BCM' columns from it). Let's say you want pin 17. Then you can use the PTT like this:
```
rigctld -P GPIO -p 17
```
Then connect to `rigctld` with whichever program supports it (QSSTV does, I didn't test WSJT-X).

nouse4anick, 2020-01-22:
so you're telling me this is an undocumented feature that no one supports?

There is nothing in the documentation that says there is support for the pi's gpio. I've done some googling and no one has anything on using the gpio pins to control ptt!

N0NB, 2020-01-22:
Yes...it was undocumented.I just added it to rigctl.1
de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer