Hamlib/Hamlib #52: rotctl: RQUAD USB Interface Rotor Card - wrong azimuth (Yaesu G-450A)

Green678, 2018-09-24:
Hello!
I m using RQUAD USB Interface Rotor Card with Yaesu G450A rotator.
http://www.quad.ru/production/rotator.php
602  Yaesu/Kenpro GS-232

Look this:

r0au@rpi3:~$ rotctl --model=602 --rot-file=/dev/rotator --serial-speed=9600 -C serial_handshake=Hardware,post_write_delay=10,timeout=1000,retry=0 -vvvvv
rotctl, Hamlib 3.2
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
initrots3_gs232a called
rot_register (601)
rot_register (603)
rot_register (602)
rot_register (604)
rot_token_lookup called
rot_confparam_lookup called
rot_set_conf called
rot_confparam_lookup called
rot_set_conf: serial_handshake='Hardware'
frontrot_set_conf called
rot_token_lookup called
rot_confparam_lookup called
rot_set_conf called
rot_confparam_lookup called
rot_set_conf: post_write_delay='10'
frontrot_set_conf called
rot_token_lookup called
rot_confparam_lookup called
rot_set_conf called
rot_confparam_lookup called
rot_set_conf: timeout='1000'
frontrot_set_conf called
rot_token_lookup called
rot_confparam_lookup called
rot_set_conf called
rot_confparam_lookup called
rot_set_conf: retry='0'
frontrot_set_conf called
rot_open called
serial_open called
serial_setup called
Opened rot model 602, 'GS-232'
rig_strstatus called
Backend version: 0.1, Status: Beta

Rotator command: p
rot_get_position called
gs232_rot_get_position called
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.
read_string called
read_string(): RX 6 characters
0000    2b 30 31 33 36 0d                                   +0136.
Azimuth: 163242760729689631064600000529361272832.000000

=============================================
RX 6 Charaster - +0136 - it is normal real azimuth.
But in Azimuth - error )
I think, error in parsing in gs232.c

Can you fix it?
Thanks!
73 de R0AU
mailto r0au@r0au.ru

Green678, 2018-09-25:
===================================
r0au@rpi3:~$ rotctl --model=601 --rot-file=/dev/rotator --serial-speed=9600 -C serial_handshake=Hardware,post_write_delay=10,timeout=1000,retry=0 -vvvvv
rotctl, Hamlib 3.2
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
initrots3_gs232a called
rot_register (601)
rot_register (603)
rot_register (602)
rot_register (604)
rot_token_lookup called
rot_confparam_lookup called
rot_set_conf called
rot_confparam_lookup called
rot_set_conf: serial_handshake='Hardware'
frontrot_set_conf called
rot_token_lookup called
rot_confparam_lookup called
rot_set_conf called
rot_confparam_lookup called
rot_set_conf: post_write_delay='10'
frontrot_set_conf called
rot_token_lookup called
rot_confparam_lookup called
rot_set_conf called
rot_confparam_lookup called
rot_set_conf: timeout='1000'
frontrot_set_conf called
rot_token_lookup called
rot_confparam_lookup called
rot_set_conf called
rot_confparam_lookup called
rot_set_conf: retry='0'
frontrot_set_conf called
rot_open called
serial_open called
serial_setup called
Opened rot model 601, 'GS-232A'
rig_strstatus called
Backend version: 0.3, Status: Beta

Rotator command: p
rot_get_position called
gs232a_rot_get_position called
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.
read_string called
read_string(): RX 6 characters
0000    2b 30 33 31 37 0d                                   +0317.
get_pos: error = Protocol error
===================================

Interesting, but another programs, for example, PstRotatorAZ for Woindows, remoteRotatot for linux, etc, working good!

mdblack98, 2020-01-18:
Fixed

closed 2020-01-18 by mdblack98