Hamlib/Hamlib #64: Wrong signal strength

ntoulasd, 2018-12-08:
I get different values of signal strength when I run from command line and different with rigctl cli
Example:
dimitris@dimitris:~/radio$ rigctl  -m 2 -r localhost l STRENGTH
16
dimitris@dimitris:~/radio$ rigctl

Rig command: l STRENGTH
Level Value: -28

mdblack98, 2018-12-08:
Your second example is using the dummy device.
Tryrigctl -m 2 -r localhost

 de Mike W9MDB

ntoulasd, 2018-12-08:
Yes, you are right about my command test.

 I am trying to find out why I have different values in my ft-857 instrument and in hamlib measure.
Before upgrade 16.04 to 18.04 the values are the same but now I get almost a +30 to hamlib that my meter.

mdblack98, 2018-12-08:
The FT857 code is converting the Rx Status results like this to dB.  That's what rigctl is returning is dB and not the SMeter value.

val->i = n = (p->rx_status & 0x0F)  * 6 - 20;

Turn on debugging and see what the rig is returning from that command and what your meter says.
rigctl -m 2 -vvvvv -r localhost
I did a patch in 2015 based on some testing with somebody.

de Mike W9MDB

ntoulasd, 2018-12-08:
Yes indeed its dB and not Smeter. But Grig which converting dB to Smeter shows a signal of S7 as a signal S9+20

And rigctl gives a signal of -50dB now as a signal -20dB.

Debug give me this :
Rig command: l STRENGTH
rigctl_parse: input_line: l STRENGTH
netrigctl_get_level called
write_block(): TX 11 bytes
0000    6c 20 53 54 52 45 4e 47 54 48 0a                    l STRENGTH.     
read_string(): RX 3 characters
0000    31 36 0a                                            16.             
Level Value: 16

ntoulasd, 2018-12-08:
Without antenna I have -20 (S0) and a full signal from a local repeater gives me 64 (S9+60).
I have to mention again that rigctl (and Grig) at ubuntu 16.04 was working with right values, maybe something have changed.

Thanks, DE SV2RCK

mdblack98, 2018-12-08:
Can you compile hamlib?
Looks like this conversion should work
level*9-66
Though this would make more sense with a level of 0 = -54dB
level*9-54

de Mike W(MDB

ntoulasd, 2018-12-08:
Yes, I can. Where to apply the changes.

mdblack98, 2018-12-09:
Try the attached file to replace ft857.c
This looks like it should work.
I think what we did before was actual calibrated values...which means it won't match what the rig's SMeter says since it's quite wrong (as are most SMeters).
One thing to see if if the values returned by rigctl are the same regardless of your RF gain.  I assume your SMeter changes based on the RF gain since most do (incorrectly).

Mike

mdblack98, 2018-12-14:
> I do some checks of code ft857.c and ft897.c. These two are actually the same hardware with some cosmetic changes. I found that there are differences in the get_smeter_level function.
> Also I found some comments in ft817.c (smaller rig but the same family) that explain dB to S meter conversions.
> Line 629
> /* S-meter value is returned in the lower 4 bits.
> 0x00 = S0 (-54dB)
> 0x01 = S1
> 0x02 = S2
> ...
> 0x09 = S9 (0dB)
> 0x0A = S9+10 (10dB)
> 0x0B = S9+20 and so on
> */
>
> I hope these helps

Did you try the new file I sent you?
Mike

mdblack98, 2018-12-19:
Patch submitted

closed 2018-12-31 by ntoulasd