Hamlib/Hamlib #66: Radio disconnected from USB no warning

2E0PGS, 2018-12-19:
If I have `rigctld` running against a USB port on Linux and I then disconnect the radio from that USB port. When I then go to query `rigctld` via netcat it returns some kind of cached data. I would expect it to return an error indication back to the client code at the very least. I notice the server side daemon does notice somethings a miss on it's verbose output.

mdblack98, 2018-12-19:
Can you shows us the output you're getting?  Unplugging the USB should return an error on the device I'd think.
And run it with 5 v's so we can see the debug info.
rigctld -vvvvv .....
de Mike W9MDB

2E0PGS, 2018-12-19:
Well that is odd! Testing it today it does return a `RPRT -1` as expected. Maybe the otherday when I was hammering the backend with malformed commands it caused it to get confused so a fresh run of `rigctld` today cleared any issues out.

I will continue to check it and close this issue if I cannot replicate it today.

2E0PGS, 2018-12-19:
It appears to be functioning as expected now thanks for confirmation on the other issue I think that was what caused this issue. I will re-open if the issue crops back up again.

closed 2018-12-19 by 2E0PGS