Hamlib/Hamlib #67: Multiple commands rigctld

2E0PGS, 2018-12-19:
Hi,

Is it possible to send multiple commands in one request? I have had some success doing so by seperating commands with a space however it doesn't like if you then mix in commands that require an argument. The verbose debug output of `rigctld` says unknown command a few times.

I tried looking at the manpages for any delim char for multiple arguments but there wasn't anything I could find. 

Maybe this just isn't intended functionallity and I should stick to single commands at a time when querying `rigctld` using netcat?

mdblack98, 2018-12-19:
Single commands are the only way right now.  Doesn't really buy you anything to do multiple.  
de Mike W9MDB

2E0PGS, 2018-12-19:
Hi,

Thanks for confirmation that may have been what caused issue #66 when I tried sending the daemon multiple commands.

closed 2018-12-19 by 2E0PGS