Hamlib/Hamlib #79: Supported modes missing from Kenwood TS-590?

jonathonadler, 2019-01-01:
My Kenwood TS-590 seems to support the modes RIG_MODE_PKTFM, RIG_MODE_PKTLSB and RIG_MODE_PKTUSB. However, these modes are note defined in kenwood/ts590.c

```c ts590.c
#define TS590_ALL_MODES (RIG_MODE_AM|RIG_MODE_CW|RIG_MODE_CWR|RIG_MODE_SSB|RIG_MODE_FM|RIG_MODE_RTTY|RIG_MODE_RTTYR)
```

Should the PKT modes be added to this line or am I missing something?

N0NB, 2019-01-01:
Where do you see PKT modes supported?These are the modes in the 590 reference manual that I have which is dated 2011.   http://www.kenwood.com/i/products/info/amateur/pdf/ts_590_pc_command_e.pdfde Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2019-01-01:
Actually I now see we have code in kenwood.c to allow for those 3 modes for the TS590.  So I added those modes and have the pull request in.
On my forkhttps://github.com/Hamlib/Hamlib/compare/master...mdblack98:master

de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

jonathonadler, 2019-01-02:
I actually have the TS-590S, not the TS-590. Perhaps the S model has a different feature set? Here is the manual for the TS-590S: http://manual.kenwood.com/files/4f46bfdf6b75c.pdf. On page 11 it mentions the DATA modes. 

The following commands all work as expected:

Rig command: M PKTLSB 0
Rig command: M PKTUSB 0
Rig command: M PKTFM 0

The issue is just that these modes are not reported as supported within modes_list. I didn't see any differences on your fork. Was the URL you posted correct?

jonathonadler, 2019-01-02:
I recompiled and tested https://github.com/N0NB/Hamlib/commit/681404e6b7f128f30d133ab0ecb49be8ee7d2622. It worked. Thanks!!

closed 2019-01-02 by jonathonadler