Hamlib/Hamlib #81: FT-818 support?

clumens, 2019-01-02:
Is the FT-818 expected to work with hamlib-3.3?  I figured it'd be similar enough to the FT-817 that it'd just work, but my brief testing on the command line didn't give any positive results.  I was unable to get/set the frequency, for instance.  I have the hardware, free time, and coding ability to work on this if needed.  I'd just need some pointers for how to get started.  Thanks!

N0NB, 2019-01-02:
* On 2019 02 Jan 10:20 -0600, Chris Lumens wrote:
> Is the FT-818 expected to work with hamlib-3.3?  I figured it'd be
> similar enough to the FT-817 that it'd just work, but my brief testing
> on the command line didn't give any positive results.  I was unable to
> get/set the frequency, for instance.

Per the operating manual:

https://www.yaesu.com/downloadFile.cfm?FileID=13547&FileCatID=158&FileName=FT%2D818ND%5FOM%5FENG%5FE13772002%5F1802S%2DCM.pdf&FileContentType=application%2Fpdf

it appears the '818 uses teh classic Yaesu CAT command set.  Its
documentation begins on page 64 of the manual.

> I have the hardware, free time, and coding ability to work on this if
> needed.  I'd just need some pointers for how to get started.  Thanks!

Start by comparing the manual for the '818 commands with those of the
yaesu/ft817.c backend file and work from there.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: http://www.n0nb.us  GPG key: D55A8819  GitHub: N0NB

clumens, 2019-01-03:
I don't have a login for the SF mailing list, but it looks like these reports get mirrored over there, so good enough.  Here's the output of `rigctl -m 120 -r /dev/ttyUSB0 -vvvvv f`:

```rigctl, Hamlib 3.3
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
yaesu: initrigs3_yaesu called
rig_register called
rig_register: rig_register (121)
rig_register called
rig_register: rig_register (127)
rig_register called
rig_register: rig_register (110)
rig_register called
rig_register: rig_register (105)
rig_register called
rig_register: rig_register (106)
rig_register called
rig_register: rig_register (107)
rig_register called
rig_register: rig_register (109)
rig_register called
rig_register: rig_register (120)
rig_register called
rig_register: rig_register (101)
rig_register called
rig_register: rig_register (122)
rig_register called
rig_register: rig_register (123)
rig_register called
rig_register: rig_register (111)
rig_register called
rig_register: rig_register (115)
rig_register called
rig_register: rig_register (113)
rig_register called
rig_register: rig_register (114)
rig_register called
rig_register: rig_register (128)
rig_register called
rig_register: rig_register (131)
rig_register called
rig_register: rig_register (116)
rig_register called
rig_register: rig_register (103)
rig_register called
rig_register: rig_register (124)
rig_register called
rig_register: rig_register (104)
rig_register called
rig_register: rig_register (125)
rig_register called
rig_register: rig_register (129)
rig_register called
rig_register: rig_register (132)
rig_register called
rig_register: rig_register (130)
rig_register called
rig_register: rig_register (117)
rig_register called
rig_register: rig_register (119)
rig_register called
rig_register: rig_register (118)
rig_register called
rig_register: rig_register (126)
rig_register called
rig_register: rig_register (133)
rig_register called
rig_register: rig_register (134)
rig_register called
rig_register: rig_register (135)
rig_register called
rig_register: rig_register (136)
ft817: ft817_init called 
rig_open called
port_open called
serial_open called
serial_setup called
ft817: ft817_open called 
rig_get_vfo called
Opened rig model 120, 'FT-817'
rig_strstatus called
Backend version: 0.5.1, Status: Beta
rig_get_freq called
ft817: cache invalid
serial_flush called
write_block called
write_block(): TX 5 bytes
0000    00 00 00 00 03                                      .....           
read_block called
read_block(): RX 5 bytes
0000    00 00 00 00 03                                      .....           
write_block called
write_block(): TX 5 bytes
0000    00 64 00 00 bb                                      .d...           
read_block called
read_block(): RX 2 bytes
0000    00 64                                               .d              
from_bcd_be called
0
rig_close called
ft817: ft817_close called 
port_close called
ser_close called
rig_cleanup called
ft817: ft817_cleanup called 
```

N0NB, 2019-01-03:
It shouldn't be the USB device.  Should be /dev/ttyS0 or /dev/ttyS1Also be sure to use the "-s 9600" or whatever your baud rate is.
de Mike W9MDB

 
 

Hamlib-developer mailing list
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

clumens, 2019-01-03:
> It shouldn't be the USB device.  Should be /dev/ttyS0 or /dev/ttyS1Also be sure to use the "-s 9600" or whatever your baud rate is.

Yeah, /dev/ttyUSB0 should be the right device.  I'm using a similar sort of cable for controlling my IC-7000 and hamlib is cool with that.  Here's what dmesg thinks:

```[13878.800449] usb 1-1.6: new full-speed USB device number 3 using ehci-pci
[13878.880449] usb 1-1.6: New USB device found, idVendor=067b, idProduct=2303
[13878.880452] usb 1-1.6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[13878.880454] usb 1-1.6: Product: USB-Serial Controller
[13878.880455] usb 1-1.6: Manufacturer: Prolific Technology Inc.
[13878.932068] usbcore: registered new interface driver pl2303
[13878.932077] usbserial: USB Serial support registered for pl2303
[13878.932092] pl2303 1-1.6:1.0: pl2303 converter detected
[13878.933566] usb 1-1.6: pl2303 converter now attached to ttyUSB0
```

I've double checked the port speed as well and that should be correct, though I can reset it to the default on the radio if required.

N0NB, 2019-01-03:
Do you maybe have an early serial# FT-818?
The 00 00 00 00 03 is the correct command for reading freq & mode.But it just echoed the response right back which, if correct, would be 0Hz and CWR which doesn't sound right.
Try running a few of the commands from rigctl (drop the trailing f), get freq, set freq, get mode, set mode, ptt on/on.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

clumens, 2019-03-10:
I finally got around to testing this with a different cable, and it works (at least, gpredict can sit there and control the frequency).  I'll just replace my cable and close this issue out.  Thanks for your time.

closed 2019-03-10 by clumens