Hamlib/Hamlib #98: TH-D-72a  rigctld and gpredict..  unable to control radio through gpredict

jrsphoto, 2019-04-14:
I'm using rigctl 3.3 and I realize the TH-D72a driver is ALPHA but I was doing some testing for an upcoming project things didn't get off to a good start.

I start things off with:
rigctld -m 233 -r /dev/ttyUSB0 -s9600 -vvvvv

that seems to start ok, and I'm seeing the typical calls go by and no error indications.

Then I launch gpredict where I have the radio configured in preferences as Duplex TRX, PTT Status=none, and VFO up/down is MAIN up/SUB down.

When I bring up the GPredict Radio Control panel and click Engage, it connects to rigctld but after several attempts and some errors, it fails and disconnects from rigctld.

Thanks
[rigctld_output.txt](https://github.com/Hamlib/Hamlib/files/3077881/rigctld_output.txt)

mdblack98, 2019-04-14:
Are you compiling hamlib yourself?
If so, in thd72.c try changing
.set_freq = thd72_set_freq,
to.set_freq = th_set_freq,

And see if that works
de Mike W9MDB

jrsphoto, 2019-04-14:
Thanks for the feedback Michael;

So I made the suggested changes to ~/hamlib/kenwood/thd72.c and recompiled.

This time I see some errors about unknown command or rig busy, followed by some retries.  I did this test several times and on one occasion, it seemed to be caught in a loop, retrying the commands, and never stopped.  I had to kill the rigctld instance.  I've attached the output.

[rigctld_output2.txt](https://github.com/Hamlib/Hamlib/files/3077972/rigctld_output2.txt)
[image]

jrsphoto, 2019-04-15:
When looking at both logfiles side by side, it seems to me, that the first log file (before I modified thd72.c) has more reasonable command responses, in addition,  In that first log file, I only see one actual error:

`read_string called
read_string(): RX 2 characters
0000    3f 0d                                               ?.              
kenwood_transaction: Unknown command or rig busy ''
kenwood_transaction: Retrying shortly`

About the only other thing in that first log file that seems wrong is the "kenwood_transaction: NegAck" lines:

`read_string called
read_string(): RX 2 characters
0000    4e 0d                                               N.              
kenwood_transaction: NegAck for 'FO 1,0145890000,7,0,0,1,0,0,0,18,08,000,0,05000000,0'`

I have no idea what to make of this, just trying to do a little debugging on my own ;-)
-John

mdblack98, 2019-04-15:
The 2nd file has bad VFO info going on which hopefully is corrected in the two files I sent you.But I didn't change the set_freq to the th_set_freq in the ones I sent.You're showing the th72d_set_freq command below which appears to be invalide.
Are you running with the two files I sent?  And then send the log file?  Also add the -Z switch so we can some timing.
Mike

jrsphoto, 2019-04-15:
Did you email me files?

mdblack98, 2019-04-15:
Let's try again...this time via dropbox....
https://www.dropbox.com/s/poj8sfr743gwaiu/kenwood.c?dl=1
https://www.dropbox.com/s/bnq0rozv29rv5qi/thd72.c?dl=1

jrsphoto, 2019-04-15:
cool, that worked.. I assume you want me to make the same changes to the thd72.c file?

jrsphoto, 2019-04-15:
looks like you already did.. I'll test this out now

mdblack98, 2019-04-15:
I made the set_freq change in the dropbox files...but it appears the thd72_set_freq is definitely the wrong way to do it...
Mike

jrsphoto, 2019-04-15:
I'm getting a bunch of compile errors now, output below, and are you on any
discord servers?  might be easier if you want...

Making all in kenwood
make[1]: Entering directory '/home/john/hamlib-3.3/kenwood'
  CC       kenwood.lo
kenwood.c:89:5: error: 'RIG_MODEL_TS890S' undeclared here (not in a
function)
   { RIG_MODEL_TS890S, 24 },
     ^
kenwood.c:118:5: error: 'RIG_MODEL_THD74' undeclared here (not in a
function)
   { RIG_MODEL_THD74, "TH-D74" },
     ^
kenwood.c:118:5: error: initializer element is not constant
kenwood.c:118:5: note: (near initialization for
'kenwood_id_string_list[23].model')
kenwood.c:138:10: error: 'RIG_MODE_PSK' undeclared here (not in a function)
   [10] = RIG_MODE_PSK,
          ^
kenwood.c:138:10: error: initializer element is not constant
kenwood.c:138:10: note: (near initialization for 'kenwood_mode_table[10]')
kenwood.c:139:10: error: 'RIG_MODE_PSKR' undeclared here (not in a function)
   [11] = RIG_MODE_PSKR,
          ^
kenwood.c:139:10: error: initializer element is not constant
kenwood.c:139:10: note: (near initialization for 'kenwood_mode_table[11]')
kenwood.c: In function 'kenwood_set_func':
kenwood.c:2037:8: error: 'RIG_FUNC_NB2' undeclared (first use in this
function)
   case RIG_FUNC_NB2:
        ^
kenwood.c:2037:8: note: each undeclared identifier is reported only once
for each function it appears in
kenwood.c:2039:26: warning: comparison between pointer and integer
     if (RIG_MODEL_TS890S == rig->caps->rig_model) {
                          ^
kenwood.c:2062:26: warning: comparison between pointer and integer
     if (RIG_MODEL_TS890S == rig->caps->rig_model) {
                          ^
kenwood.c:2087:26: warning: comparison between pointer and integer
     if (RIG_MODEL_TS890S == rig->caps->rig_model) {
                          ^
kenwood.c:2101:8: error: 'RIG_FUNC_BC2' undeclared (first use in this
function)
   case RIG_FUNC_BC2:
        ^
kenwood.c: In function 'kenwood_get_func':
kenwood.c:2185:26: warning: comparison between pointer and integer
     if (RIG_MODEL_TS890S == rig->caps->rig_model) {
                          ^
kenwood.c:2190:8: error: 'RIG_FUNC_NB2' undeclared (first use in this
function)
   case RIG_FUNC_NB2:
        ^
kenwood.c:2220:8: error: 'RIG_FUNC_BC2' undeclared (first use in this
function)
   case RIG_FUNC_BC2: // TS-890 check Beat Cancel 2 we return boolean
true/false
        ^
kenwood.c: In function 'kenwood_get_trn':
kenwood.c:2774:28: warning: comparison between pointer and integer
   if (rig->caps->rig_model == RIG_MODEL_THD74)
                            ^
kenwood.c:2781:28: warning: comparison between pointer and integer
   if (rig->caps->rig_model == RIG_MODEL_THD74)
                            ^
kenwood.c: In function 'initrigs3_kenwood':
kenwood.c:3519:17: error: 'thd74_caps' undeclared (first use in this
function)
   rig_register(&thd74_caps);
                 ^
kenwood.c:3519:16: warning: passing argument 1 of 'rig_register' from
incompatible pointer type [-Wincompatible-pointer-types]
   rig_register(&thd74_caps);
                ^
In file included from kenwood.c:35:0:
../include/hamlib/rig.h:2204:1: note: expected 'const struct rig_caps *'
but argument is of type 'const struct kenwood_id_string (*)[1]'
 rig_register HAMLIB_PARAMS((const struct rig_caps *caps));
 ^
kenwood.c:3535:17: error: 'ts890s_caps' undeclared (first use in this
function)
   rig_register(&ts890s_caps);
                 ^
kenwood.c:3535:16: warning: passing argument 1 of 'rig_register' from
incompatible pointer type [-Wincompatible-pointer-types]
   rig_register(&ts890s_caps);
                ^
In file included from kenwood.c:35:0:
../include/hamlib/rig.h:2204:1: note: expected 'const struct rig_caps *'
but argument is of type 'const struct kenwood_id_string (*)[1]'
 rig_register HAMLIB_PARAMS((const struct rig_caps *caps));
 ^
Makefile:512: recipe for target 'kenwood.lo' failed
make[1]: *** [kenwood.lo] Error 1
make[1]: Leaving directory '/home/john/hamlib-3.3/kenwood'
Makefile:592: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1

mdblack98, 2019-04-15:
I forgot you're on 3.3.   We aren't making any changes there.
Can you please checkout my fork and replace those two files in there?
https://github.com/mdblack98/Hamlib

Mike

mdblack98, 2020-01-16:
What the status of this?
de Mike W9MDB

mdblack98, 2020-03-05:
Fixed

closed 2020-03-05 by mdblack98