Hamlib/Hamlib #101: Icom manual notch filter requency

mikaelnousiainen, 2019-05-09:
I'm in the process of fixing implementation of many Icom CI-V commands that currently either do not work at all or work incorrectly when compared to the Hamlib API spec.

However, there's one particular command that is difficult to implement with Icom radios and that is the manual notch frequency (`RIG_LEVEL_NOTCHF`). Based on Icom radio manuals I've read (and I've checked most of the modern ones >= IC-7000), the scale for adjusting manual notch frequency is simply 0 to 255 without any indication of the actual frequency.

After all, the Hamlib API specifies `RIG_LEVEL_NOTCHF` as:
```
    RIG_LEVEL_NOTCHF =      (1 << 15),      /*!< \c NOTCHF -- Notch Freq., arg int (Hz) */
```
indicating that the argument should be the frequency in Hz.

I'm wondering if anyone knows better whether the notch frequency value for Icom rigs actually corresponds to a specific frequency or if it is relative to, for example, the filter width currently selected?

mikaelnousiainen, 2019-05-11:
Thanks, that kind of makes sense, but what exactly do you mean with "display width" and how do you _change_ "display width"? I've got an IC-7300 too, but couldn't really figure out how the notch range works...

Anyway, I think it makes more sense to use a custom (extra) level with range -127 to 128, as otherwise the API is misleading and it's more or less impossible to use the feature.

What do you think?

I'm also adding more detailed enumeration of extra level/parameter details in `dump_caps`, so the ranges and combo choices are listed.

N0NB, 2019-05-12:
It's the SPAN button.

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2019-05-12:
Don't confuse the hamlib API with a GUI or user interface.  I think keeping the parameter matching the rig is a bit better.  Any GUI you wrap around it would do the centered value like you are thinking.

de Mike W9MDB

mikaelnousiainen, 2019-05-12:
Thanks for the comments.

@mdblack98 My point is exactly that now it's very difficult to develop a GUI (which I've done) that works properly with Icom rigs. Let me explain:

1) For rigs that conform to Hamlib API (e.g. some Kenwood ones), the value for manual notch is in Hz (as specified in the API docs): the UI can display the unit (Hz) and the value is easy to adjust, e.g. 1500 Hz when using SSB with filter of 2.4kHz.

2) For Icom rigs, the value range is always 0 to 255, which is not in Hz and it is not even a constant scale, as it depends on rig settings. Even if the range (`level_gran`) would be specified by the rig (which it is not currently), the resulting value between 0 and 255 is confusing. Because of the limitations of Icom rigs, the backend does not (and cannot) know the actual value in Hz.

Alternatively, we could add another type of (standardized) manual notch level that is a float value from 0 to 1.0, which does not specify any particular scale or unit, so that a GUI can present the scale and the value in a user-friendly way.

mikaelnousiainen, 2019-05-12:
@N0NB @mdblack98 Explaining the problem further: the issue is simply the fact that because of the Hamlib API definition, the UI _must_ assume that the manual notch value unit is Hz and that it is an integer value. It would be more natural to have a floating point value without any unit for Icom rigs. Having the UI display "technical" values where the user just "has to know" what the values mean (and the UI is misleading the user) is something I would like to fix here.

As we are still on the way to Hamlib 4.x release, this would be a good time to add more levels, although it does require new clients to use the new level instead of the old one for Icom rigs. But that's what major releases are useful -- it allows one to break the API (even if this change is not really breaking the API, but simply changing the level definitions, which should be possible with *any* release...)

mdblack98, 2019-05-12:
I see what you mean now.  I don't know of any software that messes with the notch filter....one of the more advanced settings not used very often from software.
So since we're in 4.0 and allowed to "break" things right now perhaps it's a good time to fix that.  
The problem is the different representations where the 0-1 scale isn't a good representation for all rigs.  The fix to me seems to be to add a parameter that tells you the meaning of the scale.  That would leave the setting still in the rig's vernacular but allow a software package to determine what the units are.  But then the question is can you actually tell the Hz representation from the Icom rigs?  If not, then the NF scale parameter would be zero meaning "don't know".  That should cover the uses.   But we may want a new level added for a scaled input/output...that may only work for some rigs. 
de Mike W9MDB

mikaelnousiainen, 2019-05-12:
@mdblack98 Yes, it is definitely one of the rarely used features, but is definitely useful with remote control applications, such as the one I'm building.

The problem with Icom rigs is that it is not possible to tell the frequency range, as it seems to vary between rigs and it also depends on other settings in the rig, such as waterfall span in IC-7300, like @N0NB mentioned.

I was wondering if we could leave the current `RIG_LEVEL_NOTCHF` as it is (as many rigs do support specifying the filter using an absolute frequency) and introduce a new level for rigs like Icom that do not specify any unit for the notch filter settings. That new level could then use a float value with either range of -1..1 or 0..1. Not sure what that should be called... `RIG_LEVEL_NOTCHF_SIMPLE`? Ideas? :)

mdblack98, 2019-05-13:
I wouldn't use "NOTCHF_SIMPLE" since I dont' really see where SIMPLE applies.  It's just different so perhaps just RIG_LEVEL_NOTCHF2.
And I'd stick with the 0..1 range as it's pretty standard.

Mike

mikaelnousiainen, 2019-09-10:
This was implemented as `NOTCHF_RAW` in #112

closed 2019-09-10 by mikaelnousiainen