Hamlib/Hamlib #108: rotctl azimuth offset

mariobaldini, 2019-05-28:
Hello, 

Is there a simple way for setting an offset in the azimuth? (for a case where an existing antena/rotor is installed with a misalignment).

Mechanically fixing would be better, but considering that it can not be done, how could I fix this in software? 

I intend to use Gpredict + hamlib to control the rotator, so if I could configure one of them, without writing new code, would be preferable. Or at least make a patch/upstream, so no extra software module need to be maintained. 

Thank you,

Mario

N0NB, 2019-05-28:
On 28/05/2019 16:44, Mario Baldini wrote:
> Hello,
>
> Is there a simple way for setting an offset in the azimuth? (for a
> case where an existing antena/rotor is installed with a misalignment).
>
> Mechanically fixing would be better, but [...]

Mario.

You may want to also ask on the Gpredict support list, I have a nagging
feeling that the latest version(s) have something in there that may
accommodate such needs.

https://sourceforge.net/projects/gpredict/lists/gpredict-discussion

Or the Libre.space community:-

https://community.libre.space/

73.

Dave B G0WBX.

-- 
Created on and sent from a Unix like PC running and using free and open source software.
::

N0NB, 2019-05-28:
On 28/05/2019 16:44, Mario Baldini wrote:
> Hello,
>
> Is there a simple way for setting an offset in the azimuth? (for a
> case where an existing antena/rotor is installed with a misalignment).
>
> Mechanically fixing would be better, but [...]

If Hamlib can't do it, then the Gpredict list would be the place to
ask.  Plenty of good people (and it's author Alexandru Csete) monitor
that list.

https://community.libre.space/c/gpredict

Though, as always, best to start a new thread.

It's also relatively easy (if you can find your way around Python) to
intercept the call's and responses passing between Gpredict and Rotctld,
to "adjust" some behaviour.

73.

Dave B G0WBX.

-- 
Created on and sent from a Unix like PC running and using free and open source software.
::

mdblack98, 2019-12-03:
I've just submitted a patch that adds azimuth and elevation offsets to rigctl and rigctld
Use my fork
https://github.com/mdblack98/Hamlib

mdblack98, 2020-01-18:
This has been fixed.

closed 2020-01-18 by mdblack98