Hamlib/Hamlib #122: rotctl: broken/stuttering control of G-5500 with GS-232B

rbw42, 2019-07-29:
Hi Folks,
I have a G-5500 rotator and GS-232B controller. I am driving the controller through a USB->RS232 converter (Prolific pl2303 converter) from a raspberry pi.

When I try driving the rotator with rotctl, the rotator moves, but stutters (i.e. what should be a single drive to the new az/el is broken up). When I drive the rotator directly via the GS-232B (using ```screen /dev/ttyUSB0``` to directly use the serial port), there is no such problem.

the output of rotctl is below:
```
mypi:~/Hamlib/tests $ ./rotctl -m 603 -r /dev/ttyUSB0 -vvvvv
rotctl, Hamlib 4.0~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
initrots4_gs232a called
rot_register (601)
rot_register (602)
rot_register (603)
rot_register (604)
rot_register (605)
rot_register (606)
rot_register (607)
rot_register (608)
rot_open called
serial_open called
serial_setup called
serial_flush called
Opened rot model 603, 'GS-232B'
rig_strstatus called
Backend version: 0.4, Status: Beta

Rotator command: p
rot_get_position called
gs232b_rot_get_position called
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.             
read_string called
read_string(): RX 15 characters
0000    41 5a 3d 31 38 30 20 20 45 4c 3d 30 39 30 0d        AZ=180  EL=090. 
gs232b_rot_get_position: (az, el) = (180.0, 90.0)
Azimuth: 180.000000
Elevation: 90.000000

Rotator command: P 180 100
Azimuth: Elevation: rot_set_position called
gs232b_rot_set_position called: 180.000000 100.000000
serial_flush called
write_block called
write_block(): TX 9 bytes
0000    57 31 38 30 20 31 30 30 0d                          W180 100.       
read_string called
read_string(): Timed out 0.400488 seconds after 0 chars
serial_flush called
write_block called
write_block(): TX 9 bytes
0000    57 31 38 30 20 31 30 30 0d                          W180 100.       
read_string called
read_string(): Timed out 0.400471 seconds after 0 chars
serial_flush called
write_block called
write_block(): TX 9 bytes
0000    57 31 38 30 20 31 30 30 0d                          W180 100.       
read_string called
read_string(): Timed out 0.400485 seconds after 0 chars
serial_flush called
write_block called
write_block(): TX 9 bytes
0000    57 31 38 30 20 31 30 30 0d                          W180 100.       
read_string called
read_string(): Timed out 0.400513 seconds after 0 chars
set_pos: error = Communication timed out

Rotator command: q
rot_close called
ser_close called
ser_close: restoring options
rot_cleanup called
```

Having looked through the code a bit, the core of the command seems to be the gs232b_transaction() function, which writes a command then waits for a reply. The problem, as far as I can tell, is that the GS-232B does not reply at all when given a pointing command, it only replies when it is given a "get" type command. (This can be easily verified by using the direct serial console.) Each time the wait for the reply times out, it does a serial flush and re-issues the command, which makes the controller stutter.

So question is: have I missed something in the use of rotctl here? Or maybe something has changed in the behaviour of the GS-232B compared to the past?

Thanks,
Randall

N0NB, 2019-07-29:
The code is written to always look for a reply...I suspect that you may see a reply when the rotor is done.
What does the rotor stop command do?  Does it get an immediate reply?  Or no reply at all?
Also...if you move the rotor just a few degrees do you see a reply then?  (i.e. short enough time so it doesn't time out).
de Mike W9MDB 

 
 

Hamlib-developer mailing list
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2019-07-29:
Good evening,

I had a similar request in September last year - Magne LA1BFA sent me 
some files to make my rotator work with version 3.1 of Hamlib.
Below the original mail and  as attachment the files he sent me (all 
dated from 2018).
Maybe this could solve the problem and be integrated !?

Here the original mail:

Hello all,

Thanks for accepting me to the list!

I own a portable rotator and would like to use it with rotctl(d) under 
Linux.

The protocol used is rather similar to Yaesu GS232B - it has the parsing 
for azimuth value included in function gs232b_rot_get_position()

Portable Rotation
Model 12PR1A
Serial: 9600 baud, 1 Start, 1 Stop, No Parity, No Handshake
Commands:
L        Start turn left (not needed)
R        Start turn right (not needed)
A or S   Stop all turn
MXXX     Auto turn to new heading (XXX = 000 to 360)
C        Report current heading - replies with 'AZ=XXX<CR>'

for the last command, the function gs232b_rot_get_position() sends the 
command C2 (Yaesu GS232B) which causes the rotator to not respond 
correctly.
Also, as it is a single axis rotator there is no communication about 
elevation.

It would be great if I could use this rotator with my well-known software.

Thanks to everyone for the great work you're doing.

73 de Peter ON4EZJ

rbw42, 2019-07-30:
Hi,
Thanks for looking at this. To answer your question:

> What does the rotor stop command do?  Does it get an immediate reply?  Or no reply at all?

Issuing a stop command gets no response at all.

I have tried looking at the raw serial port output directly by `cat -A < /dev/ttyUSB0` and I can see the carriage return characters and response when I issue valid commands or just press enter. (If I press enter, I get `?>^M$`, which is exactly what I expect based on the cat man page.)

The manual for the GS-232B says:

> Commands are not echoed by the Control Interface, but a carriage return character 0xD is returned after every command, and also a line feed character 0xA if the command invoked returned data.

This is consistent with what I see, where linux correctly interprets the linefeeds when something is actually returned. So according to this, I should be receiving a single carriage return after issuing the rotation command, but I am getting nothing. This is the crux of the problem: even a single carriage return character would satisfy the read in the gs232b_transaction() function.
I have tried changing the serial port settings, including going to raw mode, and nothing seems to make any difference.  Unfortunately I don't have another computer with a native RS232 interface, so it is not possible for me to check if the GS-232B is actually generating an output or not. Very frustrating.

Thanks again,
Randall.

N0NB, 2019-07-30:
Do you have the ability to compile hamlib?
I have to wonder if this is a behavior just of this one model or if all do the same (no response) on those two commands.
I would assume we can poll the rotor position while it's moving?
Mike

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

rbw42, 2019-08-04:
Hi again,

I have just confirmed with a handheld DSO that the GS-232B does NOT emit any signal at all after being sent a pointing command (e.g. "W180 100").

Example of DSO screen, when the trigger is set to capture a snapshot. (Trigger is on the yellow signal, which is the output of the GS-232B, the cyan signal is the input to the GS232B). In this case, I have just pressed the enter key in a serial console, and the GS-232B returns ">$CRLF" where CR is the carriage return character and LF is the line feed character. This matches what is described in the manual.

[image]

If I issue a command to get the current pointing ("C"), I see:
[image]

We can't see the command in this instance - the DSO is triggered on the reply from the DS-232B. This indicates there is a delay > 5 ms between receiving the command and responding, which seems reasonable in this case.

When I issue a command to point "W180 100", there is NO response at all from the GS-232B. I have to press enter another time to get the default "$>" response, at which point the DSO will trigger and capture the waveform.

So at this point I can pretty definitively say that the GS-232B is not working as per the manual. Can any other GS-232B users confirm they have seen this behaviour?

Thanks,
Randall VK6WR.

rbw42, 2019-08-08:
Hi all,

Ok, this is my last post with an update and suggested fix. All of the commands that are "write only" on the GS232B don't return anything. I have also confirmed this by looking at the log of a colleague who uses windows and Ham Radio Deluxe. The relevant part of the log is:

```
11:04:12  COM6        Get status> Send 43-32-0D
11:04:12  COM6        Get status> Recv 41-5A-3D-31-34-33-20-20-45-4C-3D-30-30-31-0D-0A-00-00-00-00
11:04:12  COM6        Status AZ:143.0, EL:1.0
11:04:15  COM6        Get status> Send 43-32-0D
11:04:16  COM6        Get status> Recv 41-5A-3D-31-34-33-20-20-45-4C-3D-30-30-31-0D-0A-00-00-00-00
11:04:16  COM6        Status AZ:143.0, EL:1.0
11:04:16  COM6        Setting position to AZ:288.0° (288.8°, Curr: 143.0°), EL:0.0° (0.0°, Curr: 1.0°), North: Yes
11:04:16  COM6        Purging channel 000003BC, flags 0000000F
11:04:16  COM6        Set position> Send 57-32-38-38-20-30-30-30-0D
11:04:18  COM6        Setting position to AZ:288.0° (288.7°, Curr: 143.0°), EL:0.0° (0.0°, Curr: 1.0°), North: Yes
11:04:18  COM6        Purging channel 000003BC, flags 0000000F
11:04:18  COM6        Set position> Send 57-32-38-38-20-30-30-30-0D
11:04:19  COM6        Get status> Send 43-32-0D
11:04:19  COM6        Get status> Recv 41-5A-3D-31-35-34-20-20-45-4C-3D-30-30-30-0D-0A-00-00-00-00
11:04:19  COM6        Status AZ:154.0, EL:0.0
11:04:20  COM6        Setting position to AZ:288.0° (288.5°, Curr: 154.0°), EL:0.0° (0.0°, Curr: 0.0°), North: Yes
11:04:20  COM6        Purging channel 000003BC, flags 0000000F
11:04:20  COM6        Set position> Send 57-32-38-38-20-30-30-30-0D
11:04:22  COM6        Get status> Send 43-32-0D
11:04:22  COM6        Get status> Recv 41-5A-3D-31-36-39-20-20-45-4C-3D-30-30-30-0D-0A-00-00-00-00
11:04:22  COM6        Status AZ:169.0, EL:0.0

```
You can see that HRD does not look for replies on "write only" commands, only for "query" type commands.

The solution for hamlib is to stop the function gs232b_transaction() in gs232b.c from expecting a reply for "write only" commands. The easiest way to achieve this is to return success after issuing the command successfully if the return buffer is NULL. Here's my one-line diff:

```
rwayth@mypi:~/Hamlib $ git diff gs232a/gs232b.c
diff --git a/gs232a/gs232b.c b/gs232a/gs232b.c
index f2dbbe91..a39537d6 100644
--- a/gs232a/gs232b.c
+++ b/gs232a/gs232b.c
@@ -81,7 +81,8 @@ transaction_write:
 
     /* Always read the reply to know whether the cmd went OK */
     if (!data)
-        data = replybuf;
+        return RIG_OK;
+       //        data = replybuf;
     if (!data_len)
         data_len = BUFSZ;
```

Incidentally, this also fixes the direct "move" commands in rotctl which I just confirmed also did not work (for the same reason) but work after the fix.

Regards,
Randall.

N0NB, 2019-08-08:
Can you try this change and see if it works?
I assume sending a carriage return does get an answer after sending such a command.
So change
    if (cmdstr) {        retval = write_block(&rs->rotport, cmdstr, strlen(cmdstr));        if (retval != RIG_OK)            goto transaction_quit;    }

To    if (cmdstr) {        retval = write_block(&rs->rotport, cmdstr, strlen(cmdstr));        if (retval != RIG_OK)            goto transaction_quit;        // seems the gs232b does not give CR on all commands so we ask for it        if (!data) write_block(&rs->rotport, EOM, strlen(EOM));    }
de Mike W9MDB

 
 

 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2019-08-08:
Boy that didn't format worth a hoot.  Let's try again

    if (cmdstr) {
        retval = write_block(&rs->rotport, cmdstr, strlen(cmdstr));
        if (retval != RIG_OK)
            goto transaction_quit;
    }

To
    if (cmdstr) {
        retval = write_block(&rs->rotport, cmdstr, strlen(cmdstr));
        if (retval != RIG_OK)
            goto transaction_quit;
        // seems the gs232b does not give CR on all commands so we ask for it
        if (!data) write_block(&rs->rotport, EOM, strlen(EOM));
    }

de Mike W9MDB

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

rbw42, 2019-08-09:
Hi Mike,

No, that didn't work. Sending a extra EOM after the command string just makes the controller return ">?", which is then interpreted as an error.

```
rwayth@mypi:~/Hamlib/tests $ ./rotctl -m 603 -r /dev/ttyUSB0 -vvvvv
rotctl, Hamlib 4.0~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
initrots4_gs232a called
rot_register (601)
rot_register (602)
rot_register (603)
rot_register (604)
rot_register (605)
rot_register (606)
rot_register (607)
rot_register (608)
rot_open called
serial_open called
serial_setup called
serial_flush called
Opened rot model 603, 'GS-232B'
rig_strstatus called
Backend version: 0.4, Status: Beta

Rotator command: p
rot_get_position called
gs232b_rot_get_position called
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.             
read_string called
read_string(): RX 15 characters
0000    41 5a 3d 31 38 30 20 20 45 4c 3d 31 30 31 0d        AZ=180  EL=101. 
gs232b_rot_get_position: (az, el) = (180.0, 101.0)
Azimuth: 180.000000
Elevation: 101.000000

Rotator command: P 180 100
Azimuth: Elevation: rot_set_position called
gs232b_rot_set_position called: 180.000000 100.000000
serial_flush called
write_block called
write_block(): TX 9 bytes
0000    57 31 38 30 20 31 30 30 0d                          W180 100.       
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
read_string called
read_string(): RX 3 characters
0000    3f 3e 0d                                            ?>.             
': '?>_transaction: Error for 'W180 100
set_pos: error = Protocol error

Rotator command: M 2 25
Direction: Speed: rot_move called
gs232b_rot_move called 2 25
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    58 31 0d                                            X1.             
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
read_string called
read_string(): RX 3 characters
0000    3f 3e 0d                                            ?>.             
': '?>_transaction: Error for 'X1
move: error = Protocol error

Rotator command: p
rot_get_position called
gs232b_rot_get_position called
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.             
read_string called
read_string(): RX 15 characters
0000    41 5a 3d 31 38 30 20 20 45 4c 3d 31 30 30 0d        AZ=180  EL=100. 
gs232b_rot_get_position: (az, el) = (180.0, 100.0)
Azimuth: 180.000000
Elevation: 100.000000

Rotator command: M 2 25
Direction: Speed: rot_move called
gs232b_rot_move called 2 25
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    58 31 0d                                            X1.             
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
read_string called
read_string(): RX 3 characters
0000    3f 3e 0d                                            ?>.             
': '?>_transaction: Error for 'X1
move: error = Protocol error

Rotator command: fscanf: Success
fscanf: parsing '
' with '%c'
rot_close called
ser_close called
ser_close: restoring options
rot_cleanup called
```

I'm going to stick with my solution for now, which is just to not expect any response.

Cheers,
Randall

N0NB, 2019-08-09:
OK...so try the attached.  Now that we know what to expect we can check for it.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer
  
/*
 *  Hamlib Rotator backend - GS-232B
 *  Copyright (c) 2001-2012 by Stephane Fillod
 *                 (c) 2010 by Kobus Botha
 *
 *
 *   This library is free software; you can redistribute it and/or
 *   modify it under the terms of the GNU Lesser General Public
 *   License as published by the Free Software Foundation; either
 *   version 2.1 of the License, or (at your option) any later version.
 *
 *   This library is distributed in the hope that it will be useful,
 *   but WITHOUT ANY WARRANTY; without even the implied warranty of
 *   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 *   Lesser General Public License for more details.
 *
 *   You should have received a copy of the GNU Lesser General Public
 *   License along with this library; if not, write to the Free Software
 *   Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA  02110-1301  USA
 *
 */

#ifdef HAVE_CONFIG_H
#include "config.h"
#endif

#include <stdio.h>
#include <stdlib.h>
#include <string.h>  /* String function definitions */
#include <unistd.h>  /* UNIX standard function definitions */
#include <math.h>

#include "hamlib/rotator.h"
#include "serial.h"
#include "misc.h"
#include "register.h"

#include "gs232a.h"

#define EOM "\r"
#define REPLY_EOM "\r\n"

#define BUFSZ 64

/**
 * gs232b_transaction
 *
 * cmdstr - Command to be sent to the rig.
 * data - Buffer for reply string.  Can be NULL, indicating that no reply is
 *        is needed, but answer will still be read.
 * data_len - in: Size of buffer. It is the caller's responsibily to provide
 *            a large enough buffer for all possible replies for a command.
 *
 * returns:
 *   RIG_OK  -  if no error occured.
 *   RIG_EIO  -  if an I/O error occured while sending/receiving data.
 *   RIG_ETIMEOUT  -  if timeout expires without any characters received.
 *   RIG_REJECTED  -  if a negative acknowledge was received or command not
 *                    recognized by rig.
 */
static int
gs232b_transaction (ROT *rot, const char *cmdstr,
[... 185 lines cut ...]
  .rot_type =       ROT_TYPE_OTHER,
  .port_type =      RIG_PORT_SERIAL,
  .serial_rate_min =   1200,
  .serial_rate_max =   9600,
  .serial_data_bits =  8,
  .serial_stop_bits =  1,
  .serial_parity =  RIG_PARITY_NONE,
  .serial_handshake =  RIG_HANDSHAKE_NONE,
  .write_delay =  0,
  .post_write_delay =  0,
  .timeout =  400,
  .retry =  3,

  .min_az = 	-180.0,
  .max_az =  	450.0,	/* vary according to rotator type */
  .min_el = 	0.0,
  .max_el =  	180.0, /* requires G-5400B, G-5600B, G-5500, or G-500/G-550 */

  .get_position =  gs232b_rot_get_position,
  .set_position =  gs232b_rot_set_position,
  .stop = 	       gs232b_rot_stop,
  .move =          gs232b_rot_move,
};

/* end of file */

rbw42, 2019-08-11:
Hi Guys,

OK, that patch works, example output:
```
rwayth@mypi:~/Hamlib/tests $ ./rotctl -m 603 -r /dev/ttyUSB0 -vvvvv
rotctl, Hamlib 4.0~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
initrots4_gs232a called
rot_register (601)
rot_register (602)
rot_register (603)
rot_register (604)
rot_register (605)
rot_register (606)
rot_register (607)
rot_register (608)
rot_open called
serial_open called
serial_setup called
serial_flush called
Opened rot model 603, 'GS-232B'
rig_strstatus called
Backend version: 0.4, Status: Beta

Rotator command: M 4 25
Direction: Speed: rot_move called
gs232b_rot_move called 4 25
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    58 31 0d                                            X1.             
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
read_string called
read_string(): RX 3 characters
0000    3f 3e 0d                                            ?>.             
serial_flush called
write_block called
write_block(): TX 2 bytes
0000    44 0d                                               D.              
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
read_string called
read_string(): RX 3 characters
0000    3f 3e 0d                                            ?>.             

Rotator command: p
rot_get_position called
gs232b_rot_get_position called
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.             
read_string called
read_string(): RX 15 characters
0000    41 5a 3d 31 38 30 20 20 45 4c 3d 30 38 34 0d        AZ=180  EL=084. 
gs232b_rot_get_position: (az, el) = (180.0, 84.0)
Azimuth: 180.000000
Elevation: 84.000000

Rotator command: p
rot_get_position called
gs232b_rot_get_position called
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.             
read_string called
read_string(): RX 15 characters
0000    41 5a 3d 31 38 30 20 20 45 4c 3d 30 38 31 0d        AZ=180  EL=081. 
gs232b_rot_get_position: (az, el) = (180.0, 81.0)
Azimuth: 180.000000
Elevation: 81.000000

Rotator command: S
rot_stop called
gs232b_rot_stop called
serial_flush called
write_block called
write_block(): TX 2 bytes
0000    53 0d                                               S.              
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
read_string called
read_string(): RX 3 characters
0000    3f 3e 0d                                            ?>.             

Rotator command: p
rot_get_position called
gs232b_rot_get_position called
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.             
read_string called
read_string(): RX 15 characters
0000    41 5a 3d 31 38 30 20 20 45 4c 3d 30 37 37 0d        AZ=180  EL=077. 
gs232b_rot_get_position: (az, el) = (180.0, 77.0)
Azimuth: 180.000000
Elevation: 77.000000

Rotator command: P 180 100
Azimuth: Elevation: rot_set_position called
gs232b_rot_set_position called: 180.000000 100.000000
serial_flush called
write_block called
write_block(): TX 9 bytes
0000    57 31 38 30 20 31 30 30 0d                          W180 100.       
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
read_string called
read_string(): RX 3 characters
0000    3f 3e 0d                                            ?>.             

Rotator command: p
rot_get_position called
gs232b_rot_get_position called
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.             
read_string called
read_string(): RX 15 characters
0000    41 5a 3d 31 38 30 20 20 45 4c 3d 30 38 32 0d        AZ=180  EL=082. 
gs232b_rot_get_position: (az, el) = (180.0, 82.0)
Azimuth: 180.000000
Elevation: 82.000000

Rotator command: p
rot_get_position called
gs232b_rot_get_position called
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.             
read_string called
read_string(): RX 15 characters
0000    41 5a 3d 31 38 30 20 20 45 4c 3d 30 38 35 0d        AZ=180  EL=085. 
gs232b_rot_get_position: (az, el) = (180.0, 85.0)
Azimuth: 180.000000
Elevation: 85.000000

Rotator command: fscanf: Success
fscanf: parsing '
' with '%c'
rot_close called
ser_close called
ser_close: restoring options
rot_cleanup called
```

Note however, that this doesn't actually confirm that the command was successful any more than just not expecting a reply. If you're worried about whether the comms is working or not, why not set up an internal timer that polls the position of the rotator every second or so if there are no commands in progress or waiting to be processed?

Cheers,
Randall

jwcutler, 2019-10-07:
This fix is working with my GS232B but not fully.  Azimuth is changing, but not elevation.  I can manually control elevation by connecting to the serial port directly and typing in a command.  But rotctl cannot.  Any ideas or suggestions?  Output is below.  Thanks.

Rotator command: p
rot_get_position called
gs232b_rot_get_position called
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.             
read_string called
read_string(): RX 15 characters
0000    41 5a 3d 30 30 30 20 20 45 4c 3d 30 31 30 0d        AZ=000  EL=010. 
gs232b_rot_get_position: (az, el) = (0.0, 10.0)
Azimuth: 0.000000
Elevation: 10.000000

Rotator command: P 20 20
Azimuth: Elevation: rot_set_position called
gs232b_rot_set_position called: 20.000000 20.000000
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 9 bytes
0000    57 30 32 30 20 30 32 30 0d                          W020 020.       
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
read_string called
read_string(): RX 3 characters
0000    3f 3e 0d                                            ?>.             

Rotator command: p
rot_get_position called
gs232b_rot_get_position called
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.             
read_string called
read_string(): RX 15 characters
0000    41 5a 3d 30 32 30 20 20 45 4c 3d 30 31 30 0d        AZ=020  EL=010. 
gs232b_rot_get_position: (az, el) = (20.0, 10.0)
Azimuth: 20.000000
Elevation: 10.000000

N0NB, 2019-10-07:
That looks like the correct command for the GS232B.
Have you tried any other rotator software like PSTRotator?
Have you tried doing a direct serial connection and seeing if you can type the command?
I'm wondering if your elevator is working at all.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

jwcutler, 2019-10-08:
Thanks for the feedback.  Turns out it was timing.  I updated the configuration to include a delay (100-150ms).  Now both az and el are working properly.  So far.

N0NB, 2019-10-08:
Can you send a log now with that timing that works?  Sounds like we need to add that to the backend.
And when you say "configuration" do you mean hamlib or some other file?
de Mike W9MDB 

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

jwcutler, 2019-10-08:
The log is identical.  The exact same bytes are being sent.  They are just sent slower.  Here is a diff of the change I made.  I left my string change as well in the command sent to the GS232B.  I wasn't sure how the EOM in the original was being processed.  At some point, I can return that to the original to see if it still works.

diff --git a/gs232a/gs232b.c b/gs232a/gs232b.c
index 761f2b68..afc4fe32 100644
--- a/gs232a/gs232b.c
+++ b/gs232a/gs232b.c
@@ -147,7 +147,7 @@ gs232b_rot_set_position(ROT *rot, azimuth_t az, elevation_t el)
     u_az = (unsigned)rint(az);
     u_el = (unsigned)rint(el);
 
-    sprintf(cmdstr, "W%03u %03u" EOM, u_az, u_el);
+    sprintf(cmdstr, "W%03u %03u\r", u_az, u_el );
     retval = gs232b_transaction(rot, cmdstr, NULL, 0);
 
     if (retval != RIG_OK) {
@@ -268,7 +268,7 @@ const struct rot_caps gs232b_rot_caps = {
   .serial_stop_bits =  1,
   .serial_parity =  RIG_PARITY_NONE,
   .serial_handshake =  RIG_HANDSHAKE_NONE,
-  .write_delay =  0,
+  .write_delay =  150,
   .post_write_delay =  0,
   .timeout =  400,
   .retry =  3,

N0NB, 2019-10-08:
You need to add the -Z switch to see timing info.
write_delay=150 puts a 150ms delay on every character being written which seems a bit excessive although I guess we're not in much of a hurry here.
It sounds like it may be overrunning the serial buffer
You can adjust the write_delay from the command line and see what value works.
rotctl --set-conf=write_delay=50
Try smaller values and see if they work too.
Once you find a value that works I typically put in a 2X safety factor.
There is this note in iofunc.c * The write_delay is for Yaesu type rigs..require 5 character * sequence to be sent with 50-200msec between each char.

de Mike W9MDB

 
 

   
   
   

   
   

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

jwcutler, 2019-10-08:
Perfect, thanks for the info on the additional flags.  That Yaesu comment is what led me to change the timing.  Glad it was there!  Thanks.

andykv59, 2020-04-24:
> Hi all,
>
> Ok, this is my last post with an update and suggested fix. All of the commands that are "write only" on the GS232B don't return anything. I have also confirmed this by looking at the log of [...]

Thanks a lot, had the same issue and used your solution, it works now!
de UT9UF