Hamlib/Hamlib #125: rigmem on IC-746Pro (IC-7400) doesn't work

ra1nb0w, 2019-08-26:
I am trying to use rigmem to save/change the memory of IC-746Pro on macOS 10.14. The radio is marked as stable in documentation therefore this function should work.

I am using the latest commit on master and the following OS
_macOS 10.14.6 18G87
Xcode 10.3 10G8_

Command
```
./tests/rigmem -m 346 -r /dev/cu.usbserial-14120 -a save test.csv -vvvvvv -c 0x66
```

- [rigmem_log.txt](https://github.com/Hamlib/Hamlib/files/3542301/rigmem_log.txt) contains the log of the above command
- [mem_read.txt](https://github.com/Hamlib/Hamlib/files/3542303/mem_read.txt) contains a binary serial dump of a program that works correctly reading all memory from the radio

At requests I can do more tests or test patches.

Thanks for the great work that you have done in these years!

mdblack98, 2019-08-26:
Looks like you have ci-v echo turned on?
That's why the frame length error I think.
de Mike W9MDB

ra1nb0w, 2019-08-26:
I saw but for what I know the echo is part of the ci-v specification as a form of "detect collision"; see [ci-v specification](http://www.plicht.de/ekki/civ/civ-p3.html) for more information.

mdblack98, 2020-01-16:
What's the status of this?  Working OK?
de Mike W9MDB

ra1nb0w, 2020-01-16:
Tested with the latest commit `c9c32947` and error is still the same.

mdblack98, 2020-01-17:
Think I have a fix for this....
Can you please contact me offline at mdblack98@yahoo.com?
Are you on Windows?
de Mike W9MDB

closed 2020-01-23 by mdblack98