Hamlib/Hamlib #129: Yaesu FT920 control problem with WSJT-X

W9RM, 2019-09-26:
Yaesu FT920 does not respond correctly when controlled by a combo of N1MM+ and WSJT-X.  WSJT-X uses HAMLIB data for radio control.

When using MSK144 mode of WSJT-X and operating in SPLIT mode (CQing on the default MSK144 frequency and listening elsewhere)  the FT920 switches frequency properly only once and then remains on the receive frequency.  This is not new - I first noticed this behavior when WSJT was in V1.9.

Also, when 'running' stations on FT8, radio will randomly switch to some 2 MHz frequency. and crash WSJT-X.

I will supply any further data as needed - tell me what you need.

Jay W9RM

N0NB, 2019-09-26:
How do you have N1MM+ and WSJT-X both talking to your rig?
de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

W9RM, 2019-09-26:
N1MM+ has been configured to work (fairly) seamlessly with WSJT-X.
See:  https://n1mmwp.hamdocs.com/manual-windows/wsjt-x-decode-list-window/

W9RM

Keith J Morehouse
Managing Partner
Calmesa Partners G.P.
Olathe, CO

N0NB, 2019-09-26:
OK...there have been quite a few reports of problems with the DXLab Commander interface.Don't know if they've been ironed out yet.
Bill could provide some more insight on that.
de Mike W9MDB

 

—

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2020-01-16:
Hey Keith,
What's the status of this?
Still having problems?

de Mike W9MDB

W9RM, 2020-01-16:
Mike, I haven't used WSJT-X extensively over the last few months, but I did
try a run of CQing "split" on MSK144 during the Geminid meteor shower.

The behavior of WSJT-X, controlled by DXLabs Commander was what I would
consider normal. It was not faulting the way it was earlier this year and
the FT920 seemed to be controlled the way it should have been.

I was using the latest releases of both WSJT-X and DXLabs.

-W9RM

Keith Morehouse
via MotoG

N0NB, 2020-01-16:
Closing per Mike's request.

closed 2020-01-16 by N0NB