Hamlib/Hamlib #131: Yaesu FTDX101D

RicoDG5BQ, 2019-10-09:
Hello,
I would love to see support for Yaesus new FTDX11D.

73

N0NB, 2019-10-09:
Does the FT-100 work?
de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

RicoDG5BQ, 2019-10-09:
Hi Nate,

I didn’t try the FT-100 
But of course I made a mistake. It should read FTDX101D. 
73

mdblack98, 2019-10-10:
Were you able to test the FT-100?  If that works I'll make an entry for the DX101.
de Mike W9MDB

RicoDG5BQ, 2019-10-10:
Hi Mike,

I tried the settings for the FT-991. They seem to work just fine. Even though the FTDX101D has two separate transceivers at least switching frequencies and so on on the primary side does work. 

73 de Rico

RicoDG5BQ, 2019-10-10:
Hi again,

for further explanation: I need the settings for my logbook which is cqrlog on my raspberry pi. So the changes that have to be done on the rig are only basic. 
If you’d like I could try the settings for the FT 100 as well later this evening. Just let me know. 

73 de Rico, DG5BQ

RicoDG5BQ, 2019-10-11:
Hello Mike,

I’ve tried the FT-100 now. It doesn’t work at all.
The only issue I have with the setting for the FT991 on cqrlog is that I cannot switch between the 2 VFO‘s. I only can work and control on the main VFO. Maybe I could configure the sub VFO as the second radio. But I don’t know how. 
On flrig I can switch between the two VFO’s. But in cqrlog I only have the option to control a second radio. 

73 de Rico

RicoDG5BQ, 2019-10-11:
Hi Mike,

maybe this file helps:

73 de Rico, DG5BQ

mdblack98, 2019-10-11:
I've submitted a patch for the FT-101 so should be in the next release of WSJT-X.
de Mike W9MDB

N0NB, 2019-10-11:
Are you able to compile?  If so try my fork
https://github.com/mdblack98/Hamlib

de Mike W9MDB

 

—

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

RicoDG5BQ, 2019-10-11:
Hi Mike,

Unfortunately I’m not able to compile. 

73 de Rico

N0NB, 2019-10-11:
Contact me off list and I may be able to send you a new rigctld to test.What OS are you running rigctld on?
mdblack98@yahoo.com
de Mike W9MDB

 

—

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

RicoDG5BQ, 2019-10-11:
Hi Mike,
That would be absolutely perfect. I just wir an email to the address you gave me. 
Thank you very much. To be honest: the support you give to me is far more than I expected to get. So thank you so much for your help. 

73 de Rico, DG5BQ

mdblack98, 2020-01-18:
Fixed

closed 2020-01-18 by mdblack98