Hamlib/Hamlib #132: Icom PCR-100 default output volume too low and windows help

magellan-13016, 2019-10-09:
Hello,  I am very involved in many educational projects with the
International Space Station.

In these projects, I need to use "rigctld" to control my receiver and
correct Doppler effects.

I use the ICOM PCR-100 model with GPredict software. However, with this
receiver, when you take control of it with HAMLib, the volume of the
output sound is too low.

To control the receiver, I use this command on linux :

rigctld -v -m 402 -r /dev/ttyUSB0 -s 38400 -t 4532 -C stop_bits=1

And a kind of this one on Windows :

"C:Program Files\HAMLib\rigctld.exe" -v -m 402 -r /dev/ttyUSB0 -s 38400
-t 4532 -C stop_bits=1

To increase the output volume on Linux, the only way I've found is to
open a second terminal window and to use this command to set up the volume to the maximum :

echo 'L AF 1' | nc localhost 4532

But I didn't find any way to do the same on Windows.

So here are my questions : How can we take control of the receiver with
the output volume set to the maximum without having to open a second terminal window and using another command line ? If there is no way to do that, how can I use the same command on Windows to set the volume at its maximum ?

Thanks for the answer.

Best regards, David.

N0NB, 2019-10-09:
> Hamlib-developer mailing list
> Hamlib-developer@lists.sourceforge.net

David

You can get netcat (nc) for windows.  https://nmap.org/ncat/

Then use the same script both O.S.’s

Tom

magellan-13016, 2019-10-09:
Many thanks Tom but your link point to nmap.org and it turns out that ncat is now integrated to Nmap software... There is two links : Latest stable release self-installer: nmap-7.80-setup.exe and 
Latest Npcap release self-installer: npcap-0.9983.exe. I don't know which of them I should choose. Could you tell me which is the good one or give me a direct link to the program I should use ?

Many thanks, David.

N0NB, 2019-10-09:
David

If the rigctl bit that Bill suggested (better solution as only one program) does not do it - or you just want a copy of nc - guess google best option - there are a pile of sites, some source only some with binaries. Depending on your view on internet downloads you can take your pick.

I downloaded it years ago as used nc a lot on Unix and have carried the binaries for windows only - I can’t remember the original  site.

Tom

mdblack98, 2020-01-18:
Solution found.

closed 2020-01-18 by mdblack98