Hamlib/Hamlib #136: hamlib error with WSJT-X and FT-897

G0ODR, 2019-10-18:
cannot use WSJT software due to "hamlib error switching vfo's" message with FT-897. can anyone suggest a fix for this? i have justed tried to install version 3.3 for win64 but get errors trying to install .dll files. i cannot find any Hamlib folder/files on the PC to even look at them. Please any suggestions would be gratefully welcomed otherwise there will be a radio on sale soon.

very frustrated,
Colin G0ODR

N0NB, 2019-10-18:
> Hamlib-developer mailing list
> Hamlib-developer@lists.sourceforge.net

Colin.

Make sure the radio is not in any memory mode.  I don't have one of
those radios, but there have been lots of messages elsewhere indicating
that if they are in a memory mode (tunable memories?) what you see happens.

73.

Dave G8KBV

mdblack98, 2019-10-18:
There was a bug in the ft897 backend where the cache had to be forced to time out.Not sure that's the problem here but it might be.  I believe the old behavior was that it would get the wrong freq/mode after setting That patch will be in the next release.
de Mike W9MDB

G0ODR, 2019-10-18:
Definitely not in memory mode as as you say it is mentioned else wherethanks anyway

Colin G0ODR

N0NB, 2019-10-18:
WSJT-X bundles a static version of the hamlib library.

G0ODR, 2019-12-23:
Sorry to drag this one back up but still no joy in using FT897. Can I check version/load/replace the Hamlib file in the software with the latest release? I have searched for file name and no results found. Where is it stored, in the code itself or as a file somewhere?  I can only dream of the amount of knowledge you guys have in computer programming and control so please be gentle with this “user”. Thanks for all people like you do for the like of mere mortals.
Colin G0ODR

N0NB, 2019-12-23:
What exactly is the problem?
Can you run rigctl and give some debug output?  I'll be glad to fix whatever is wrong.  I've been working hard on fixing up outstanding issues.
https://github.com/mdblack98/Hamlib

We'll be doing it on my fork so test this one and post the results...we can start off with get freq/mode/ptt

rigctl -m 123 -r com1 -s 9600  f m tYou should output like this145000000FM150000
If you don't get this or get an error re-run with debug and send me the debug.txt file.
rigctl -m 123 -r com1 -s 9600  f m t >debug.txt 2>&1

de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

G0ODR, 2019-12-24:
Sorry Nate, 
Forgot to answer first question about what is the issue. Answer is as per first post regarding WSJT-X software and the post error message. Equipment being used is a ft897, microham microkeyer 2 and their urouter software.

Merry Christmas 
73s
Colin G0ODR

G0ODR, 2020-01-06:
Hi Nate,
Did you get my email with the attachment showing output from the rigctl command?

Colin

N0NB, 2020-01-06:
No
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-01-07:
Colin, the messages may appear to come from me due to an email forwarder I use to copy these posts to the hamlib mailing list as GitHub doesn't allow me to subscribe the mailing list to GitHub mail.  The replies have been coming from Mike, W9MDB, and not me.  And no, I did not receive a direct email from you as near as I can tell.  Please just post the output here.

G0ODR, 2020-01-10:
Looks like I need some more instruction on how to run rigctl. I have used the rigctl-wsjtx.exe application to open the cod window. I have typed in exactly as above and it gives me several lines returned which looks like the settings in the radio but not what you are expecting. Thinking that I need to change a few thinks to suit my set up I have tried varies combination but still get the same result. Sorry to be a pain but could you please point me in the correct direction of using the rigctl function to give you the information you require.

Colin G0ODR

N0NB, 2020-01-10:
> i have read the help file and some other stuff and attached is the output i get. if i just us the f m t commands i do get some output more like you are looking for.
>
> it looks like it is taking the -m 123 command to get all the data as 1 dump_caps, 2 power2mW and 3 dump_conf instead of looking for radio type 123 ie ft897, if thats what it should be doing???

The "rig command:" indicates you are already running rigctl and resending the command line to rigctl itself.
Ensure you are at the C:\ prompt when you run the command.
de Mike W9MDB

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-01-10:
Plus...try run "cmd" to open the command window...not rigctl.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

G0ODR, 2020-01-10:
thanks Mike but still no joy. below are the two returns from the cmd window opened from the run window.
C:\>rigctl -m 123 -r com1 -s 9600  f m t
'rigctl' is not recognized as an internal or external command,
operable program or batch file.

C:\>rigctl -m 123 -r com1 -s 9600  f m t >debug.txt 2>&1
A required privilege is not held by the client.

Should i be further into the directory path. my radio software is on another internal drive would that make a difference?

Colin G0ODR

N0NB, 2020-01-10:
You will have to be in the same directory as rigctl or rigctl-wsjtx
Or you have to give complete path to rigctl/rigctl-wsjtx
If you are running WSJTX's installation should be here
cd \WSJT\WSJTX\binrigctl-wsjtx -m 123 -r com1 -s 9600 f m t
Or perhaps you have another installation of hamlib?
Mike

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

G0ODR, 2020-01-10:
Ok getting something a bit more meaningful, possibly.
this what i get now but questions. the -com1 command is that the port number used? I have tried changing this to my port number but no difference. also the -s 9600 is that the baud rate?

E:\RadioProgs\wsjtx\bin>rigctl-wsjtx -m 123 -r com1 -s 9600 f m t
get_freq: error = Communication timed out
get_mode: error = Communication timed out
get_ptt: error = Communication timed out

just tried the debug option and it returns nothing.
de Colin G0ODR

mdblack98, 2020-01-10:
Yes....
-r is your rig's com port-s is the rig's baud rate
And when you say you tried the debug you're not showing what you tried...but this work now assuming you get correct COM port and baud rate.
rigctl-wsjtx -m 123 -r com1 -s 9600 -Z -vvvvv f m t

de Mike W9MDB

G0ODR, 2020-01-10:
below are the responses.

E:\RadioProgs\wsjtx\bin>rigctl-wsjtx -m 123 -r com9 -s 4800  f m t
get_freq: error = Communication timed out
get_mode: error = Communication timed out
get_ptt: error = Communication timed out
E:\RadioProgs\wsjtx\bin>

E:\RadioProgs\wsjtx\bin>rigctl-wsjtx -m 123 -r com9 -s 4800  f m t >debug.txt 2>&1
E:\RadioProgs\wsjtx\bin>

and the debug text files shows:
get_freq: error = Communication timed out
get_mode: error = Communication timed out
get_ptt: error = Communication timed out

de G0ODR

G0ODR, 2020-01-10:
Also just tried it with the added commands you put ind:-

E:\RadioProgs\wsjtx\bin>rigctl-wsjtx -m 123 -r com9 -s 4800  -Z -vvvvv f m t
2020-01-10:22:43:46.564370: rigctl, Hamlib 4.0~git
2020-01-10:22:43:46.564370: Report bugs to <hamlib-developer@lists.sourceforge.net>

2020-01-10:22:43:46.564370: rig_init called
2020-01-10:22:43:46.564370: yaesu: initrigs4_yaesu called
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (121)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (127)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (110)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (105)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (106)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (107)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (139)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (109)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (120)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (101)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (122)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (123)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.564370: rig_register: rig_register (111)
2020-01-10:22:43:46.564370: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (115)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (113)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (114)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (128)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (131)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (116)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (103)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (124)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (104)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (125)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (129)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (137)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (132)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (130)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (117)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (119)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (118)
2020-01-10:22:43:46.579991: rig_register called
2020-01-10:22:43:46.579991: rig_register: rig_register (126)
[... 45 lines cut ...]
2020-01-10:22:43:46.955012: write_block(): TX 5 bytes
2020-01-10:22:43:46.955012: 0000    00 78 00 00 bb                                      .x...
2020-01-10:22:43:46.955012: read_block called
2020-01-10:22:43:47.267362: read_block(): Timed out 0.312350 seconds after 0 chars
2020-01-10:22:43:47.267362: rig_passband_normal called
get_mode: error = Communication timed out
2020-01-10:22:43:47.267362: rig_get_ptt called
2020-01-10:22:43:47.267362: ft897: cache invalid
2020-01-10:22:43:47.267362: serial_flush called
2020-01-10:22:43:47.267362: serial_flush: tcflush
2020-01-10:22:43:47.267362: write_block called
2020-01-10:22:43:47.267362: write_block(): TX 5 bytes
2020-01-10:22:43:47.267362: 0000    00 00 00 00 f7                                      .....
2020-01-10:22:43:47.267362: read_block called
2020-01-10:22:43:47.579873: read_block(): Timed out 0.312511 seconds after 0 chars
get_ptt: error = Communication timed out
2020-01-10:22:43:47.579873: rig_close called
2020-01-10:22:43:47.579873: ft897:ft897_close called
2020-01-10:22:43:47.579873: port_close called
2020-01-10:22:43:47.579873: ser_close called
2020-01-10:22:43:47.579873: ser_close: restoring options
2020-01-10:22:43:47.595495: rig_cleanup called
2020-01-10:22:43:47.595495: ft897:ft897_cleanup called

E:\RadioProgs\wsjtx\bin>

N0NB, 2020-01-10:
The timeout indicates wrong com port or wrong baud rate.
What kind of cable do you have hooked?
And menu 19 is your baud rate which defaults to 4800 but you may have changed it.
de MIke W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

G0ODR, 2020-01-10:
Hi Mike,
Baud rate is 4800. Microhams U router software communicates ok with the radio and it says that the port settings are 4800 bps 8N2. Not sure where the 2 stop bits get set from as i cannot set or change it. The cable from the microham microkeyer 2 to the radio is their own and a usb from PC to Microham.
would it make any difference with teh port being a virtual port??

de Colin G0ODR

G0ODR, 2020-01-10:
just been playing and opened up Logger 32. the radio will change frequence when a DX spot is clicked so the comms seems to be OK to Logger32. Could that be an issue??

de Colin G0ODR

N0NB, 2020-01-10:
Are you trying to split the com port?  Yes...that makes a difference and usually doesn't work.
Mike

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

G0ODR, 2020-01-10:
Not sure what you mean by splitting the com port and does that mean I can’t use the WSJt software with the Microham unit? There isn’t the option for a second cat port in the U router software or I can set up a completely different set of options in the u router software but it will always be a virtual port I guess. Any ideas on how to make it work. It would be nice to know if anyone else has this setup working.

de Colin G0ODR

mdblack98, 2020-01-10:
There are only certain programs that allow you to share a com port across multiple applications.
FLRigrigctld (along with rigctlcom)OmniRig
What all do you want to run?
I would recommend you start with FLRig and hook up WSJT-X to FLRig.
Then we'll hook up rigctld  (multiple program can talk to it if they know how)Then we'll hook up rigctlcom (for generic COM port access like Logger32 probably wants)Then we'll hook up Logger32 to rigctlcom
You got anything else that wants rig access?
de Mike W9MDB

N0NB, 2020-01-10:
I will post a youtube video of setting up WSJT-X, FLRig, and Logger32 by tomorrow both talking at the same time.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

G0ODR, 2020-01-10:
That all sounds interesting. WhenI use the WSJT-X software on HF I call it from Logger32 and it cuts out the comms from Logger 32 to the radio and takes direct control, as I understand it. Is the issue the virtual prot as the flex SDR I use for HF runs virtual ports but through a VSP manager?
OK on th evideo i will look out for that and look into downloading the FLrig software.
 Thanks for your help this evening it is much appreciated. 
73s
de Colin G0ODR

G0ODR, 2020-01-11:
Mike,
To answer your question I guess the best way is to tell you my setup. I have a FT897 which I want to use on VHF for FT8 via the Microham Microkeyer 2. I use Logger 32 as my logging software. I know that I can use the MMTTY software embedded in Logger32 for RTTY and PSK modes but I have been told these are cut down version of the full software so was considering using the full versions instead and getting that to talk to Logger 32. Also I may want to get N1MM into the mix as well.
Thanks again for all your time,

de Colin G0ODR

G0ODR, 2020-01-19:
All,
I have been able to get the setup, FT897, Microham Microkeyer 2, logger32 and WSJT-X, working on 2meters using the flrig software. Did not see a video though Mike.

Thanks for all your efforts Mike really appreciated.

73s from. Happy Colin

GratingLobe, 2020-04-20:
Hi,
I have the same issue as Colin had getting hamlib error with WSJT-X, but using IC-7000 transceiver and a CAT/Serial interface.
The transceiver and the interface are working fine with all the digital mode programs that I use, including DigitalMaster/Ham Radio Deluxe.
Flrig software also works, but this doesn't help for working with WSJT-X program. 
I have another radio IC-7300 which works fine with WSJT-X, but this radio has internal interface directly on USB.
Please let me know if have any idea how to solve this issue.
thanks,

mdblack98, 2020-04-20:
Contact me off list and we can hook up to work on the problem.
mdblack98@yahoo.com
de Mike W9MDB

mdblack98, 2020-04-20:
> Hi Mike,
> Many thanks for your replay, but after few days of struggling I managed to fix the issue using OmniRig program with information provided by the link below:
>
> https://dxham.de/?p=2364
>
> The link is in German, but is no problem to use Google Translate.
> There are many transceivers that suffer from this issue. I don't know why the designers of WSJT-X program don't try to fix this problem.
> Thanks again,

Because I need debug information to fix it.
Please contact me directly so I can send you some information on helping to debug this.
mdblack98@yahoo.com
de Mike W9MDB

closed 2020-09-05 by mdblack98