Hamlib/Hamlib #138: TS-590 - split frequency reportting via cat not working

oh1haq, 2019-10-23:
Hi! 

Noticed that the hamlib with ts-590 is not reporting the split frequency via cat when transmitting. If you press TF-SET on the rig, the frequency is reported correctly. 

Omnirig and dx commander are reporting the frequency correctly when transmitting, so maybe it's just a simple definition file issue. Hamlib is used in RF-KIT ampliler for cat following, so the split is not reported for the atu etc. 

Thanks in advance if someone has time to take a look.

mdblack98, 2019-10-23:
What software are you running hamlib with and what version of hamlib are you using?
Can you provide a debug log to help us see what's going on?
de Mike W9MDB

oh1haq, 2019-10-23:
Hi! 

Thank you for reply. The hamlib is used in an rf-kit amplifier running rasperry pi. I can access the terminal, so what would me the easiest way to see the version information/debug info. Sorry for being almost complete linux newbie.

N0NB, 2019-10-23:
Not sure of your setup but there are two ways to get debug info.
#1 If you can set up the software to use rigctld is best because then you see what the software is asking for.#2 Using rigctl directly which doesn't require the application.
Let's start with rigctl -- this should work....TS-590S I assume and not TS-590SG?  If SG use -m 237 instead

rigctl -m 231 -r /dev/xxxx -s 9600 -vvvvvAdjust COM port and baud rate to match your setup.
Using the following command this is what I'd expect to see.
V VFOAF 14074000fFrequency: 14074000V VFOBF 14075000fFrequency: 14075000T 1fFrequency: 14075000V VFOAfFrequency: 14074000T 0
You can also run rigctl (and rigctld) requiring the vfo be provided and again you'll get whatever VFO you ask for.rigctl -m 231 -r /dev/xxxx -s 9600 -vvvvv
F VFOA 14074000F VFOB 14075000
f VFOAFrequency: 14074000f VFOB
Frequency: 14075000

In other words it depends on which VFO you ask for as to what frequency you get.Some other software may assume you want the TX VFO when asking for generic frequency...hamlib does not make that assumption.  You get what you ask for regardless of PTT status
Mike

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2019-10-23:
Not sure of your setup but there are two ways to get debug info.
#1 If you can set up the software to use rigctld is best because then you see what the software is asking for.#2 Using rigctl directly which doesn't require the application.
Let's start with rigctl -- this should work....TS-590S I assume and not TS-590SG?  If SG use -m 237 instead

rigctl -m 231 -r /dev/xxxx -s 9600 -vvvvvAdjust COM port and baud rate to match your setup.
Using the following command this is what I'd expect to see.
V VFOAF 14074000fFrequency: 14074000V VFOBF 14075000fFrequency: 14075000T 1fFrequency: 14075000V VFOAfFrequency: 14074000T 0
You can also run rigctl (and rigctld) requiring the vfo be provided and again you'll get whatever VFO you ask for.rigctl -m 231 -r /dev/xxxx -s 9600 -vvvvv
F VFOA 14074000F VFOB 14075000
f VFOAFrequency: 14074000f VFOB
Frequency: 14075000

In other words it depends on which VFO you ask for as to what frequency you get.Some other software may assume you want the TX VFO when asking for generic frequency...hamlib does not make that assumption.  You get what you ask for regardless of PTT status
Mike

mdblack98, 2019-10-23:
Not sure of your setup but there are two ways to get debug info.

#1 If you can set up the software to use rigctld is best because then you see what the software is asking for.
#2 Using rigctl directly which doesn't require the application.

Let's start with rigctl -- this should work....TS-590S I assume and not TS-590SG?  If SG use -m 237 instead

rigctl -m 231 -r /dev/xxxx -s 9600 -vvvvv
Adjust COM port and baud rate to match your setup.

Using the following command this is what I'd expect to see.

V VFOA
F 14074000
f
Frequency: 14074000
V VFOB
F 14075000
f
Frequency: 14075000
T 1
f
Frequency: 14075000
V VFOA
f
Frequency: 14074000
T 0

You can also run rigctl (and rigctld) requiring the vfo be provided and again you'll get whatever VFO you ask for.
rigctl -m 231 -r /dev/xxxx -s 9600 -vvvvv

F VFOA 14074000
F VFOB 14075000
f VFOA
Frequency: 14074000
f VFOB
Frequency: 14075000

In other words it depends on which VFO you ask for as to what frequency you get.
Some other software may assume you want the TX VFO when asking for generic frequency...hamlib does not make that assumption.  You get what you ask for regardless of PTT status

Mike
Boy that really formatted poorly...let's try again

oh1haq, 2019-10-23:
This is what I got:

rigctl, Hamlib 3.2~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
initrigs3_kenwood called
rig_register called
rig_register: rig_register (213)
rig_register called
rig_register: rig_register (201)
rig_register called
rig_register: rig_register (225)
rig_register called
rig_register: rig_register (203)
rig_register called
rig_register: rig_register (204)
rig_register called
rig_register: rig_register (216)
rig_register called
rig_register: rig_register (224)
rig_register called
rig_register: rig_register (205)
rig_register called
rig_register: rig_register (207)
rig_register called
rig_register: rig_register (209)
rig_register called
rig_register: rig_register (210)
rig_register called
rig_register: rig_register (222)
rig_register called
rig_register: rig_register (214)
rig_register called
rig_register: rig_register (230)
rig_register called
rig_register: rig_register (221)
rig_register called
rig_register: rig_register (229)
rig_register called
rig_register: rig_register (238)
rig_register called
rig_register: rig_register (202)
rig_register called
rig_register: rig_register (211)
rig_register called
rig_register: rig_register (206)
rig_register called
rig_register: rig_register (208)
rig_register called
rig_register: rig_register (215)
rig_register called
rig_register: rig_register (226)
rig_register called
rig_register: rig_register (217)
rig_register called
rig_register: rig_register (233)
rig_register called
rig_register: rig_register (220)
rig_register called
rig_register: rig_register (223)
rig_register called
rig_register: rig_register (227)
rig_register called
rig_register: rig_register (234)
rig_register called
rig_register: rig_register (231)
rig_register called
rig_register: rig_register (239)
rig_register called
rig_register: rig_register (237)
rig_register called
rig_register: rig_register (228)
rig_register called
rig_register: rig_register (219)
rig_register called
rig_register: rig_register (232)
rig_register called
rig_register: rig_register (236)
rig_register called
rig_register: rig_register (240)
kenwood_init called
kenwood_init: if_len = 37
rig_open called
port_open called
serial_open called
serial_setup called
kenwood_open called
kenwood_transaction called
kenwood_transaction: cmdstr = FV
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    46 56 3b                                            FV;             
read_string called
read_string(): RX 7 characters
0000    46 56 32 2e 30 34 3b                                FV2.04;         
kenwood_open: found f/w version 2.04
kenwood_get_id called
kenwood_transaction called
kenwood_transaction: cmdstr = ID
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string called
read_string(): RX 6 characters
0000    49 44 30 32 31 3b                                   ID021;          
kenwood_open: found match 021
kenwood_get_trn called
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = AI
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    41 49 3b                                            AI;             
read_string called
read_string(): RX 4 characters
0000    41 49 30 3b                                         AI0;            
kenwood_set_trn called
kenwood_transaction called
kenwood_transaction: cmdstr = AI0
serial_flush called
write_block called
write_block(): TX 4 bytes
0000    41 49 30 3b                                         AI0;            
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string called
read_string(): RX 6 characters
0000    49 44 30 32 31 3b                                   ID021;          
rig_get_vfo called
kenwood_get_vfo_if called
kenwood_get_if called
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = IF
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    49 46 3b                                            IF;             
read_string called
read_string(): RX 38 characters
0000    49 46 30 30 30 31 34 31 38 35 30 31 30 20 20 20     IF00014185010   
0010    20 20 20 30 30 30 30 30 30 30 30 35 30 32 30 30        0000000050200
0020    30 30 30 38 30 3b                                   00080;          
Opened rig model 231, 'TS-590S'
rig_strstatus called
Backend version: 0.9.1, Status: Beta

oh1haq, 2019-10-23:
Some observations:
f VFOB does not give me vfo-b frequency. This is what I get:

----------------------
Rig command: f VFOB
Frequency: 14188310

Rig command: VFO: set_vfo: error = Invalid parameter

--------------------------

I can read the frequency with simple f, but this returns vfo a frequency 14188310.

oh1haq, 2019-10-24:
Just got an e-mail from the amp support. I take the bug is in the ts-590. 
Maybe hamlib could somehow workaround this?

"Hi. 
Hamlib is requesting the TX frequency, but Transceiver to just give the main VFO.
If Split is the TX, than you'll get the wrong frequency.
It's a known bug.
Workaround is to use the main VFO as TX and VFO/B as receive.
DX Commander is looking at both VFO's and check the TX flag, so they do have the right TX frequency.
N1MM is also doing it and is working fine."

mdblack98, 2019-10-24:
Still waiting to see the result of using the --vfo switch
With the rig in split mode try these commands
F VFOA 14074000F VFOB 14075000f VFOAFrequency: 14074000f VFOB
Frequency: 14075000T 1
f VFOAFrequency: 14074000f VFOB
Frequency: 14075000

I expect to see the frequencies returned correctly for "f VFOA and "f VFOB"hamlib makes no presumption about which VFO you're asking for which may work for some particular software programs but hamlib is generic.If the requesting software asks for VFOA it gets VFOA and asking for VFOB gives VFOB results.
Mike

oh1haq, 2019-10-24:
Sorry Mike, this is kind of a learning by doing. 

If I give the  F VFOA 14074000 command after the connection init I get the Frequency: set_freq: error = Invalid parameter -error. 

If I select the VFO A using the command you wrote in you previous message, right after I command V VFOA, the rig jumps out of split and only vfo-a is displayed.  Is this as designed?

Other than that I can change the vfo a and b freqs by selecting the correct vfo by V and entering a frequency by F

N0NB, 2019-10-24:
Please show the command line you are using.  You apparently don't have the --vfo switch enabled.
Mike

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oh1haq, 2019-10-24:
The init command is: rigctl -o -m 231 -r /dev/ttyUSB0 -s 9600 
As soon as I give V VFOA -command and hit enter the split is turned off on the rig.

mdblack98, 2019-10-24:
You don't need to use the "V" command in vfo mode.
F VFOA14074000F VFOB 14075000
f VFOAFrequency: 14074000f VFOBFrequency: 14075000

N0NB, 2019-10-24:
This seems to be formatting incorrectly when I receive this through my email so this is in text mode now.

F VFOA14074000
F VFOB 14075000
f VFOA
Frequency: 14074000
f VFOB
Frequency: 14075000

Mike

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oh1haq, 2019-10-24:
Mike, thanks! 

Now working as intended so no V needed, 

Now to transmit. without -o option I can toggle the PTT with T 1. 
But with -o the T 1 seems to want the VFO info. I tried with T 1 VFOB, but it did not like it.

mdblack98, 2019-10-24:
You can set the split vfo.
S VFOA 1 VFOBThen transmit will work no matter which VFO you useT VFOA 1orT VFOB 1
The VFO is always the 1st thing after the command

N0NB, 2019-10-24:
Dang formatting again....

You can set the split vfo.
S VFOA 1 VFOB
Then transmit will work no matter which VFO you use
T VFOA 1
or
T VFOB 1

The VFOA/B is always the 1st thing after the command

Mike

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2019-10-24:
I should add for clarity that the "T" command VFO is not actually used.  But the vfo option mandates VFO on all commands anyways.

 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oh1haq, 2019-10-25:
After 

S VFOA 1 VFOB

This is what I get. If I try T VFOB 1 the split disables.

Rig command: T VFOA 1
VFO: PTT: 
Rig command: f VFOA
VFO: Frequency: 14074000

Rig command: f VFOB
VFO: Frequency: 14075000

Rig command: T VFOA 0

mdblack98, 2019-10-25:
I thought the T VFOB 1 would still work but perhaps not.

You are saying that "T VFOA 1" is working for you, correct?

de Mike W9MDB

oh1haq, 2019-10-25:
Yes, "T VFOA 1"  is working. But now the million dollar question is that is there a bug in 590 or somethere else so that the transmit frequency of the rig in split mode is not correctly transmitted to the amp, even though the amp is asking the transmit frequency.

mdblack98, 2019-10-25:
It seems you have confirmed that "f VFOB" correctly returns VFOB.

I don't quite understand what software is asking for what in your setup.    

There is no command in any rig I know of the "asks for transmit freq".  The client needs to know how it set up the split mode and ask for the appropriate freq from the VFO that it wants.  I can see where some software packages may make the assumption you want the Tx freq but that doesn't work for hamlib since it doesn't know what the client software is doing.  It's up to the client to ask for what it want.s

There's no logical reason to automatically return the TX freq when asking for the current frequency since both frequencies could be asked for.

Mike

N0NB, 2019-10-25:
Can you provide any debug on what is going on with the amplifier and what it's asking for?

Mike

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oh1haq, 2019-10-25:
Yes, the f VFOB returns the vfo-b frequency.  But it's the "original" vfo-b frequency. If you press ptt or use tf-set in split mode, the vfo-b moves to the place of vfo-a on the rig's frequency display. 

But what puzzles me is that when I press the TF-set button on the rig, the amp gets the "transmit" (vfo-b) frequency.  The tf-set function performs an indential function vfo-wise than pressing the PTT. 

There is some debug information on the amp, but I don't think there is any help in this case. 

The amp needs a frequency information via cat so that the solid state amp can automatically select the correct band and correct sub-band for ATU. There is a problem with ATU segments if the split is very large and in this condition amp only gets the VFO-a frequency when transmitting and not the transmit frequency (vfo-b)

mdblack98, 2019-10-25:
I need to know how the amp is requesting the frequency to see if there's a fix we can do.

oh1haq, 2019-10-25:
rgr, I'll contact the amp support.

N0NB, 2019-10-25:
> Am 25.10.2019 um 18:13 schrieb oh1haq <notifications@github.com>:
> 
> Yes, the f VFOB returns the vfo-b frequency. But it's the "original" vfo-b frequency. If you press ptt or use tf-set in split mode, the vfo-b moves to the place of vfo-a on the rig's frequency display.
> 

Yes, but it is still the „VFO B“ frequency, isn’t it?

For me it seems that the algorithm is as follows

if (split) then
   TX freq is VFO_B freq
else
   TX freq is VFO_A free
endif

so one can also regard this as an error „downstream“ to assume that
TX freq == VFO_A freq

Yours

DL1YCF Christoph

mdblack98, 2019-10-25:
So then this should return if that is true.
T 0F VFOA 14075000F VFOB 14076000S VFOA 1 VFOB
f VFOAf VFOBT 1f VFOAf VFOB
After the "T 1" do you see the VFOA/B frequencies reverse?
If so we can code an exception for the TS-590 on this.  I wonder if any other rigs are like this.
Mike

 
 

Yes, but it is still the „VFO B“ frequency, isn’t it?

For me it seems that the algorithm is as follows

if (split) then
 TX freq is VFO_B freq
else
 TX freq is VFO_A free
endif

so one can also regard this as an error „downstream“ to assume that
TX freq == VFO_A freq

Yours

DL1YCF Christoph

mdblack98, 2019-10-25:
Once again with text formatting now

So then this should return if that is true.
T 0F VFOA 14075000F VFOB 14076000S VFOA 1 VFOB
f VFOAf VFOBT 1f VFOAf VFOB
After the "T 1" do you see the VFOA/B frequencies reverse?
If so we can code an exception for the TS-590 on this.  I wonder if any other rigs are like this.
Mike

 

    On Friday, October 25, 2019, 05:25:55 PM CDT, Black Michael <mdblack98@yahoo.com> wrote:  
 
 So then this should return if that is true.
T 0F VFOA 14075000F VFOB 14076000S VFOA 1 VFOB
f VFOAf VFOBT 1f VFOAf VFOB
After the "T 1" do you see the VFOA/B frequencies reverse?
If so we can code an exception for the TS-590 on this.  I wonder if any other rigs are like this.
Mike

 
 

Yes, but it is still the „VFO B“ frequency, isn’t it?

For me it seems that the algorithm is as follows

if (split) then
 TX freq is VFO_B freq
else
 TX freq is VFO_A free
endif

so one can also regard this as an error „downstream“ to assume that
TX freq == VFO_A freq

Yours

DL1YCF Christoph

mdblack98, 2019-10-25:
T 0
F VFOA 14075000
F VFOB 14076000
S VFOA 1 VFOB
f VFOA
f VFOB
T 1
f VFOA
f VFOB

After the "T 1" do you see the VFOA/B frequencies reverse?

If so we can code an exception for the TS-590 on this.  I wonder if any other rigs are like this.

oh1haq, 2019-10-26:
Here are the results, so no, they are not reversed, 

Rig command: F VFOA 14075000
Rig command:  F VFOB 14076000
Rig command: S VFOA 1 VFOB
VFO: Split: TX VFO: 
Rig command: f VFOA
VFO: Frequency: 14075000
Rig command: f VFOB
VFO: Frequency: 14076000
Rig command: T VFOA 1
VFO: PTT: 
Rig command: f VFOA
VFO: Frequency: 14075000
Rig command:  f VFOB
Rig command: VFO: Frequency: 14076000

oh1haq, 2019-10-26:
I also made a test with tf-set button pressed on the rig. (amp now gets the correct frequency info) When doing f VFOA and f VFOB the frequencies are not reversed.

N0NB, 2019-10-26:
Dear Michael,
I am involved in the programming of piHPSDR, an SDR program
which has a CAT interface meant to simulate the TS-2000.

In this case, a request „FA;“ always returns the frequency of VFO_A,
and „FB;“ returns the frequency of VFO_B. This seems to be quite logical,
but of course, when TXing, the TX frequency may be VFOA/VFOB depending on
split/no_split.

It is only my $0.02, but I find the assumption a frequency request for
VFO_A should always return the TX frequency even if it comes from VFO_B
not more logical.

But since this is about hamlib rather than the TS590:
=====================================================

However, I could run a test on my TS-590SG if there is still ambiguity,
but I would rather code a small test program directly using hamlib,
e.g. coding

rig_get_freq(rig, RIG_VFO_CURR, &freq1);
rig_get_freq(rig, RIG_VFO_RX, &freq2);
rig_get_freq(rig, RIG_VFO_TX, &freq3);
rig_get_freq(rig, RIG_VFO_MAIN, &freq4);
rig_get_freq(rig, RIG_VFO_SUB, &freq5);
rig_get_freq(rig, RIG_VFO_A, &freq6);
rig_get_freq(rig, RIG_VFO_B, &freq7);

and do this in a loop and reporting the frequencies
while playing with the rig. In my opinion, we should strictly
keep the discussion to the hamlib API rather than to the capabilities of some
downstream application **including rigctl**.

If I understand well, for amplifier control anything but using RIG_VFO_TX
is just an error in using the API rather an error in hamlib.

Should I run such a test (I think the original poster should do), I could also
report by activating debug output which CAT commands are actually sent to the
rig.

Yours,

Christoph DL1YCF.

mdblack98, 2019-10-26:
Now add these three frequency requests between every command so we can see how they change.  Start with the rig NOT in split mode since we're setting split explicitly.

f currVFO
f RX
f TX

Mike

oh1haq, 2019-10-26:
Before I go any further, f RX works, but f  TX returns invalid parameter.

mdblack98, 2019-10-26:
Can you please show a screen shot including your starting command?

oh1haq, 2019-10-26:
Just tested quickly like this:

rigctl -o -m 231 -r /dev/ttyUSB0 -s 9600 

Rig command: f RX
VFO: Frequency: 14342150

Rig command: f TX
VFO: get_freq: error = Invalid parameter

Rig command:

mdblack98, 2019-10-26:
That all depends on whether the rig has full dual VFOs.  We've got a generic interface for Kenwoods and some have dual, some not.So without putting in exceptions for all rigs the proper API method, as you've pointed out, is to use the proper A/B/RX/TX as the client wants.Can't assume VFOA contains the TX frequencyI'm wondering if this really isn't a display problem since there's only one frequency display.  Does the display actually show VFOA when transmitting in split?  I don't see a VFO indicator in the 850 manual on the display.The manual does mention "Dual Digital VFOs".
Mike

oh1haq, 2019-10-26:
>  I don't see a VFO indicator in the 850

You mean 590?

Here is a short video how the split works. https://youtu.be/9_uYHwAQcQ0?t=43

Wondering what is done differently when using tf-set vs. transmitting.

mdblack98, 2019-10-26:
Forgot to show you need to add the VFO to those commands when using -o

f VFOB TX
f VFOA RX

I think there is a small bug as this doesn't work the way I think it should.

I think you should be able to do this:
f VFOA TX
f VFOA RX
f VFOB TX
f VFOB RX
The vfo argument required by -o should not affect the TX/RX VFO selection.
I'll take a look at this one.

Rig command: S VFOA 1 VFOB

Rig command: F VFOA 14000000

Rig command: F VFOB 15000000

Rig command: f VFOB TX
Frequency: 15000000

Rig command: f VFOA RX
Frequency: 14000000

N0NB, 2019-10-26:
Yes...I had an 850 manual open and was looking at the wrong thin.The 590 manual doesn't mention dual VFOs.
Mike

 
 

 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oh1haq, 2019-10-26:
rigctl -o -m 231 -r /dev/ttyUSB0 -s 9600 

Rig command: S VFOA 1 VFOB
VFO: Split: TX VFO: 
Rig command: F VFOA 14000000
VFO: Frequency: 
Rig command: F VFOB 15000000
VFO: Frequency: 
Rig command: f VFOB TX
VFO: Frequency: 15000000
Rig command: VFO: PTT: ^Z
[6]+  Stopped                 rigctl -o -m 231 -r /dev/ttyUSB0 -s 9600
pi@RF-KIT-PA:~ $ rigctl -o -m 231 -r /dev/ttyUSB0 -s 9600 
Rig command: f VFOA RX
VFO: Frequency: 14000000
Rig command: VFO: Rptr Shift: ^Z

As you see I needed to crtl-z out in order to f VFOA RX to work, and after that the prompt is looking for repeater shift...

oh1haq, 2019-10-26:
And to my knowledge ts-590 has dual vfo's (separare vfo a and b)

mdblack98, 2019-10-26:
mdblack98/Hamlib  
|  
|   
|   
|   |    |

   |

  |
|  
|   |  
mdblack98/Hamlib
 
Ham radio control library. Contribute to mdblack98/Hamlib development by creating an account on GitHub.
  |   |

  |

  |

  Can you compile the latest Hamlib?Use the one from my fork in case we patch this thing.

oh1haq, 2019-10-26:
Sadly I've never compiled before as being a linux newbie. And I'm a little bit worried not to break my the linear software.  :) 

But the rig has dual cat (linear is connected to the com port and pc is in usb) so I could do tests with my windows machine connected to the rig, if you could do a windows version also.

oh1haq, 2019-10-26:
Tested with  Hamlib 3.3 (win) Getting the same error. 

C:\Program Files (x86)\hamlib-w32-3.3\bin>rigctl -o -m 231 -r COM3 -s 115200

Rig command: f
VFO: VFOA
Frequency: 14200000

Rig command: f VFOB TX
VFO: Frequency: 14200000

Rig command: Command ' ' not found!

Rig command: VFO: PTT:

mdblack98, 2020-01-16:
Following up...is it behaving for you now using my fork?
https://github.com/mdblack98/Hamlib
de Mike W9MDB

mdblack98, 2020-03-11:
No response received on this so closing this issue for now...assuming it's fixed.

closed 2020-03-11 by mdblack98