Hamlib/Hamlib #141: new release

ra1nb0w, 2019-11-14:
why don't release a new version?
thanks

N0NB, 2019-11-14:
First of all, there is still work being done by Mike.  Also, I seem to recall there might be some questions about the new API/ABI.

Second, I've been quite busy in real life and have been a bit slow in keeping up.  Perhaps it's time for me to move on and let someone else do this?

ra1nb0w, 2019-11-14:
Was only a simple question from a maintainer (macports), nothing more :)
Come from user questions that from time to time ask me about support for this and that and I say that that change is in master branch a not release exist.
Maybe, we can wait an answer from Mike.

mdblack98, 2020-01-18:
Not a bug so closing

closed 2020-01-18 by mdblack98