Hamlib/Hamlib #144: Filter width setting doesn't work with data mode on Icom

arodland, 2019-11-20:
`icom_set_mode_with_data` calls `icom_set_mode`, passing the filter width through, and then does a data mode (`1A 06`, except on IC-7200) call to set data mode to 0 or 1 depending on whether the mode was a `PKT` one.

The problem is that (at least on the 7300 and 7610, and probably on most newer rigs), the filter width passed to the "set mode" command only affects the *non-data* filter width, and to change the data filter width, one must pass a filter width to the `1A 06` command. Since hamlib omits this parameter, the width is never changed for packet modes.

It seems easy enough to drag a call to `rig2icom_mode`/`r2i_mode` into `icom_set_mode_with_data` to get the proper width and then pass it along to the datamode command, but perhaps a refactor is in order — should there be one function which outputs `icmode`/`icmode_ext`/`datamode`/`datamode_ext` and the whole sequence keyed off of those?

Also, to research: are there rigs that support 1A 06, but where it doesn't have a second byte for filter width?

mdblack98, 2019-11-20:
I'm working on several Icom things right now so I'll add this to my list.  Might be a while before I get a fix in though as my queue is getting pretty long here.
de Mike W9MDB

arodland, 2019-11-20:
Thanks! At least in the process I figured out what magic string I have to send to the radio to fix it on my own :)

mdblack98, 2020-01-05:
Try my fork...with a bit o'luck it will set the mode correct for the 7300 and 7610 now.
I have to research all the other ICOM rigs to see who else needs this.
de Mike W9MDB

arodland, 2020-01-08:
Nope, that didn't work quite right on my 7300:

```
Rig command: m
Mode: PKTUSB
Passband: 3600

Rig command: M PKTUSB 500
set_mode: error = Invalid parameter

Rig command: m
Mode: USB
Passband: 2200

Rig command: M PKTUSB 3600
set_mode: error = Invalid parameter

Rig command: m
Mode: USB
Passband: 3600

Rig command: M USB 3000
set_mode: error = Command rejected by the rig
```

I was able to fix it, though; see https://github.com/arodland/Hamlib/commit/2584e74ca721814cae81f391861d770a11e40f6b which works for me.

arodland, 2020-01-08:
Okay, one more problem... `rig2icom_mode` can set `pd` to `-1`. I'm locally converting that to `2`, but this is starting to seem like a bit of a pile of hacks, so I don't want to commit it :)

N0NB, 2020-01-08:
Hmmm...PKTUSB is not one of the icom modes.....and 0x02 is AM mode....
So rig2icommode is not what we want here....
Try my fork again....
https://github.com/mdblack98/Hamlib

de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

arodland, 2020-01-08:
Haven't tested it but that doesn't look like it can be right... `width2` is a `pbwidth_t`, which shouldn't go in an icom command. My version does work.

N0NB, 2020-01-08:
pbwidth_t is the correct type to make the C compiler happy.  It's the same as shortfreq_t which is how all the PASSBAND settings are defined.

It gets treated in the code as just a single byte here so is the same effect as using a char.typedef signed long shortfreq_t;
typedef shortfreq_t pbwidth_t;

#define s_Hz(f)     ((shortfreq_t)(f))#define s_kHz(f)    ((shortfreq_t)((f)*(shortfreq_t)1000))#define s_MHz(f)    ((shortfreq_t)((f)*(shortfreq_t)1000000))#define s_GHz(f)    ((shortfreq_t)((f)*(shortfreq_t)1000000000))
#define s_Hz(f)     ((shortfreq_t)(f))#define s_kHz(f)    ((shortfreq_t)((f)*(shortfreq_t)1000))#define s_MHz(f)    ((shortfreq_t)((f)*(shortfreq_t)1000000))#define s_GHz(f)    ((shortfreq_t)((f)*(shortfreq_t)1000000000))
 #define RIG_PASSBAND_NORMAL     s_Hz(0)#define RIG_PASSBAND_NOCHANGE   s_Hz(-1)
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2020-01-11:
I got the filter setting working correctly now.  It does only do filter 1 and 3 and not filter two.  Do we need filter  two (i.e. Medium)?  Not sure we do.  Wide and Narrow seem plenty.
That's already built-in to the rig2icom_mode function.  
de Mike W9MDB

mdblack98, 2020-01-18:
Fixed

closed 2020-01-18 by mdblack98