Hamlib/Hamlib #146: 7851 icom_get_freq fails

roelandjansen, 2019-11-24:
together with @mdblack98 did some reasearch why the 'f' command fails on the hash rev 74718f9b6b06332250d23b704c60e879fc159674.

7851
==================================
get vfo
get_vfo: error = Feature not available
set Main
get freq
get_freq: error = Command rejected by the rig
set Sub
get freq
get_freq: error = Command rejected by the rig

And with the proposed and tested PR I made, the function works as supposed. It also does not break the 7300 and 9700 code.

7851
==================================
get vfo
get_vfo: error = Feature not available
set Main
get freq
14074000
set Sub
get freq
15273000

I

N0NB, 2019-11-24:
and a reference to the patch link: 
https://github.com/Hamlib/Hamlib/pull/147

N0NB, 2019-11-25:
Is this going to be removed permanently or revisited in the future?  If it can be revisited, I'd rather see it just temporarily taken out of compilation with CPP directives #IF 0 ... #ENDIF when possible.

roelandjansen, 2019-11-25:
Mike stated:

"Just remove that whole switch statement...it's bad."

Roeland

mdblack98, 2019-11-25:
It's permanent....doesn't do what we want.  I thought it was the ability to read freq but alas....no....Icom still hasn't given the ability to give current vfo or ask for frequency by any specific Main/Sub VFOA/VFOB ability.
I'm working on a few more mods to icom.c for TX/RX VFO queries and Main/Sub changes for the newer SATMODE rigs.
Mike

N0NB, 2019-11-25:
Okay, fair enough.  Thanks for the replies!

N0NB, 2019-11-25:
Applied and pushed.  Thanks, Roeland.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

PR #147 (2019-11-25): removed the code that breaks the 'f' function (icm Mike)
this fixes #146 by removing code that inadvertedly misassigns variables that were OK in the first place  and removes unreachable code as well, incuding a double break statement.
closed 2019-11-25 by N0NB