Hamlib/Hamlib #157: TS-590S get_level segmentation fault (Hamlib 4.0)

hlnurse, 2019-12-12:
Hi Folks,

I am testing the latest beta code and have run into a problem with the get_level function on the TS-590S.  This problem does not exist on version 3.3.  Hamlib was built on a Raspberry Pi Buster. A segmentation fault results for STRENGTH and SQL (at least).

Here is a debug dump:

pi@rigpi4:~ $ rigctl -m 231 -r /dev/ttyUSB0 -s 115200 -vvvvv
rigctl, Hamlib 4.0~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
initrigs4_kenwood called
rig_register called
rig_register: rig_register (213)
rig_register called
rig_register: rig_register (201)
rig_register called
rig_register: rig_register (225)
rig_register called
rig_register: rig_register (203)
rig_register called
rig_register: rig_register (204)
rig_register called
rig_register: rig_register (216)
rig_register called
rig_register: rig_register (224)
rig_register called
rig_register: rig_register (205)
rig_register called
rig_register: rig_register (207)
rig_register called
rig_register: rig_register (209)
rig_register called
rig_register: rig_register (210)
rig_register called
rig_register: rig_register (222)
rig_register called
rig_register: rig_register (214)
rig_register called
rig_register: rig_register (230)
rig_register called
rig_register: rig_register (221)
rig_register called
rig_register: rig_register (229)
rig_register called
rig_register: rig_register (243)
rig_register called
rig_register: rig_register (244)
rig_register called
rig_register: rig_register (245)
rig_register called
rig_register: rig_register (238)
rig_register called
rig_register: rig_register (202)
rig_register called
rig_register: rig_register (211)
rig_register called
rig_register: rig_register (206)
rig_register called
rig_register: rig_register (208)
rig_register called
rig_register: rig_register (215)
rig_register called
rig_register: rig_register (226)
rig_register called
rig_register: rig_register (217)
rig_register called
rig_register: rig_register (233)
rig_register called
rig_register: rig_register (242)
rig_register called
rig_register: rig_register (220)
rig_register called
rig_register: rig_register (223)
rig_register called
rig_register: rig_register (227)
rig_register called
rig_register: rig_register (234)
rig_register called
rig_register: rig_register (231)
rig_register called
rig_register: rig_register (239)
rig_register called
rig_register: rig_register (237)
rig_register called
rig_register: rig_register (228)
rig_register called
rig_register: rig_register (219)
rig_register called
rig_register: rig_register (232)
rig_register called
rig_register: rig_register (236)
rig_register called
rig_register: rig_register (240)
rig_register called
rig_register: rig_register (241)
kenwood_init called, version 1.3
kenwood_init: if_len = 37
rig_open called
port_open called
serial_open called
serial_setup called
serial_flush called
serial_flush: tcflush
kenwood_open called
kenwood_transaction called
kenwood_transaction: cmdstr = FV
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 3 bytes
0000    46 56 3b                                            FV;             
read_string called
read_string(): RX 7 characters
0000    46 56 32 2e 30 34 3b                                FV2.04;         
kenwood_open: found f/w version 2.04
kenwood_get_id called
kenwood_transaction called
kenwood_transaction: cmdstr = ID
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string called
read_string(): RX 6 characters
0000    49 44 30 32 31 3b                                   ID021;          
kenwood_open: found match 021
kenwood_get_trn called
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = AI
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 3 bytes
0000    41 49 3b                                            AI;             
read_string called
read_string(): RX 4 characters
0000    41 49 30 3b                                         AI0;            
kenwood_set_trn called
kenwood_transaction called
kenwood_transaction: cmdstr = AI0
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 4 bytes
0000    41 49 30 3b                                         AI0;            
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string called
read_string(): RX 6 characters
0000    49 44 30 32 31 3b                                   ID021;          
kenwood_get_split_vfo_if called
kenwood_get_if called
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = IF
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 3 bytes
0000    49 46 3b                                            IF;             
read_string called
read_string(): RX 38 characters
0000    49 46 30 30 30 30 37 31 39 30 39 39 30 20 20 20     IF00007190990   
0010    20 20 20 30 30 30 30 30 30 30 30 32 30 31 30 30        0000000020100
0020    30 30 30 38 30 3b                                   00080;          
rig_strvfo called
rig_strvfo called
kenwood_get_split_vfo_if: priv->tx_vfo=VFOA
rig_strvfo called
rig_strvfo called
kenwood_open: priv->tx_vfo=VFOA
rig_get_vfo called
kenwood_get_vfo_if called
kenwood_get_if called
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = IF
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 3 bytes
0000    49 46 3b                                            IF;             
read_string called
read_string(): RX 38 characters
0000    49 46 30 30 30 30 37 31 39 30 39 39 30 20 20 20     IF00007190990   
0010    20 20 20 30 30 30 30 30 30 30 30 32 30 31 30 30        0000000020100
0020    30 30 30 38 30 3b                                   00080;          
rig_strvfo called
rig_strvfo called
kenwood_get_vfo_if: priv->tx_vfo=VFOA
Opened rig model 231, 'TS-590S'
rig_strstatus called
rig_strstatus called
Backend version: 1.3.2, Status: Stable
Rig command: \get_level
Level: STRENGTH
rig_parse_level called
rig_has_get_level called
rig_get_level called
rig_has_get_level called
ts590_get_level called
rig_get_level called
rig_has_get_level called

.....several hundred more repetitions....

Segmentation fault
pi@rigpi4:~ $ 

Thanks for your help!  Looking forward to official release.

Howard W6HN

mdblack98, 2019-12-13:
I can confirm this is a defect.
I checked in a patch which should fix the problem.
Use my fork
https://github.com/mdblack98/Hamlib.git

de Mike W9MDB

hlnurse, 2019-12-13:
Perfect, Mike, thanks!  The get_level function works well.  I found another problem, I'll open a new issue.

--Howard

closed 2019-12-13 by hlnurse