Hamlib/Hamlib #158: -Werror=format-security errors in rotctl_parse.c (in master f1d0340)

barjac, 2019-12-13:
Testing build of current master I hit a format-security error.
This is in Mageia (dev) branch. I could normally patch these errors for printf but since the fix would extend the string then I have no idea how to handle this for snprintf() - I'm not a programmer, just the package maintainer ;)

`rotctl_parse.c:1012:13: error: format not a string literal and no format arguments [-Werror=format-security]
 1012 |             snprintf(cmd_name, sizeof(cmd_name), parsed_input[0] + 1);

Thanks,
Barry

mdblack98, 2019-12-13:
Try my fork where that has been patched.
https://github.com/mdblack98/Hamlib

barjac, 2019-12-14:
Thanks Michael,
Yes that fixes build.
I am confused somewhat about whether to use your fork or upstream with patches from your fork for our distribution sources. Do you merge your changes directly into upstream (if so how often) or do you submit pull requests and wait? ;)
Also since 3.3 release I see many changes in the upstream git log but no incremental point releases, despite the lib major change to 4.
Any clarifications would be appreciated.
Thanks again,
Barry

mdblack98, 2020-01-18:
Fixed

closed 2020-01-18 by mdblack98