Hamlib/Hamlib #159: TS-590 \set_powerstat 1 fails

hlnurse, 2019-12-13:
Using this command with a TS-590S:

rigctl -m 231 -r /dev/ttyUSB0 -s 115200 \set_powerstat 1 

fails with a communications timed out error.  \set_powerstat 0 works fine once the radio is on.  \set_powerstat 1 works when used from 'Rig command:'.

Changing the radio model to TS-950SDX, 213, works for power on with the 590S.  If the 950SDX power on code could be moved to the 950S it should fix the problem.

This is with the Linux build on a Raspberry Pi.  It does not work on earlier versions back to 3.3.

--Howard

N0NB, 2019-12-13:
They both call the same function.
Can you please send a debug log from both including the -Z switch and -vvvvv?
de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

hlnurse, 2019-12-14:
Here is are the results from both.  In the case of the 590S, it looks for firmware versions and fails.  For the 950SDX it tries IF then does a PS!;, but no VF.

See below.

--Howard

RADIO TS-590S, using -m 231 (590S)

pi@rigpi4:~ $ rigctl -m 231 -r /dev/ttyUSB0 -s 115200 -vvvvv -Z \set_powerstat 1
2019-12-14:00:01:48.020508: rigctl, Hamlib 4.0~git
2019-12-14:00:01:48.020693: Report bugs to <hamlib-developer@lists.sourceforge.net>

2019-12-14:00:01:48.020784: rig_init called
2019-12-14:00:01:48.021107: initrigs4_kenwood called
2019-12-14:00:01:48.021182: rig_register called
2019-12-14:00:01:48.021229: rig_register: rig_register (213)
2019-12-14:00:01:48.021278: rig_register called
2019-12-14:00:01:48.021345: rig_register: rig_register (201)
2019-12-14:00:01:48.021386: rig_register called
2019-12-14:00:01:48.021418: rig_register: rig_register (225)
2019-12-14:00:01:48.021466: rig_register called
2019-12-14:00:01:48.021501: rig_register: rig_register (203)
2019-12-14:00:01:48.021544: rig_register called
2019-12-14:00:01:48.021582: rig_register: rig_register (204)
2019-12-14:00:01:48.021622: rig_register called
2019-12-14:00:01:48.021667: rig_register: rig_register (216)
2019-12-14:00:01:48.021706: rig_register called
2019-12-14:00:01:48.021747: rig_register: rig_register (224)
2019-12-14:00:01:48.021795: rig_register called
2019-12-14:00:01:48.021832: rig_register: rig_register (205)
2019-12-14:00:01:48.021871: rig_register called
2019-12-14:00:01:48.021912: rig_register: rig_register (207)
2019-12-14:00:01:48.021943: rig_register called
2019-12-14:00:01:48.021989: rig_register: rig_register (209)
2019-12-14:00:01:48.022102: rig_register called
2019-12-14:00:01:48.022137: rig_register: rig_register (210)
2019-12-14:00:01:48.022176: rig_register called
2019-12-14:00:01:48.022212: rig_register: rig_register (222)
2019-12-14:00:01:48.022250: rig_register called
2019-12-14:00:01:48.022288: rig_register: rig_register (214)
2019-12-14:00:01:48.022325: rig_register called
2019-12-14:00:01:48.022362: rig_register: rig_register (230)
2019-12-14:00:01:48.022402: rig_register called
2019-12-14:00:01:48.022437: rig_register: rig_register (221)
2019-12-14:00:01:48.022476: rig_register called
2019-12-14:00:01:48.022514: rig_register: rig_register (229)
2019-12-14:00:01:48.022553: rig_register called
2019-12-14:00:01:48.022589: rig_register: rig_register (243)
2019-12-14:00:01:48.022632: rig_register called
2019-12-14:00:01:48.022670: rig_register: rig_register (244)
2019-12-14:00:01:48.022707: rig_register called
2019-12-14:00:01:48.022744: rig_register: rig_register (245)
2019-12-14:00:01:48.022790: rig_register called
2019-12-14:00:01:48.022827: rig_register: rig_register (238)
2019-12-14:00:01:48.022866: rig_register called
2019-12-14:00:01:48.022907: rig_register: rig_register (202)
2019-12-14:00:01:48.022938: rig_register called
2019-12-14:00:01:48.022975: rig_register: rig_register (211)
2019-12-14:00:01:48.023015: rig_register called
2019-12-14:00:01:48.023054: rig_register: rig_register (206)
2019-12-14:00:01:48.023092: rig_register called
2019-12-14:00:01:48.023130: rig_register: rig_register (208)
2019-12-14:00:01:48.023169: rig_register called
2019-12-14:00:01:48.023212: rig_register: rig_register (215)
2019-12-14:00:01:48.023252: rig_register called
2019-12-14:00:01:48.023295: rig_register: rig_register (226)
2019-12-14:00:01:48.023333: rig_register called
2019-12-14:00:01:48.023377: rig_register: rig_register (217)
2019-12-14:00:01:48.023418: rig_register called
2019-12-14:00:01:48.023455: rig_register: rig_register (233)
[... 337 lines cut ...]
2019-12-14:00:06:42.741485: rigctl_parse: debug5
2019-12-14:00:06:42.741540: rigctl_parse: debug10
2019-12-14:00:06:42.741667: rig_set_powerstat called
2019-12-14:00:06:42.741730: kenwood_transaction called
2019-12-14:00:06:42.741786: kenwood_transaction: cmdstr = PS1
2019-12-14:00:06:42.741840: serial_flush called
2019-12-14:00:06:42.741896: serial_flush: tcflush
2019-12-14:00:06:42.741943: write_block called
2019-12-14:00:06:42.741988: write_block(): TX 4 bytes
2019-12-14:00:06:42.742015: 0000    50 53 31 3b                                         PS1;            
2019-12-14:00:06:42.742057: write_block called
2019-12-14:00:06:42.742090: write_block(): TX 3 bytes
2019-12-14:00:06:42.742111: 0000    49 44 3b                                            ID;             
2019-12-14:00:06:42.742152: read_string called
2019-12-14:00:06:42.797885: read_string(): RX 6 characters
2019-12-14:00:06:42.797964: 0000    49 44 30 32 31 3b                                   ID021;          
2019-12-14:00:06:42.797998: kenwood_set_powerstat called status=1
2019-12-14:00:06:42.798018: kenwood_set_powerstat: timeout waiting for powerup, try 1
set_powerstat: error = Communication timed out
2019-12-14:00:06:42.798103: rig_close called
2019-12-14:00:06:42.798128: port_close called
2019-12-14:00:06:42.798151: ser_close called
2019-12-14:00:06:42.798178: ser_close: restoring options
2019-12-14:00:06:42.800860: rig_cleanup called
2019-12-14:00:06:42.800975: kenwood_cleanup called

N0NB, 2019-12-14:
Try my fork again.  I made it so that Kenwoods  always execute set_powerstat 1 on startup.I can't think of reason not to do this for all rigs.  Can anybody else?
de Mike W9MDB 

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

hlnurse, 2019-12-14:
So far so good on the fix.  Several other 590S owners will be testing and I'll confirm.

--Howard

hlnurse, 2019-12-20:
All tests for \set_powerstat with the TS-590 are good, thanks for the fix!  I will close this issue.

Howard W6HN

closed 2019-12-20 by hlnurse