Hamlib/Hamlib #161: FT-950 Get Mode SSB reversal

hlnurse, 2019-12-20:
With the FT-950 tuned to 7.125 MHz LSB, rigctl m command returns USB.  The debug log shows MD01; which should be LSB.  Here is the debug log:

Radio is set to 7.251MHz LSB
 
Rig command: m
rigctl_parse: debug1
rigctl_parse: debug5
rigctl_parse: debug10
rig_get_mode called
newcat_get_mode called
newcat_valid_command called
newcat_valid_command MD
rig_strvfo called
rig_strvfo called
newcat_set_vfo_from_alias: alias vfo = currVFO
newcat_get_mode: cmd_str = MD0;
cmd_str = MD0;
write_block called
write_block(): TX 4 bytes
0000    4d 44 30 3b                                         MD0;
read_string called
read_string(): RX 5 characters
0000    4d 44 30 31 3b                                      MD01;
newcat_get_cmd: read count = 5, ret_data = MD01;
rig_passband_normal called
newcat_get_rx_bandwidth called
newcat_valid_command called
newcat_valid_command SH
rig_strvfo called
rig_strvfo called
newcat_set_vfo_from_alias: alias vfo = currVFO
cmd_str = SH0;
write_block called
write_block(): TX 4 bytes
0000    53 48 30 3b                                         SH0;
read_string called
read_string(): RX 6 characters
0000    53 48 30 31 33 3b                                   SH013;
newcat_get_cmd: read count = 6, ret_data = SH013;
Mode: USB
Passband: 2400

Giving the Rig command w \0x4d\0x44\0x30\0x32\0x3b to change to USB, according to the CAT manual, I get:
 
Rig command: m
rigctl_parse: debug1
rigctl_parse: debug5
rigctl_parse: debug10
rig_get_mode called
newcat_get_mode called
newcat_valid_command called
newcat_valid_command MD
rig_strvfo called
rig_strvfo called
newcat_set_vfo_from_alias: alias vfo = currVFO
newcat_get_mode: cmd_str = MD0;
cmd_str = MD0;
write_block called
write_block(): TX 4 bytes
0000    4d 44 30 3b                                         MD0;
read_string called
read_string(): RX 5 characters
0000    4d 44 30 32 3b                                      MD02;
newcat_get_cmd: read count = 5, ret_data = MD02;
rig_passband_normal called
newcat_get_rx_bandwidth called
newcat_valid_command called
newcat_valid_command SH
rig_strvfo called
rig_strvfo called
newcat_set_vfo_from_alias: alias vfo = currVFO
cmd_str = SH0;
write_block called
write_block(): TX 4 bytes
0000    53 48 30 3b                                         SH0;
read_string called
read_string(): RX 6 characters
0000    53 48 30 31 33 3b                                   SH013;
newcat_get_cmd: read count = 6, ret_data = SH013;
Mode: LSB
Passband: 2400

Thanks!

--Howard W6HN

mdblack98, 2019-12-20:
Check my fork for the patch.
https://github.com/mdblack98/Hamlib

It appears this has been broken for quite a while.  The newcat.c set_mode assumes all Yaesu's are 1=LSB, 2=USB so I made get_mode the same.  Looking through some of my manuals I don't see any different behavior.
de Mike W9MDB

hlnurse, 2019-12-24:
This issue (SSB mode reversal on the FT-950) is fixed, thank you!  I'll close the issue.

Howard W6HN

closed 2019-12-24 by hlnurse