Hamlib/Hamlib #162: FT-950 \set_powerstat delay

hlnurse, 2019-12-23:
Using \set_powerstat 1 for the FT-950 takes about 30 seconds for the radio to respond.  The 950 CAT manual states PS1 needs to be preceded by a dummy character, then the PS1 sent between 1 and 2 seconds later.  This delay probably also occurs for the FT-450, FT991A, FTdx-1200, FTdx-3000 and perhaps the FTdx-101D/MP (not tested).

The log for the FT-950 is below.

--Howard W6HN

pi@rigpi4:~ $ rigctl -m 128 -s 38400 -r /dev/ttyUSB0 -vvvvv -Z \set_powerstat 1
2019-12-23:01:43:33.468039: rigctl, Hamlib 4.0~git
2019-12-23:01:43:33.468137: Report bugs to <hamlib-developer@lists.sourceforge.net>
 
2019-12-23:01:43:33.468237: rig_init called
2019-12-23:01:43:33.468448: yaesu: initrigs4_yaesu called
2019-12-23:01:43:33.468546: rig_register called
2019-12-23:01:43:33.468600: rig_register: rig_register (121)
2019-12-23:01:43:33.468641: rig_register called
2019-12-23:01:43:33.468677: rig_register: rig_register (127)
2019-12-23:01:43:33.468711: rig_register called
2019-12-23:01:43:33.468743: rig_register: rig_register (110)
2019-12-23:01:43:33.468773: rig_register called
2019-12-23:01:43:33.468803: rig_register: rig_register (105)
2019-12-23:01:43:33.468835: rig_register called
2019-12-23:01:43:33.468867: rig_register: rig_register (106)
2019-12-23:01:43:33.468897: rig_register called
2019-12-23:01:43:33.468927: rig_register: rig_register (107)
2019-12-23:01:43:33.468963: rig_register called
2019-12-23:01:43:33.468996: rig_register: rig_register (139)
2019-12-23:01:43:33.469027: rig_register called
2019-12-23:01:43:33.469057: rig_register: rig_register (109)
2019-12-23:01:43:33.469087: rig_register called
2019-12-23:01:43:33.469118: rig_register: rig_register (120)
2019-12-23:01:43:33.469148: rig_register called
2019-12-23:01:43:33.469180: rig_register: rig_register (101)
2019-12-23:01:43:33.469210: rig_register called
2019-12-23:01:43:33.469242: rig_register: rig_register (122)
2019-12-23:01:43:33.469276: rig_register called
2019-12-23:01:43:33.469317: rig_register: rig_register (123)
2019-12-23:01:43:33.469353: rig_register called
2019-12-23:01:43:33.469383: rig_register: rig_register (111)
2019-12-23:01:43:33.469413: rig_register called
2019-12-23:01:43:33.469444: rig_register: rig_register (115)
2019-12-23:01:43:33.469474: rig_register called
2019-12-23:01:43:33.469504: rig_register: rig_register (113)
2019-12-23:01:43:33.469537: rig_register called
2019-12-23:01:43:33.469567: rig_register: rig_register (114)
2019-12-23:01:43:33.469598: rig_register called
2019-12-23:01:43:33.469631: rig_register: rig_register (128)
2019-12-23:01:43:33.469661: rig_register called
2019-12-23:01:43:33.469691: rig_register: rig_register (131)
2019-12-23:01:43:33.469721: rig_register called
2019-12-23:01:43:33.469751: rig_register: rig_register (116)
2019-12-23:01:43:33.469781: rig_register called
2019-12-23:01:43:33.469813: rig_register: rig_register (103)
2019-12-23:01:43:33.469847: rig_register called
2019-12-23:01:43:33.469877: rig_register: rig_register (124)
2019-12-23:01:43:33.469910: rig_register called
2019-12-23:01:43:33.469940: rig_register: rig_register (104)
2019-12-23:01:43:33.469974: rig_register called
2019-12-23:01:43:33.470009: rig_register: rig_register (125)
2019-12-23:01:43:33.470040: rig_register called
2019-12-23:01:43:33.470071: rig_register: rig_register (129)
2019-12-23:01:43:33.470105: rig_register called
2019-12-23:01:43:33.470138: rig_register: rig_register (137)
2019-12-23:01:43:33.470169: rig_register called
2019-12-23:01:43:33.470199: rig_register: rig_register (132)
2019-12-23:01:43:33.470230: rig_register called
2019-12-23:01:43:33.470259: rig_register: rig_register (130)
2019-12-23:01:43:33.470289: rig_register called
2019-12-23:01:43:33.470319: rig_register: rig_register (117)
2019-12-23:01:43:33.470352: rig_register called
2019-12-23:01:43:33.470381: rig_register: rig_register (119)
[... 160 lines cut ...]
Opened rig model 128, 'FT-950'
2019-12-23:01:44:06.509714: rig_strstatus called
2019-12-23:01:44:06.509782: rig_strstatus called
2019-12-23:01:44:06.509826: Backend version: 0.24.2, Status: Stable
2019-12-23:01:44:06.509887: rigctl_parse: debug1
2019-12-23:01:44:06.509922: rigctl_parse: debug3
2019-12-23:01:44:06.509955: rigctl_parse: debug5
2019-12-23:01:44:06.509988: rigctl_parse: debug10
2019-12-23:01:44:06.510093: rig_set_powerstat called
2019-12-23:01:44:06.510132: newcat_set_powerstat called
2019-12-23:01:44:06.510166: newcat_valid_command called
2019-12-23:01:44:06.510200: newcat_valid_command PS
2019-12-23:01:44:06.510241: write_block called
2019-12-23:01:44:06.510303: write_block(): TX 4 bytes
2019-12-23:01:44:06.510344: 0000    50 53 31 3b                                         PS1;
2019-12-23:01:44:08.010555: write_block called
2019-12-23:01:44:08.010653: write_block(): TX 4 bytes
2019-12-23:01:44:08.010697: 0000    50 53 31 3b                                         PS1;
2019-12-23:01:44:08.010793: rig_close called
2019-12-23:01:44:08.010826: newcat_close called
2019-12-23:01:44:08.010884: port_close called
2019-12-23:01:44:08.010922: ser_close called
2019-12-23:01:44:08.010992: ser_close: restoring options
2019-12-23:01:44:08.015034: rig_cleanup called
2019-12-23:01:44:08.015095: newcat_cleanup called

N0NB, 2019-12-23:
The problem is all the commands it's trying to do during open.  Try my fork as I've now added powerup during the open before anything else is tried.
https://github.com/mdblack98/Hamlib.git

de Mike W9MDB

 
 

Hamlib-developer mailing list
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

hlnurse, 2019-12-24:
Thanks, Mike.  (I have to test on the TS-950 several thousand miles away!)  The problem is still there.  The PS1; is being sent first as you know, but it apparently takes a dummy character to wake up the 950 and the first real command must come no sooner than 1 and no more than 2 seconds later.  I did get a power on 30 seconds later, but the PS1 is sent there too.

--Howard

N0NB, 2019-12-24:
I need a debug log.  That command should be sent twice and it appears whoever initially set it up though that sending PS1 twice would do it with a 1.5 second interval between them.Perhaps a different or single dummy character is needed?
Mike

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

hlnurse, 2019-12-24:
Mike, it appears the delay is missing in the upfront PS1;’s.

Howard

pi@rigpi4:~ $ rigctl -m 128 -r /dev/ttyUSB0 -s 38400 -vvvvv -Z \set_powerstat 1
2019-12-24:17:51:23.195583: rigctl, Hamlib 4.0~git
2019-12-24:17:51:23.195680: Report bugs to <hamlib-developer@lists.sourceforge.net>

2019-12-24:17:51:23.195903: rig_init called
2019-12-24:17:51:23.196274: yaesu: initrigs4_yaesu called
2019-12-24:17:51:23.196441: rig_register called
2019-12-24:17:51:23.196580: rig_register: rig_register (121)
2019-12-24:17:51:23.196711: rig_register called
2019-12-24:17:51:23.196816: rig_register: rig_register (127)
2019-12-24:17:51:23.196980: rig_register called
2019-12-24:17:51:23.197103: rig_register: rig_register (110)
2019-12-24:17:51:23.197229: rig_register called
2019-12-24:17:51:23.197354: rig_register: rig_register (105)
2019-12-24:17:51:23.197477: rig_register called
2019-12-24:17:51:23.197605: rig_register: rig_register (106)
2019-12-24:17:51:23.197708: rig_register called
2019-12-24:17:51:23.197832: rig_register: rig_register (107)
2019-12-24:17:51:23.197956: rig_register called
2019-12-24:17:51:23.198084: rig_register: rig_register (139)
2019-12-24:17:51:23.198207: rig_register called
2019-12-24:17:51:23.198324: rig_register: rig_register (109)
2019-12-24:17:51:23.198427: rig_register called
2019-12-24:17:51:23.198551: rig_register: rig_register (120)
2019-12-24:17:51:23.198679: rig_register called
2019-12-24:17:51:23.198782: rig_register: rig_register (101)
2019-12-24:17:51:23.198910: rig_register called
2019-12-24:17:51:23.199012: rig_register: rig_register (122)
2019-12-24:17:51:23.199139: rig_register called
2019-12-24:17:51:23.199262: rig_register: rig_register (123)
2019-12-24:17:51:23.199366: rig_register called
2019-12-24:17:51:23.199460: rig_register: rig_register (111)
2019-12-24:17:51:23.199554: rig_register called
2019-12-24:17:51:23.199647: rig_register: rig_register (115)
2019-12-24:17:51:23.199741: rig_register called
2019-12-24:17:51:23.199834: rig_register: rig_register (113)
2019-12-24:17:51:23.199928: rig_register called
2019-12-24:17:51:23.200021: rig_register: rig_register (114)
2019-12-24:17:51:23.200115: rig_register called
2019-12-24:17:51:23.200210: rig_register: rig_register (128)
2019-12-24:17:51:23.200305: rig_register called
2019-12-24:17:51:23.200398: rig_register: rig_register (131)
2019-12-24:17:51:23.200491: rig_register called
2019-12-24:17:51:23.200584: rig_register: rig_register (116)
2019-12-24:17:51:23.200678: rig_register called
2019-12-24:17:51:23.200771: rig_register: rig_register (103)
2019-12-24:17:51:23.200864: rig_register called
2019-12-24:17:51:23.200956: rig_register: rig_register (124)
2019-12-24:17:51:23.201049: rig_register called
2019-12-24:17:51:23.201143: rig_register: rig_register (104)
2019-12-24:17:51:23.201239: rig_register called
2019-12-24:17:51:23.201333: rig_register: rig_register (125)
2019-12-24:17:51:23.201427: rig_register called
2019-12-24:17:51:23.201528: rig_register: rig_register (129)
2019-12-24:17:51:23.201652: rig_register called
2019-12-24:17:51:23.201775: rig_register: rig_register (137)
2019-12-24:17:51:23.201880: rig_register called
2019-12-24:17:51:23.201973: rig_register: rig_register (132)
2019-12-24:17:51:23.202067: rig_register called
2019-12-24:17:51:23.202161: rig_register: rig_register (130)
2019-12-24:17:51:23.202256: rig_register called
2019-12-24:17:51:23.202350: rig_register: rig_register (117)
2019-12-24:17:51:23.202444: rig_register called
2019-12-24:17:51:23.202537: rig_register: rig_register (119)
2019-12-24:17:51:23.202631: rig_register called
2019-12-24:17:51:23.202723: rig_register: rig_register (118)
[... 80 lines cut ...]
2019-12-24:17:51:34.224385: 0000    41 49 30 3b                                         AI0;            
2019-12-24:17:51:34.224417: cmd_str = ID;
2019-12-24:17:51:34.224447: write_block called
2019-12-24:17:51:34.224497: write_block(): TX 3 bytes
2019-12-24:17:51:34.224533: 0000    49 44 3b                                            ID;             
2019-12-24:17:51:34.224564: read_string called
2019-12-24:17:51:36.226659: read_string(): Timed out 2.2049 seconds after 0 chars
2019-12-24:17:51:36.226724: serial_flush called
2019-12-24:17:51:36.226756: serial_flush: tcflush
2019-12-24:17:51:36.226794: cmd_str = AI0;
2019-12-24:17:51:36.226829: write_block called
2019-12-24:17:51:36.226887: write_block(): TX 4 bytes
2019-12-24:17:51:36.226985: 0000    41 49 30 3b                                         AI0;            
2019-12-24:17:51:36.227021: cmd_str = ID;
2019-12-24:17:51:36.227051: write_block called
2019-12-24:17:51:36.227103: write_block(): TX 3 bytes
2019-12-24:17:51:36.227154: 0000    49 44 3b                                            ID;             
2019-12-24:17:51:36.227201: read_string called
2019-12-24:17:51:38.229301: read_string(): Timed out 2.2052 seconds after 0 chars
2019-12-24:17:51:38.229363: serial_flush called
2019-12-24:17:51:38.229408: serial_flush: tcflush
2019-12-24:17:51:38.229453: cmd_str = AI0;
2019-12-24:17:51:38.229495: write_block called
2019-12-24:17:51:38.229560: write_block(): TX 4 bytes
2019-12-24:17:51:38.229606: 0000    41 49 30 3b                                         AI0;

N0NB, 2019-12-24:
Try my fork again...I had the sleep macros wrong and have improved the flow for Yaesu set_powerstat now.
I'm hoping it only needs the one byte to wake up.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

hlnurse, 2019-12-24:
Mike, initial tests are good.  I'll confirm once the op has had a chance to test locally.  Thank you!

Merry Christmas!

Howard W6HN

hlnurse, 2020-01-04:
This fix looks good.  Thank you Mike!  I will close it.

Howard W6HN

closed 2020-01-04 by hlnurse