Hamlib/Hamlib #164: FT-991A Set VFOA and VFOB only possible when GPredict is running

oe3jtb, 2020-01-03:
I am using Windows 10, RigCtl last version  and GPredirect 2.3.37 as decribed here
http://w7hd.ddns.net/yaesu/ft991a/gpredict-ft991a.pdf
Everthing is fine as long as I run GPredirect  in the background
But I want to set VFOA and VFOB without running GPredirect
https://www.systutorials.com/docs/linux/man/1-rigctl/

First I tried to set split operation with
"rigctl.exe -m 2 -r localhost:4532 V VFOA F 145620000 V VFOB F 434020000"
When I do that the 434Mhz is in VFO A, VFOB is unchanged
This command
rigctl.exe -m 2 -r localhost:4532 V VFOA F 145620000 I 434020000
is also not working, till I start GPRedict, than VFO A has 145... and VFO B 434

Any hints to make it without running GPRedict in the background or what I am doing wrong.
tnx 73 Alex, OE3JTB

mdblack98, 2020-02-08:
Try running this like this with the extra freq request and let's see what i says
rigctl.exe -m 2 -r localhost:4532 V VFOA F 145620000 f V VFOB F 434020000 f
rigctl.exe -m 2 -r localhost:4532 V VFOA F 145620000 f I 43402000

oe3jtb, 2020-02-08:
HI Mike here is the result
C:\Users\OE3JTB>rigctl.exe -m 2 -r localhost:4532 V VFOA F 145620000 f V VFOB F 434020000 f
set_vfo: error = Feature not available
145620000
set_vfo: error = Feature not available
434020000

C:\Users\OE3JTB> rigctl.exe -m 2 -r localhost:4532 V VFOA F 145620000 f I 43402000
set_vfo: error = Feature not available
145620000

C:\Users\OE3JTB> rigctl.exe -m 2 -r localhost:4532 V VFOA F 145620000 f I 434020000
set_vfo: error = Feature not available
145620000

C:\Users\OE3JTB>

in VFO A there ist 434020.00 VFOB is unchanged
tnx
73 ALex

mdblack98, 2020-02-10:
I need to see debug info from the rigctld side.  So run rigctld with 5 'vs and send me the log.
Mike

oe3jtb, 2020-02-16:
ok so I did
Recommend using --vfo switch for rigctld
rigctl and netrigctl will automatically detect vfo mode
rigctld, Hamlib 4.0~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
yaesu: initrigs4_yaesu called
rig_register called
rig_register: rig_register (121)
rig_register called
rig_register: rig_register (127)
rig_register called
rig_register: rig_register (110)
rig_register called
rig_register: rig_register (105)
rig_register called
rig_register: rig_register (106)
rig_register called
rig_register: rig_register (107)
rig_register called
rig_register: rig_register (139)
rig_register called
rig_register: rig_register (109)
rig_register called
rig_register: rig_register (120)
rig_register called
rig_register: rig_register (101)
rig_register called
rig_register: rig_register (122)
rig_register called
rig_register: rig_register (123)
rig_register called
rig_register: rig_register (111)
rig_register called
rig_register: rig_register (115)
rig_register called
rig_register: rig_register (113)
rig_register called
rig_register: rig_register (114)
rig_register called
rig_register: rig_register (128)
rig_register called
rig_register: rig_register (131)
rig_register called
rig_register: rig_register (116)
rig_register called
rig_register: rig_register (103)
rig_register called
rig_register: rig_register (124)
rig_register called
rig_register: rig_register (104)
rig_register called
rig_register: rig_register (125)
rig_register called
rig_register: rig_register (129)
rig_register called
rig_register: rig_register (137)
rig_register called
rig_register: rig_register (132)
rig_register called
rig_register: rig_register (130)
rig_register called
rig_register: rig_register (117)
rig_register called
rig_register: rig_register (119)
rig_register called
rig_register: rig_register (118)
rig_register called
rig_register: rig_register (126)
[... 659 lines cut ...]
newcat_close called
newcat_set_trn called
newcat_valid_command called
newcat_valid_command AI
cmd_str = AI0;
serial_flush called
serial_flush: tcflush
cmd_str = AI0;
write_block called
write_block(): TX 4 bytes
0000    41 49 30 3b                                         AI0;
cmd_str = ID;
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;
read_string called
read_string(): RX 7 characters
0000    49 44 30 36 37 30 3b                                ID0670;
newcat_set_cmd: read count = 7, ret_data = ID0670;
port_close called
ser_close called
ser_close: restoring options
Closed rig model 135, 'FT-991 - no clients, will reopen for new clients'
sync_callback: client lock disengaged
Connection closed from cpm:53551

mdblack98, 2020-03-01:
Please try the master branch now and let me know how it works.

oe3jtb, 2020-03-07:
Hi Mike 
now the switching of mode in VFOA and VFOB is working. Have to select new model 1035 instead of 135. 
But the big issue is, that now no tracking with GPredict is possible :-(
tnx 73 Alex

N0NB, 2020-03-07:
I just tested this with a couple of IC9700 owners and VFOA/B tracking was working fine.
But...you have to set up gpredict with a simplex setup instead of duplex.
de Mike W9MDB

 
 

Hamlib-developer mailing list
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oe3jtb, 2020-03-07:
Hi Mike
when I am using former versions 3.3 and 4.0 (model 135) I see in the deamon window action, when I do changes in GPredict. And its working fine tracking
When I start new 4.0. version model 1035 I see no action, even I configure simplex (with older one duplex works fine) 
I see only action, if I call e.g. the change of mode as I wrote above in an extra window

mdblack98, 2020-03-07:
Can you  send a debug log?

oe3jtb, 2020-03-07:
Yes shure, only from the frequency change commands, because GPredict make no action? cheers Alex

mdblack98, 2020-03-07:
I need debug from rigctld side...not rigctl....

rigctld is talking to the rig.

de Mike W9MDB

oe3jtb, 2020-03-08:
HI Mike
first I did the debug with working chande mode, 
than two versions of trying changing frequency (nothing happens at the display)
and last but not least what happens, when I press engange at GPredict
(also no further action possible, same config as with hamlib 3.3 or elder 4.0)
tnx
73 Alex

rigctl.exe -m 2 -r localhost:4532 M USB 0 X LSB 0
Microsoft Windows [Version 10.0.18363.657]
(c) 2019 Microsoft Corporation. Alle Rechte vorbehalten.

C:\Users\OE3JTB>rigctld.exe -vvvvv -r \\.\com14 -m 1035 -s 38400 -t 4532 -C "serial_speed=38400,stop_bits=2,rts_state=ON,dtr_state=OFF,serial_handshake=None"
Recommend using --vfo switch for rigctld
rigctl and netrigctl will automatically detect vfo mode
rigctld, Hamlib 4.0~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
yaesu: initrigs4_yaesu called
rig_register called
rig_register: rig_register (1021)
rig_register called
rig_register: rig_register (1027)
rig_register called
rig_register: rig_register (1010)
rig_register called
rig_register: rig_register (1005)
rig_register called
rig_register: rig_register (1006)
rig_register called
rig_register: rig_register (1007)
rig_register called
rig_register: rig_register (1039)
rig_register called
rig_register: rig_register (1009)
rig_register called
rig_register: rig_register (1020)
rig_register called
rig_register: rig_register (1001)
rig_register called
rig_register: rig_register (1022)
rig_register called
rig_register: rig_register (1023)
rig_register called
rig_register: rig_register (1011)
rig_register called
rig_register: rig_register (1015)
rig_register called
rig_register: rig_register (1013)
rig_register called
rig_register: rig_register (1014)
rig_register called
rig_register: rig_register (1028)
rig_register called
rig_register: rig_register (1031)
rig_register called
rig_register: rig_register (1016)
rig_register called
rig_register: rig_register (1003)
rig_register called
rig_register: rig_register (1024)
rig_register called
rig_register: rig_register (1004)
rig_register called
rig_register: rig_register (1025)
rig_register called
rig_register: rig_register (1029)
rig_register called
rig_register: rig_register (1037)
[... 2258 lines cut ...]
newcat_close called
newcat_set_trn called
newcat_valid_command called
newcat_valid_command AI
cmd_str = AI0;
serial_flush called
serial_flush: tcflush
cmd_str = AI0;
write_block called
write_block(): TX 4 bytes
0000    41 49 30 3b                                         AI0;
cmd_str = ID;
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;
read_string called
read_string(): RX 7 characters
0000    49 44 30 36 37 30 3b                                ID0670;
newcat_set_cmd: read count = 7, ret_data = ID0670;
port_close called
ser_close called
ser_close: restoring options
Closed rig model 1035, 'FT-991 - no clients, will reopen for new clients'
sync_callback: client lock disengaged
Connection closed from cpm:56409

N0NB, 2020-03-09:
Update your hamlib...this was fixed.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2020-03-11:
Need to know how this is behaving now with the current master.
If I don't hear anything back by 2020-03-17 will close this ticket.

oe3jtb, 2020-03-15:
Ok the behavior is
GPredict is working again

rigctl.exe -m 2 -r localhost:4532 V VFOA F 145620000 V VFOB F 434020000 

sets first 145620 and miliseconds later 434020 in VFO A, VFO B is unchanged
73 ALex

mdblack98, 2020-03-15:
That doesn't make sense.  Did you mean "434020 in VFO B, VFO A is unchanged" or do you mean gpredict is changing VFOB?

oe3jtb, 2020-03-15:
Hi If I am running GPRedict, everthing is working fine, frequency is changed due to Doppler fine.

But when I only start in a cmd window 
rigctl.exe -m 2 -r localhost:4532 V VFOA F 145620000 V VFOB F 434020000 V

as above decribed than there is the behavior, that no frequency is changed in VFO B

Perhaps I need another command :-)
tnx
73 Alex

oe3jtb, 2020-03-15:
ok now I saw my first post 
this one 
rigctl.exe -m 2 -r localhost:4532 V VFOA F 145620000 I 434020000
is working
tnx Mike
73 Alex

oe3jtb, 2020-03-15:
works

closed 2020-03-15 by oe3jtb

N0NB, 2020-03-15:
What version of hamlib are you running?  Can you please test my fork?  https://github.com/mdblack98/Hamlib
And I need to see the debug log from rigctld (not rigctl)
And I guess you're in satellite mode on the rig since your'e doing cross-band split.So this should workrigctl.exe -m 2 -r localhost:4532 V Main F 145620000 V Sub F 434020000 V
Or thisrigctl.exe -m 2 -r localhost:4532 V Main F 145620000 V TX F 434020000 V

de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer