Hamlib/Hamlib #165: Yaesu FT-991A and set CTCSS

oe3jtb, 2020-01-03:
Hi nice team, 
would it ever planned that setting CTCSS with rigctl is possible. WIth 3.3 message not supported appears
would be nice for sat op
tnx 
73 Alex, OE3JTB

N0NB, 2020-01-03:
You tried the latest github version?
It appears to be supported now.
rigctl -m 135 | grep CTCSS
 Can set CTCSS: YCan get CTCSS: Y
I guess 3.3 shows "N' for these two entries?
de Mike W9MDB 

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oe3jtb, 2020-01-04:
Tnx for the hint 
I installed 4.0 from http://n0nb.users.sourceforge.net, same result but also this command , which worked with 3.3. rigctl.exe -m 2 -r localhost:4532 M USB 0 X LSB 0
isnt working with 4.0 only main VFO is set correct..

N0NB, 2020-01-05:
Can you send a debug log from each version?  Add "-vvvvv -Z" to the command line and redirect to a log file -- you can't use the -Z on 3.3 though....that's a new switch that provides timing info.
 rigctl.exe -m 2 -r localhost:4532 M USB 0 X LSB 0 -vvvvv -Z >&log.txt

de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-01-05:
I may have fixed the problem....
Try my fork.
https://github.com/mdblack98/Hamlib

de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oe3jtb, 2020-01-05:
HI Mike

tnx for fast  answer

with &log.txt the command dosent work, without

C:\Users\OE3JTB> rigctl.exe -m 2 -r localhost:4532 M USB 0 X LSB 0 -vvvvv -Z >log.txt
Recommend using --vfo switch for rigctl with rigctld
Command ' ' not found!
Command ' ' not found!

and only in the main VFOA the mode is set to LSB (not USB as expected)

tnx

73 Alex

BTW

nightly built tomorrow has the new binaries with your change?

N0NB, 2020-01-05:
Try this...switches have to be before any rig commands.
 rigctl.exe -m 2 -r -vvvvv -Z localhost:4532 M USB 0 X LSB 0 2>1 >log.txt

de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oe3jtb, 2020-01-05:
Hi Mike
so with switches before nothing happens, in the deamon window I see no action until I remove -Z running nightly from 3/JAN
73 Alex

N0NB, 2020-01-05:
Can you show what you're doing?
Mike

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oe3jtb, 2020-01-05:
HI MIke 
yes shure
https://youtu.be/5OwNgBBsxWE
what is on
deamon 4.0 32 bit (64 bit is the same) is running .  First I ran cmd as you wrote, you see no action in deamon window. THan deleting the log statement, see error rig_open
Than without -vvvvvv and -Z and action in the deamon window.
But the mode is only set in VFOA, but with the mode X FM, which should be in VFOB
With 3.3. its working fine
tnx
73 ALex

N0NB, 2020-01-05:
Wrong format on the command line...the address must be right after the -r switch.
For 4.0
rigctl.exe -m 2 -r localhost:4532 -vvvvv -Z M FM 0 X FM 0 2>1 >log4.txt
For 3.3rigctl.exe -m 2 -r localhost:4532 -vvvvv M FM 0 X FM 0 2>1 >log3.txt
de Mike W9MDB

 
 

https://youtu.be/5OwNgBBsxWE

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oe3jtb, 2020-01-06:
Hi Mike 
ok now I ran the command with hamlib 4.0 
result in the log
Opened rig model 2, 'NET rigctl'
set_mode: error = Invalid parameter

result on the FT-991A
only VFOA has the right mode, VFOB is unchanged

with hamlib 3.3.
result in the log
Opened rig model 2, 'NET rigctl'

result on the Yaesu is correct setting of both VFOs

tnx
73 Alex

N0NB, 2020-01-06:
Can you please attach the logs?
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oe3jtb, 2020-01-06:
HI Mike 
here they are
[log3.txt](https://github.com/Hamlib/Hamlib/files/4026441/log3.txt)
[log4.txt](https://github.com/Hamlib/Hamlib/files/4026442/log4.txt)
tnx
73 Alex

mdblack98, 2020-01-06:
You didn't have the -vvvvv switch apparently...nothing in those files.
de Mike W9MDB

oe3jtb, 2020-01-06:
Hi Mike
I ran the cmds with cut and paste out of your posting and changed nothing, even the name of the logfiles are log3 and 4
I am in my qth next saturday , I will do it once more and make a screen image as before.
tnx
73 Alex

oe3jtb, 2020-01-26:
Hi MIke 
so there is a video , latest 64 bit version installed see https://youtu.be/WdDPIKJ4EWo

result in log4
Opened rig model 2, 'NET rigctl'
set_mode: error = Invalid parameter
hmh any ideas
tnx
73 Alex

mdblack98, 2020-01-26:
Can you please run that again so I can get the whole debug log?
Change 2>1 to 2>&1 and it will all go to the file that you can email me.
de Mike W9MDB

oe3jtb, 2020-02-08:
Mike did you get the log by mail?

N0NB, 2020-02-10:
No I didn't.  I only saw old logs with nothing in them.
Please send logs again.

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oe3jtb, 2020-02-11:
qrz.com email ok?

N0NB, 2020-02-11:
Yes

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oe3jtb, 2020-02-15:
ok send it via mail last week, did you receive it?

N0NB, 2020-02-15:
I never saw any debug from the rigctld side -- that's where the problem would be.
Mike

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

oe3jtb, 2020-02-15:
Ok I send it to both email addresses but no luck
so here ist the log I sent

C:\Users\OE3JTB>rigctl.exe -m 2 -r localhost:4532 -vvvvv -Z M FM 0 X FM
0 2>&1 >log4.txt

result
2020-01-26:17:08:50.161135: rigctl, Hamlib 4.0~git
2020-01-26:17:08:50.161135: Report bugs to
<hamlib-developer@lists.sourceforge.net>

2020-01-26:17:08:50.162134: rig_init called
2020-01-26:17:08:50.162134: initrigs4_dummy: _init called
2020-01-26:17:08:50.162134: rig_register called
2020-01-26:17:08:50.162134: rig_register: rig_register (1)
2020-01-26:17:08:50.162134: rig_register called
2020-01-26:17:08:50.162134: rig_register: rig_register (2)
2020-01-26:17:08:50.163133: rig_register called
2020-01-26:17:08:50.163133: rig_register: rig_register (4)
2020-01-26:17:08:50.163133: rig_register called
2020-01-26:17:08:50.163133: rig_register: rig_register (5)
2020-01-26:17:08:50.163133: netrigctl_init version 1.2
2020-01-26:17:08:50.163133: rig_open called
2020-01-26:17:08:50.164133: port_open called
2020-01-26:17:08:50.164133: network_open called
2020-01-26:17:08:50.164133: network_open version 1.0
2020-01-26:17:08:50.176121: netrigctl_open called
2020-01-26:17:08:50.176121: netrigctl_transaction: called len=9
2020-01-26:17:08:50.176121: network_flush called
2020-01-26:17:08:50.177120: write_block called
2020-01-26:17:08:50.177120: write_block(): TX 9 bytes
2020-01-26:17:08:50.177120: 0000    5c 63 68 6b 5f 76 66 6f
0a                          \chk_vfo.
2020-01-26:17:08:50.177120: read_string called
2020-01-26:17:08:52.119319: read_string(): RX 2 characters
2020-01-26:17:08:52.120318: 0000    30
0a                                               0.
2020-01-26:17:08:52.121317: netrigctl_open: vfo_mode=0
2020-01-26:17:08:52.121317: netrigctl_transaction: called len=12
2020-01-26:17:08:52.122317: network_flush called
2020-01-26:17:08:52.122317: write_block called
2020-01-26:17:08:52.123314: write_block(): TX 12 bytes
2020-01-26:17:08:52.124315: 0000    5c 64 75 6d 70 5f 73 74 61 74 65
0a                 \dump_state.
2020-01-26:17:08:52.125312: read_string called
2020-01-26:17:08:52.132306: read_string(): RX 2 characters
2020-01-26:17:08:52.132306: 0000    30
0a                                               0.
2020-01-26:17:08:52.133307: read_string called
2020-01-26:17:08:52.133307: read_string(): RX 4 characters
2020-01-26:17:08:52.134304: 0000    31 33 35
0a                                         135.
2020-01-26:17:08:52.135304: read_string called
2020-01-26:17:08:52.135304: read_string(): RX 2 characters
2020-01-26:17:08:52.136303: 0000    32
0a                                               2.
2020-01-26:17:08:52.136303: read_string called
2020-01-26:17:08:52.137301: read_string(): RX 51 characters
2020-01-26:17:08:52.137301: 0000    33 30 30 30 30 2e 30 30 30 30 30 30
20 34 37 30     30000.000000 470
2020-01-26:17:08:52.138300: 0010    30 30 30 30 30 30 2e 30 30 30 30 30
30 20 30 78     000000.000000 0x
2020-01-26:17:08:52.138300: 0020    31 64 39 66 20 2d 31 20 2d 31 20 30
78 33 20 30     1d9f -1 -1 0x3 0
2020-01-26:17:08:52.141298: 0030    78 33
0a                                            x3.
2020-01-26:17:08:52.142296: read_string called
2020-01-26:17:08:52.143296: read_string(): RX 14 characters
2020-01-26:17:08:52.143296: 0000    30 20 30 20 30 20 30 20 30 20 30 20
30 0a           0 0 0 0 0 0 0.
[... 462 lines cut ...]
2020-01-26:17:08:52.503961: netrigctl_set_split_mode called
2020-01-26:17:08:52.503961: netrigctl_transaction: called len=7
2020-01-26:17:08:52.504960: network_flush called
2020-01-26:17:08:52.504960: write_block called
2020-01-26:17:08:52.505959: write_block(): TX 7 bytes
2020-01-26:17:08:52.506958: 0000    58 20 46 4d 20 30
0a                                X FM 0.
2020-01-26:17:08:52.509956: read_string called
2020-01-26:17:08:52.856634: read_string(): RX 7 characters
2020-01-26:17:08:52.856634: 0000    52 50 52 54 20 30
0a                                RPRT 0.
2020-01-26:17:08:52.857633: rig_close called
2020-01-26:17:08:52.857633: netrigctl_close called
2020-01-26:17:08:52.858631: write_block called
2020-01-26:17:08:52.859632: write_block(): TX 2 bytes
2020-01-26:17:08:52.860631: 0000    71
0a                                               q.
2020-01-26:17:08:52.861629: port_close called
2020-01-26:17:08:52.863628: network_close called
2020-01-26:17:08:52.864627: rig_cleanup called

The logs shows the same as before

Opened rig model 2, 'NET rigctl'
set_mode: error = Invalid parameter

N0NB, 2020-02-15:
I need the rigctld side... not rigctl.
Mike

oe3jtb, 2020-02-16:
ok here the deamon 
Recommend using --vfo switch for rigctld
rigctl and netrigctl will automatically detect vfo mode
rigctld, Hamlib 4.0~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
yaesu: initrigs4_yaesu called
rig_register called
rig_register: rig_register (121)
rig_register called
rig_register: rig_register (127)
rig_register called
rig_register: rig_register (110)
rig_register called
rig_register: rig_register (105)
rig_register called
rig_register: rig_register (106)
rig_register called
rig_register: rig_register (107)
rig_register called
rig_register: rig_register (139)
rig_register called
rig_register: rig_register (109)
rig_register called
rig_register: rig_register (120)
rig_register called
rig_register: rig_register (101)
rig_register called
rig_register: rig_register (122)
rig_register called
rig_register: rig_register (123)
rig_register called
rig_register: rig_register (111)
rig_register called
rig_register: rig_register (115)
rig_register called
rig_register: rig_register (113)
rig_register called
rig_register: rig_register (114)
rig_register called
rig_register: rig_register (128)
rig_register called
rig_register: rig_register (131)
rig_register called
rig_register: rig_register (116)
rig_register called
rig_register: rig_register (103)
rig_register called
rig_register: rig_register (124)
rig_register called
rig_register: rig_register (104)
rig_register called
rig_register: rig_register (125)
rig_register called
rig_register: rig_register (129)
rig_register called
rig_register: rig_register (137)
rig_register called
rig_register: rig_register (132)
rig_register called
rig_register: rig_register (130)
rig_register called
rig_register: rig_register (117)
rig_register called
rig_register: rig_register (119)
rig_register called
rig_register: rig_register (118)
rig_register called
rig_register: rig_register (126)
[... 284 lines cut ...]
newcat_close called
newcat_set_trn called
newcat_valid_command called
newcat_valid_command AI
cmd_str = AI0;
serial_flush called
serial_flush: tcflush
cmd_str = AI0;
write_block called
write_block(): TX 4 bytes
0000    41 49 30 3b                                         AI0;
cmd_str = ID;
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;
read_string called
read_string(): RX 7 characters
0000    49 44 30 36 37 30 3b                                ID0670;
newcat_set_cmd: read count = 7, ret_data = ID0670;
port_close called
ser_close called
ser_close: restoring options
Closed rig model 135, 'FT-991 - no clients, will reopen for new clients'
sync_callback: client lock disengaged
Connection closed from cpm:53602

N0NB, 2020-02-16:
According to the notes in the source code the FT-991 was returning an error trying to set passband for FM mode.

PKTFM should work....which is probably what you want anyways I would imagine.

But can you see if this works in rigctl?  I think the first may return an error where the PKTFM version will work.  

M FM
w NA00;
w NA01;

M PKTFM
w NA00;
w NA01;

de Mike W9MDB

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2020-03-11:
What's the status of this?  If I don't hear back by 2020-03-17 will close this ticket.

oe3jtb, 2020-03-15:
Hi the behavior with new hamlib 4.0 is
changing modes in both VFOs works fine
CTCSS command implemented?
tnx 73 Alex

mdblack98, 2020-03-15:
Fixed

closed 2020-03-15 by mdblack98