Hamlib/Hamlib #167: FT-991A not working with hamlib 3.3

HB9FIH, 2020-01-05:
Linux Mint Mate 19.3 64
Latest hamlib 3.3 compiled installed fresh.
Together with WSJT-X, JTDX no cat possible 
(both have problems with the interface - altough I set same parameter like  in FLDigi - FLDigi runs UFB)

says only after timeout, unable to communicate with the rig (or similar)

btw - same issue on WIN10 

73 Erich

N0NB, 2020-01-05:
Please try my fork.....
https://github.com/mdblack98/Hamlib

de Mike W9MDB 

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

HB9FIH, 2020-01-06:
Tnx Mike, 

I try now to rebuild ...
Hangs always in the  build-aux  - where is missing the aclocal.m4 -
seems there is used an old version.
I still check and research... 

73 de Erich 
---

..................
vy 73 Erich Rieder
HB9FIH
HS0ZLS

Webseite ist NEU unter: http://www.hb9fih.org 

 

Links:
------
[1]
https://github.com/Hamlib/Hamlib/issues/167?email_source=notifications&amp;email_token=AITJN53YGQV6KADSFAVQVMTQ4JHXDA5CNFSM4KC5G33KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEIEAHUI#issuecomment-570950609
[2]
https://github.com/

mdblack98, 2020-01-06:
aclocal.m4 is created by "bootstrap".  You have to run that first, the configure, make 
de Mike W9MDB

 
 

 

Links:
------
[1]
https://github.com/Hamlib/Hamlib/issues/167?email_source=notifications&amp;email_token=AITJN53YGQV6KADSFAVQVMTQ4JHXDA5CNFSM4KC5G33KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEIEAHUI#issuecomment-570950609
[2]
https://github.com/

HB9FIH, 2020-01-06:
Hi Mike, just done (short before I go to sleep - here = UTC) with full
success ! - now JTDX runs - also WSJT-X under Linux.
- so I am ready to sleep quiet ;-) 

Yes I MUST  become familiar with the tools again.....I've forgot so many
when I did not use Linux for nearly 20 years.

(I was maintainer and owner of a server farm and built the webservers
(Apache) and the Mail Servers always from source...but this is
history....) 

Many thanks - all the best for you 

---

..................
vy 73 Erich Rieder
HB9FIH
HS0ZLS

Webseite ist NEU unter: http://www.hb9fih.org 

 

Links:
------
[1]
https://github.com/Hamlib/Hamlib/issues/167?email_source=notifications&amp;email_token=AITJN5Y4KE4JNNQER4WHK3TQ4OS3JA5CNFSM4KC5G33KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEIG55SA#issuecomment-571334344
[2]
https://github.com/

mdblack98, 2020-01-18:
Fixed in 4.0

closed 2020-01-18 by mdblack98