Hamlib/Hamlib #169: IC-7100 & rigctl - bad CI-V address

malcolmh, 2020-01-06:
Running rigctl for IC7100 (rig 341), the ci-v address output is not the value in priv_caps. If I use the -c switch with the correct value, that works OK. If I edit the value in ic7100.c, it still outputs the same wrong value. FYI, 0x88 is the correct value - 0x34 is the value that is output.

My fork is synced with the latest mdblack fork.

N0NB, 2020-01-06:
That's because the IC7100 is rig 370The IC-R7000 is 341
   341  Icom                   IC-R7100                0.19.0          Untested   370  Icom                   IC-7100                 0.19.0          Untested

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

malcolmh, 2020-01-06:
My Bad!

closed 2020-01-06 by malcolmh