Hamlib/Hamlib #171: Handling for waterfall/bandscope data on Icom SDR rigs

arodland, 2020-01-11:
Icom SDR rigs (7300, 7610, 9700, etc.) support sending waterfall data over USB for display on a PC. When the scope data output (command 27 subcommand 20) is enabled, the rig sends a constant stream of unsolicited scope data reports (command 27 subcommand 00), with raw binary payloads, between command responses. Their RS-BA1 remote control software and a few other third-party Windows programs support decoding and displaying it.

I want to make something similar but cross-platform, and of course I don't want my program to take sole ownership of the serial port; I want to have rigctld running and do other hamlib things at the same time. Practically speaking, this means that hamlib needs to be the one handling the scope data.

I don't care if hamlib *interprets* it, but what I would hope for would be

* For hamlib not to desync or show any behavior difference to uninterested clients, when connected to a rig with scope data output enabled.
* For a TCP client to have a way to subscribe to the scope data, receiving it as it comes in, with sensible framing (It could also be used for other kinds of transcieve data, e.g. Icoms will tell you about frequency changes and PTT toggles if you ask... but I don't care if it doesn't). I wouldn't mind if starting one of these subscriptions made the socket totally unable to issue other commands (to avoid framing problems).

I think this should be more or less possible with the existing event loop code in hamlib, and I'm willing to make an attempt at it myself, but I wanted to run the approach by someone first, and get some input, to increase the chances of actually getting a patch accepted.

N0NB, 2020-01-11:
My first thought would be to add an audio command that would turn into an audio stream.  Pretty simple to implement.Then...if you want audio...you'd tell rigctl, for example, "\audio_start 4900" and you'd get UDP data to start streaming to port 4900.  And make it multicast so it can go to multiple applications.audio_start would spawn a thread to handle it and "\audio_stop" would stop it.
de Mike W9MDB

 
 

   

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-01-11:
Actually my first thought was wrong.
As Bill noted hamlib is synchronous and that audio data is async.
If it's all coming over the same serial port won't work as-is since any CAT command will flush the buffer.  It should be possible to change that logic so that before flushing it checks for audio...that's the one possibility of getting this to work....so read any available data and if it's audio send it to the UDP port otherwise flush it.

de Mike W9MDB

 

    On Saturday, January 11, 2020, 07:47:30 AM CST, Black Michael <mdblack98@yahoo.com> wrote:  
 
 My first thought would be to add an audio command that would turn into an audio stream.  Pretty simple to implement.Then...if you want audio...you'd tell rigctl, for example, "\audio_start 4900" and you'd get UDP data to start streaming to port 4900.  And make it multicast so it can go to multiple applications.audio_start would spawn a thread to handle it and "\audio_stop" would stop it.
de Mike W9MDB

 
 

   

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

roelandjansen, 2020-01-12:
audio does not come via cat packets; that's "just" pcm coming in and I do have some sample code that can do that and with some socat stuff it's possible to send it to different addresses.

Biggest issues so far is that audio is very different on different platforms and with linux you will
find yourself that you can find out which rig uses what USB ports (as the serial numbers and the name of the rigs are announced); the audio codecs is a but different as they do not carry that information.

You can find back in the logs with some challenges which codecs are related to what rig.

N0NB, 2020-01-12:
Then the threaded solution I mentioned earlier would work.  You would want the audio device and port number as an argument.
\audio_start 4400 /dev/....\audio_stop\audio_list
audio_list would show the selectable audio devices that can be arguments for audio_start
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

arodland, 2020-01-12:
I'm not sure how audio ever got into this discussion, can we start over? This is about waterfall data, not audio.

mdblack98, 2020-01-12:
Because it's audio data -- no such thing as waterfall data coming from a rig.  Waterfall is post-processed audio data.
de Mike W9MDB

N0NB, 2020-01-12:
On 12/01/2020 16:59, Michael Black wrote:
> Because it's audio data -- no such thing as waterfall data coming from 
> a rig.  Waterfall is post-processed audio data.
> de Mike W9MDB 

Mike,

I don't believe that is correct, the WF data from the IC7300 and similar 
is a bitmap image, i.e. exactly what is displayed on the WF display on 
the rig.

73
Bill
G4WJS.

mdblack98, 2020-01-12:
Yup...you're right...just looked at the manual...doesn't quite say what the format is though...mentions RGB values.
Mike 

 

Mike,

I don't believe that is correct, the WF data from the IC7300 and similar 
is a bitmap image, i.e. exactly what is displayed on the WF display on 
the rig.

73
Bill
G4WJS.

mikaelnousiainen, 2020-01-12:
Based on my understanding (and by reading the IC-7300 or IC-7610 manuals), the "Scope waveform data" (what the manual calls it) consists of bytes, which resemble FFT results.

On page 19-12 of the IC-7300 FULL MANUAL:

    The transceiver outputs the drawn waveform data.
    The data range or data length of the waveform
    data is judged by the controller. (The data range is
    basically the same as the display size of the scope
    on the controller.)

    Data range: 0~160
    Data length: 475

I think the above means that the packet contains 475 bytes with values from a minimum of 0 to maximum of 160 for each frequency "bucket" (as in FFT). The "Scope waveform data" includes some header bytes, which indicate the edges or the center of the waterfall frequencies, so one can calculate which frequency each "bucket" indicates.

It seems that the data length and range are different for IC-7610 (and I'd assume they may be different for IC-9700 too), but otherwise the format seems to be similar.

It would certainly be nice to be able to extract this data in Hamlib, but the issue is that it is very Icom-specific. It's not I/Q, audio or even bitmap data, but some kind of pseudo-FFT results, which allow you to plot waterfall based on the intensity levels for the frequency buckets.

arodland, 2020-01-14:
It's *basically* a bitmap ­— @mikaelnousiainen  has the details about right. It's a bunch of bytes, each one represents the energy in a frequency bin, 0 is least, more is more, details are dependent on the rig and on the various scope settings. The complexity of interpreting it is why I wasn't gunning for hamlib to interpret these messages — I was just looking for a way to pass them through to a client that can.

mikaelnousiainen, 2020-01-14:
Would it make sense to do the following to extract the data to clients:

1) Implement some (or why not all) of the 27 xx CI-V commands as Icom-specific (but not model-specific) ext params (not ext levels, as the feature isn't directly RX or TX related), so that Hamlib clients can turn on/off and configure the waterfall. The commands seem to be identical (with some additions in 7610) for IC-7300, IC-7610 and IC-9700 at least. There may be other rigs that support this too, but I haven't checked. This way one could just enable the spectrum support on rigs that have it.

2) Design a push-based response format for rig-specific data. There should be an indicator in the beginning of this response about the contents, e.g. `Data: ICOM_WATERFALL` or similar. In my opinion, Hamlib should do some interpretation of the waterfall data and expose it in a format that is easier to parse by the clients. The Hamlib protocol is text-based after all, so we can't (and shouldn't) even push the binary data directly out. A simple way (key-value pairs, e.g. `Center: 3700000 Hz`, like with other command responses) to express the waterfall edges/dimensions with Base64-encoded or even numeric hex data for the waterfall bytes could be enough. Another option would be to use a JSON object to contain all of the fields and waterfall data. That would make parsing relatively easy in most clients and the boundaries of the response would be easy to interpret.

Comments?

malcolmh, 2020-01-14:
As a test, I implemented the 0x27 commands for my ICR8600 & sent the 27 11 ON command. Indeed, a continuous stream of 27 00 frames started arriving until I sent the 27 11 OFF command.

To be able to use this function, hamlib would have to intercept theses frames & send them to a call-back in the application.

N0NB, 2020-01-14:
So what happens if you send a CAT command while that's streaming?
I assume the response from the CAT command is just being mixed with the audio stream packets.
It might be possible to do this with a thread and a mutex.  The icom cat command would request the mutex and the 27 00 thread would release and re-request the mutex between packets.  I don't think we want to stop/start the streaming.
I'm curious as the amount of time it takes for a CAT command to return a packet while the streaming is going on.  There's no handshaking and I guess the streaming doesn't require an ACK from the software to continue so there's going to be some buffering going on.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

malcolmh, 2020-01-14:
Yes, the response frames to CAT command frames sent during the 27 11 ON state are mixed in with the scope frames.

The stream of frames is continuous & does not wait for an ACK. My test used the serial line & each scope sweep is chopped up into 11 separate frames, each with the 27 00 overhead. The documentation suggests that when using a LAN, the complete sweep is contained withing one frame.

arodland, 2020-01-15:
@mikaelnousiainen yes, that would make me abundantly happy. Proper decoding for the "setting" bits and hex payload for the waterfall data seems like a fair compromise. I think that clients will probably have to send a command that says they're interested in, and want, such unsolicited data; otherwise having the waterfall turned on would break the request/response loop of a great many existing clients :)

mdblack98, 2020-01-18:
I believe this is best handled by a separate program and running two CAT cables on the Icom rigs.  Do CAT control over the CI-V cable and waterfall data over the regular COM port.  That's because the CI-V can only do 19,200 and the COM port 115,200.

closed 2020-01-18 by mdblack98

roelandjansen, 2020-01-18:
This is not entirely correct, you can run civ and scope data on a single
port at 115k2.

It entirely depends on the link/unlink from remote, and then you either
want one USB cable _or_ lan. The latter is preferred but so far there has
not been enough efforts to find out how rs-ba1 does this.

For a picture how the rates and link/unlink works, I suggest to look at the

rs-ba1- rev-engineering@groups.io

group. It contains some links on how the "hardware" is connected internally
and what setting affects what port, rate etc.

Roeland.

N0NB, 2020-01-18:
I didn't say the "only" way to handle it.  I said the "best".  The added latencies on the CAT control could prove to have problems...for example with DTR/RTS keying for CW and RTTY in FLRig for example....which of course won't work over CI-V anyways so may be moot point.
The LAN connection would be another matter....it's not a secret.  That would likely work OK but with the synchronous nature of hamlib a mutex controlled thread might work.
Mike

 

—

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

roelandjansen, 2020-01-18:
I know it's not the only way to handle it.

When it comes to latencies, I run both rs-ba1 as well as hamlib on the same
rig and even when the rig sends out scope data (in this case the 7300 so
surely only over usb) I have not noticed any latency differences. If I have
a tight loop that queries the frequency for instance, it works as snappy as
using the controls or remotely changed by rs-ba1.

I recall the dtr/rts keying is done on the "second" virtual usb port
anyways. can't check for sure as I am not home now.

The lan connection is a secret because there is encryption involved; that
is what I refer to.