Hamlib/Hamlib #175: Hamlib error: IO error while opening connection to rig (FT-991A)

PA3VOS, 2020-01-15:
Since two days I get errors after starting WSJT-X (version 2.1.0) running on my Win10 PC which is connected to my FT-991A. 
I am using this configuration already since 2 months without any problems. The mentioned problem started last sunday after I made a QSO in the mode C4FM. I tried to start WSJT-X again but got a error message HAMLib could not read the mode. In the first place I was not aware the this probably had to do with the C4FM-mode I selected on my FT-991a. So after some try  and error I reset my config in WSJT-X and reconfigured it. Still the same error message refering to HAMLib. So I decided to remove HamLib and reinstalled it with version 3.3.

All com-ports and as well as the audio interface work properly. And are not reinstalled. Als Windows does not show any disconnection of the comports. No hardware has been changed. USB cable is OK and provided with several RF beads MIX31. 

I googled and found more same problems that rever to this HamLib IO error. 
Please help!
73's PA3VOS

N0NB, 2020-01-15:
Turn off C4FM mode in the rig.  C4FM is not currently supported in hamlib for any Yaesu rig.
I'll add it.  Can you compile hamlib?
de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-01-15:
I've submitted a patch to include C4FM mode so hamlib at least shouldn't die when you start it.
I can send you a compiled version if you need it...contact me off-list.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

PA3VOS, 2020-01-15:
Hi Mike,
I turned off C4FM when I got this error message. But that was to late. Before I did that I already reset the settings accidentely. So from that point the.problems started to get worse. I reinstalled the latest version of HamLib. I am working with Windows 10 not with Umbuntu or a Raspberry Pi. So I need a windows exacutable.

mdblack98, 2020-03-01:
Was a hardware problem so closing.

closed 2020-03-01 by mdblack98