Hamlib/Hamlib #176: Rig failure with Yaesu FT-900

Zazzmatazz, 2020-01-17:
Good afternoon;

I am receiving an error when attempting to configure a Yaesu FT-900 in WSJT-X.

My settings are as follows:

- Baud: 4800
- Data Bits: 8
- Stop Bits: Two
- Handshake: None
- PTT Method: CAT
- Mode: USB
- Split Operation: None
- Polling Interval: 1s

The error I am receiving is:
"Rig failure.  Hamlib error: Invalid parameter while testing getting current VFO"

I am seeing the "CAT" indicator appearing on screen, which at least indicates that the radio is seeing the DTR signal, but I don't have any other information beyond that.  I will attempt to connect a serial terminal between the application and the radio in order to examine the communications in more detail.

N0NB, 2020-01-17:
What version of WSJT-X?
Is your rig in some non-standard mode?  Try setting the rig to USB before starting WSJT-X.
de Mike W9MDB

 
 

   

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

Zazzmatazz, 2020-01-17:
Hello @N0NB,

The radio is currently in VFO-A, frequency dial is locked and frequency is currently on 18.100 MHz USB.

N0NB, 2020-01-17:
Unlock the frequency dial.  That may be the problem.

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

Zazzmatazz, 2020-01-17:
There is no change sadly, same exact error.

N0NB, 2020-01-17:
Contact me off list at mdblack98@yahoo.com
Mike

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

closed 2020-01-17 by Zazzmatazz