Hamlib/Hamlib #180: Loading Python 2.7 module gives '_Hamlib.so: undefined symbol: rig_mem_count'

N0NB, 2020-01-18:
With the many changes taking place I thought it would be time to test the bindings.  So far I have tested Perl with success.  Next in line was Python and the build went fine but now I'm seeing the following run time error:

```
$ python pytest.py 
Traceback (most recent call last):
  File "pytest.py", line 11, in <module>
    import Hamlib
  File "/home/nate/local/lib/python2.7/site-packages/Hamlib.py", line 17, in <module>
    _Hamlib = swig_import_helper()
  File "/home/nate/local/lib/python2.7/site-packages/Hamlib.py", line 16, in swig_import_helper
    return importlib.import_module('_Hamlib')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: /home/nate/local/lib/python2.7/site-packages/_Hamlib.so: undefined symbol: rig_mem_count
```

Looking for `rig_mem_count` I see it occurs in only two places in the current master branch:

```
$ git grep rig_mem_count
rig.swg:449:            return rig_mem_count(self->rig);
rig.swg:563:    int nb_chans = rig_mem_count(self->rig);
```

Oddly, these are also the only references to `rig_mem_count` in the Hamlib-3.2 and Hamlib-3.3 branches.  So far as I know, the Python binding worked in those releases.

Just to be sure, I recompiled for a system installation in `/usr/local` and get the same error:

```
$ python pytest.py 
Traceback (most recent call last):
  File "pytest.py", line 11, in <module>
    import Hamlib
  File "/usr/local/lib/python2.7/dist-packages/Hamlib.py", line 17, in <module>
    _Hamlib = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/Hamlib.py", line 16, in swig_import_helper
    return importlib.import_module('_Hamlib')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: /usr/local/lib/python2.7/dist-packages/_Hamlib.so: undefined symbol: rig_mem_count
```

mdblack98, 2020-01-18:
Patch submitted....rig_mem_count was in one of the XXREMOVEDXX blocks.Interesting that python uses it but nothing else does.
de Mike W9MDB

N0NB, 2020-01-18:
Thanks, Mike.  I will pull that later and test it.

N0NB, 2020-01-18:
tcl uses it too.

N0NB, 2020-01-18:
And lua!

Next I'll pull and test.

N0NB, 2020-01-18:
It helps to be in the main directory with `git grep`, otherwise you only get what's in that directory and below.

"I knew that!"

ARRRRGHHHHH!!!

N0NB, 2020-01-19:
I get this now from Lua:

```
$ lua bindings/luatest.lua 
Lua 5.3 test, Hamlib 4.0~git

status(str):		Command completed successfully
get_conf:		path = /dev/Rig, retry = 5, ITU region = 2, ITU region (str) = 2
freq:			5700000000
mode:			FM
bandwidth:		15000.0
lua: bindings/luatest.lua:44: Error in Rig::set_mode (arg 2), expected 'rmode_t' got 'string'
stack traceback:
	[C]: in method 'set_mode'
	bindings/luatest.lua:44: in function 'doStartup'
	bindings/luatest.lua:118: in main chunk
	[C]: in ?
```
I've looked in the various .swg files, but this is above my pay grade.  :-D

N0NB, 2020-01-19:
Looks like Lua is the only binding with an issue after your patch.

mdblack98, 2020-01-19:
I submitted a patch to fix that and put some notes about the needed versions of swig and lua.
de Mike W9MDB

mdblack98, 2020-01-27:
All binding problems fixed

closed 2020-01-27 by mdblack98

N0NB, 2020-01-28:
On 2020-01-28 00:16, Michael Black wrote:
> All binding problems fixed

while python should imho die for all versions...(also add perl and some 
others too here) ;-) :

python v2.7 is out of support since beginning of 2020

https://pythonclock.org/

N0NB, 2020-01-28:
* On 2020 28 Jan 01:46 -0600, roeland@linux-it.nl wrote:
> On 2020-01-28 00:16, Michael Black wrote:
> > All binding problems fixed
> 
> while python should imho die for all versions...(also add perl and some
> others too here) ;-) :
> 
> python v2.7 is out of support since beginning of 2020
> 
> https://pythonclock.org/

So far as I know, the Python bindngs should build with either version.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819