Hamlib/Hamlib #186: Lack of sufficent Pi documentation

nouse4anick, 2020-01-22:
- GPIO support is left out of man pages
- use examples of how to use hamlib with GPIO pins for PTT do not exist in documentation
- no raspberry pi specific build instructions in the wiki's documentation (nor in the install documentation for compiling for use on a pi)

Reason for update: many hams are turning to Pi's for cheap shack computers. Also many hams do not have the ability to control their rig through a cable (most HT's, older rigs and DIY kits) or otherwise do not want the added overhead (might be backpacking and every extra cable adds weight and a fail point). Most Digital Station setups (see the NW Digital Radio's DRAW/UDRC boards and WB7FHC's Nexus Board for examples) require the use of a GPIO pin to control the PTT on a radio (or radios). I would love to use wspr modes with my pi, but due to the lack of documentation I cannot figure out how to use hamlib to key the rig through a GPIO pin. Also with no steps in recompiling the library for use on the pi I have no way of testing to see if it works.

I'd be happy to help with testing and examples (application specific as well) if you need it.

73
Art - KC7SDA

N0NB, 2020-01-22:
It is no problem to compile hamlib on a pi, I have done so.
However I cannot tell about GPIO support, just compiling hamlib
to part of, say, fldigi.

> 
> 	• GPIO support is left out of man pages
> 	• use examples of how to use hamlib with GPIO pins for PTT do not exist in documentation
> 	• no raspberry pi specific build instructions in the wiki's documentation (nor in the install documentation for compiling for use on a pi)
> Reason for update: many hams are turning to Pi's for cheap shack computers. Also many hams do not have the ability to control their rig through a cable (most HT's, older rigs and DIY kits) or otherwise do not want the added overhead (might be backpacking and every extra cable adds weight and a fail point). Most Digital Station setups (see the NW Digital Radio's DRAW/UDRC boards and WB7FHC's Nexus Board for examples) require the use of a GPIO pin to control the PTT on a radio (or radios). I would love to use wspr modes with my pi, but due to the lack of documentation I cannot figure out how to use hamlib to key the rig through a GPIO pin. Also with no steps in recompiling the library for use on the pi I have no way of testing to see if it works.
> 
> I'd be happy to help with testing and examples (application specific as well) if you need it.
> 
> 73
> Art - KC7SDA
> 
> 
> Hamlib-developer mailing list
> Hamlib-developer@lists.sourceforge.net
> https://lists.sourceforge.net/lists/listinfo/hamlib-developer

+---------------------------------+----------------------------------+
| Prof. Christoph van Wüllen      | Tele-Phone (+49) (0)631 205 2749 |
| TU Kaiserslautern, FB Chemie    | Tele-Fax   (+49) (0)631 205 2750 |
| Erwin-Schrödinger-Str.          |                                  |
| D-67663 Kaiserslautern, Germany | vanWullen@Chemie.Uni-KL.De       |
|                                                                    |
| HomePage:  http://www.chemie.uni-kl.de/vanwullen                   |
+---------------------------------+----------------------------------+

nouse4anick, 2020-01-23:
> It is no problem to compile hamlib on a pi, I have done so.

I believe one of the images I was using had hamlib on it and it compiles just fine. What I'm after is GPIO support which (if i read the issues right) was 'solved' by adding in a flag that specifically adds the gpio support on the pi (it won't work on a desktop or laptop), however I can find no mention of how to use those flags to configure hamlib specifically for the pi or to really use the gpio pins to control ptt (I think one thread that I chimed in on said use rigctld with some options.

I have very little experience with hamlib as 90% of my radios do not support the functions or its not needed (ptt is handled through the mic connector), I would love to get this working so I can go to the rest of the hams in my community and show them 'hey this is possible without buying more stuff!' and to use the setup they already have to run wspr modes.

N0NB, 2020-01-23:
Hopefully a pi user can help more...but you need two flags.  You should be opening the COM port for the device of course.
-P GPIO -p 1
The -p is the GPIO pin # you want.
de Mike W9MDB 

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-01-23:
Just compile Hamlib on the Pi.
The GPIO module is compiled by default on all (!) platforms.

It uses „file IO“ to /sys/class/gpio to do the GPIO actions. This
is not very real-time, but more than sufficient for PTT and such
things. I compiles on non-RaspPi Linux systems, but of course does not
work there.

> 
> It is no problem to compile hamlib on a pi, I have done so.
> 
> I believe one of the images I was using had hamlib on it and it compiles just fine. What I'm after is GPIO support which (if i read the issues right) was 'solved' by adding in a flag that specifically adds the gpio support on the pi (it won't work on a desktop or laptop), however I can find no mention of how to use those flags to configure hamlib specifically for the pi or to really use the gpio pins to control ptt (I think one thread that I chimed in on said use rigctld with some options.
> 
> I have very little experience with hamlib as 90% of my radios do not support the functions or its not needed (ptt is handled through the mic connector), I would love to get this working so I can go to the rest of the hams in my community and show them 'hey this is possible without buying more stuff!' and to use the setup they already have to run wspr modes.
> 
> 
> Hamlib-developer mailing list
> Hamlib-developer@lists.sourceforge.net
> https://lists.sourceforge.net/lists/listinfo/hamlib-developer

+---------------------------------+----------------------------------+
| Prof. Christoph van Wüllen      | Tele-Phone (+49) (0)631 205 2749 |
| TU Kaiserslautern, FB Chemie    | Tele-Fax   (+49) (0)631 205 2750 |
| Erwin-Schrödinger-Str.          |                                  |
| D-67663 Kaiserslautern, Germany | vanWullen@Chemie.Uni-KL.De       |
|                                                                    |
| HomePage:  http://www.chemie.uni-kl.de/vanwullen                   |
+---------------------------------+----------------------------------+

nouse4anick, 2020-01-24:
Ok, so here is the command I use:
`sudo rigctld -P GPIO -p 12`
it doesn't return to the command prompt (I'm assuming it doesn't fork off into its own process)

I go into WSJT-X and configure the radio tab with hamlib NET rigctl as the rig, network server is: localhost:4532
set ptt method to CAT
transmit audio source as front/mic, mode and split are set to none.

I hit test cat, test ptt becomes enabled, hit test ptt and my indicator light does not light up that its keyed.

I go into fldigi -> configure -> rig control -> GPIO and double check what I have:
[image]
I have BCM 12 selected (GPIO pin 26, pin 32). Now I've tried all 3 of these, and I double checked and the version that I have on here is 3.3 (which i believe is current)

So, any ideas? I'm pretty close to getting this working with WSJT-X and if I can a LOT of people would be super happy!

73
Art

nouse4anick, 2020-01-24:
also, i just ran it in verbose  and it shows that its taking the commands from WSJT-X, what should I be looking for in terms of ptt-key/ptt-unkey? right now it shows rig_get_ptt called and dummy_get_ptt called

mdblack98, 2020-01-24:
You didn't pick a radio model or device to connect to with the -m and -r switches so you got the dummy devicee.g. to find the model# for a 7300:rigctl -l | grep 7300  373  Icom                   IC-7300                 0.24.5          Stable
You can run rigctl first to ensure you are talking to the rigUsing the model# you getrigctl -m 373 -P GPIO -p 12 -r /dev/ttyUSB0You should be able to run f to get frequency, m to get mode/bandwidth, and "T 1" "T 0" to toggle PTT.
I'm not sure of the mapping on the pin# though.  From a bit of prior experience I think if you have PTT on pin 12 that would be GPIO pin 1rigctl -m 373 -P GPIO -p 1 -r /dev/ttyUSB0

nouse4anick, 2020-01-24:
> You didn't pick a radio model or device to connect to with the -m and -r switches so you got the dummy devicee.g. to find the model# for a 7300:rigctl -l | grep 7300  373  Icom [...]

but i'm not really interested in controlling a rig, I don't care about getting the frequency either. All I'm after is getting the GPIO to trigger.

N0NB, 2020-01-24:
I didn't know you didn't have a rig....
This was from somebody else who seems to have it working....

Our fork works fine, however as it turns out, this feature was already in Hamlib, but not documented in the manpage!

Basically, what you want to do is figure out the physical GPIO pin number of the wanted PTT pin. You can do it by data-mining datasheets, or you can install wiringPi and use the gpio readall command which will print a graphic representation of the pin header on your Pi (you'll need the 'BCM' columns from it). Let's say you want pin 17. Then you can use the PTT like this:
sudo rigctld -P GPIO -p 17

Then connect to rigctld with whichever program supports it (QSSTV does, I didn't test WSJT-X).

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-01-24:
This is a bug in Win4Icom that they can easily fix.It's not responding with the rig ID when asked to do so...just times out....so it's not correctly emulating an Icom rig.All Icom rigs respond to this command and is part of new code we have added to hamlib to determine rig capabilities automatically and handle power on/off.
2020-01-24:05:51:19.999424: write_block(): TX 7 bytes2020-01-24:05:51:19.999424: 0000    fe fe 94 e0 19 00 fd                                .......         2020-01-24:05:51:19.999424: read_string called2020-01-24:05:51:20.010425: read_string(): RX 7 characters2020-01-24:05:51:20.010425: 0000    fe fe 94 e0 19 00 fd                                .......         2020-01-24:05:51:20.010425: read_string called2020-01-24:05:51:21.040824: read_string(): Timed out 1.30399 seconds after 0 chars
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

nouse4anick, 2020-01-24:
Ok, to re-iterate my goals in doing ptt with GPIO:
- not everyone has a rig that can talk to a computer
- not everyone wants to use the same rig day after day, or in the same location
- ptt with the GPIO means that your rig interface is now UNIVERSAL among all of your radios!

So, 99% of my radios cannot be controlled by a computer period, this makes a majority of hamlib useless to me, however I can still connect them to my pi with the proper cables for audio and ptt (nw digital radio's DRAWS/UDRC boards and the Nexus board from WB7FHC for example). With a cable swap I can use ANY of my rigs for digital. I do not have to change the settings in fldigi either! This is what I'm trying to accomplish with hamlib and WSJT-X. Getting this to work will mean a lot of hams with similar setups will be able to use WSJT-X software (or any software that can use hamlib) on any radios (including really old ones!).

mdblack98, 2020-01-24:
So did you get the GPIO to work?
Mike

N0NB, 2020-01-25:
If the rig cannot talk with hamlib, then most likely
Hamlib’s rig_open() fails, and then most probably hamlib
cannot control ptt (because it does not have a rig).

So I think the only way to try out is to use the „Hamlib dummy“
rig and set GPIO PTT as an option before opening it.

> Am 25.01.2020 um 00:07 schrieb Michael Black <notifications@github.com>:
> 
> So did you get the GPIO to work?
> Mike
> 
> 

+---------------------------------+----------------------------------+
| Prof. Christoph van Wüllen      | Tele-Phone (+49) (0)631 205 2749 |
| TU Kaiserslautern, FB Chemie    | Tele-Fax   (+49) (0)631 205 2750 |
| Erwin-Schrödinger-Str.          |                                  |
| D-67663 Kaiserslautern, Germany | vanWullen@Chemie.Uni-KL.De       |
|                                                                    |
| HomePage:  http://www.chemie.uni-kl.de/vanwullen                   |
+---------------------------------+----------------------------------+

nouse4anick, 2020-01-25:
> So did you get the GPIO to work? Mike

yes and no.... the program didn't error out on me when I went to test the ptt, and the rigctrld program didn't exit with an error so its trying to do something.... however it didn't actually key anything that I can tell.

I am testing this on a NEXUS hat (http://wb7fhc.com/nexus-dr-x.html) which has indicator lights for which rig is keying (left rig or right rig). neither light actually lights when I test it in the WSJT-X program. However when testing with FLDIGI (see pic above for settings) it lights up no problem. I've tried all 3 pin designations (actual pin number, the BCM number, and the GPIO number) and none of them trigger.

I'm hoping someone who actually worked on that particular portion of the code can tell me how they designed it, or if there's a way to test it (i'll do some digging on my own but i'm getting ready for Winter Field Day) without using WSJT-X. Or if I have to pass a rig to the program to get the key to work (maybe a bug?).

nouse4anick, 2020-01-25:
here is a log that i created with -vvvv:

```
rigctld, Hamlib 3.3
Report bugs to <hamlib-developer@lists.sourceforge.net>
rig_init called
dummy: _init called
rig_register called
rig_register: rig_register (1)
rig_register called
rig_register: rig_register (2)
rig_register called
rig_register: rig_register (4)
rig_register called
rig_register: rig_register (5)
dummy_init called
rig_strvfo called
rig_passband_normal called
rig_strvfo called
rig_passband_normal called
rig_open called
port_open called
dummy_open called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
rig_strstatus called
Backend version: 0.5, Status: Beta
rig_close called
dummy_close called
port_close called
Connection opened from localhost:46634
client lock engaged
rig_open called
port_open called
dummy_open called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
client lock disengaged
client lock engaged
rig_strvfo called
client lock disengaged
client lock engaged
rig_strvfo called
client lock disengaged
client lock engaged
rig_strvfo called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
rig_strvfo called
client lock disengaged
client lock engaged
rig_strvfo called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
rig_strvfo called
client lock disengaged
client lock engaged
rig_strvfo called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
rig_strvfo called
client lock disengaged
client lock engaged
rig_strvfo called
rig_get_mode called
[... 253 lines cut ...]
rig_strvfo called
rig_get_mode called
rig_strvfo called
dummy_get_mode called: currVFO
rig_strrmode called
client lock disengaged
client lock engaged
rig_strvfo called
rig_get_ptt called
dummy_get_ptt called
client lock disengaged
select
client lock engaged
1 outstanding client(s)
rig_close called
dummy_close called
port_close called
client lock disengaged
rig_cleanup called
dummy_cleanup called
Opened rig model 1, 'Dummy'
Closed rig model 1, 'Dummy - will reopen for clients'
Opened rig model 1, 'Dummy'
```
I ran rigctrld as above, went over to WSJT-X, hit 'test cat' then 'test ptt' then exited rigctrld

N0NB, 2020-01-25:
Add one more v to that tests (5 total).
Mike

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

nouse4anick, 2020-01-25:
here's with 5 -v:
rigctld, Hamlib 3.3
Report bugs to <hamlib-developer@lists.sourceforge.net>

```
rig_init called
dummy: _init called
rig_register called
rig_register: rig_register (1)
rig_register called
rig_register: rig_register (2)
rig_register called
rig_register: rig_register (4)
rig_register called
rig_register: rig_register (5)
dummy_init called
rig_strvfo called
rig_passband_normal called
rig_strvfo called
rig_passband_normal called
rig_open called
port_open called
dummy_open called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
rig_strstatus called
Backend version: 0.5, Status: Beta
rig_close called
dummy_close called
port_close called
Connection opened from localhost:46654
client lock engaged
rig_open called
port_open called
dummy_open called
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
client lock disengaged
client lock engaged
rig_strvfo called
rigctl(d): ð 'currVFO' '' '' ''
client lock disengaged
client lock engaged
rig_strvfo called
rigctl(d):  'currVFO' '' '' ''
client lock disengaged
client lock engaged
rig_strvfo called
rigctl(d): v 'currVFO' '' '' ''
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
rig_strvfo called
client lock disengaged
client lock engaged
rig_strvfo called
rigctl(d): v 'currVFO' '' '' ''
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
rig_strvfo called
client lock disengaged
client lock engaged
rig_strvfo called
rigctl(d): v 'currVFO' '' '' ''
rig_get_vfo called
rig_strvfo called
dummy_get_vfo called: VFOA
[... 265 lines cut ...]
rig_get_mode called
rig_strvfo called
dummy_get_mode called: currVFO
rig_strrmode called
client lock disengaged
client lock engaged
rig_strvfo called
rigctl(d): t 'currVFO' '' '' ''
rig_get_ptt called
dummy_get_ptt called
client lock disengaged
select
client lock engaged
1 outstanding client(s)
rig_close called
dummy_close called
port_close called
client lock disengaged
rig_cleanup called
dummy_cleanup called
Opened rig model 1, 'Dummy'
Closed rig model 1, 'Dummy - will reopen for clients'
Opened rig model 1, 'Dummy'

```

mdblack98, 2020-01-25:
Now let's try this with rigctl as Bill recommended.
And please show your command line every time......
Once you run rigctl to the "T 1" "T 0" and the GPIO pin should toggle if you have the right one selected.
I've got a Pi in the closet I could probably yank out and test this myself.
MIke

nouse4anick, 2020-01-26:
> Now let's try this with rigctl as Bill recommended. And please show your command line every time...... Once you run rigctl to the "T 1" "T 0" and the GPIO pin should toggle if you have the right one selected. I've got a Pi in the closet I could probably yank out and test this myself. MIke

The bands went dead so I'm back to tinkering with this... the command i ran was:
`sudo rigctld -P GPIO -p 12 -m 1 -vvvvv`
for the first tests
This last time I tried rigctl with the following:
`sudo rigctl -P GPIO -p 12 -m 1 -vvvvv`
then it prompted for a command, typed 'T 1' and sure enough the light on my Nexus board lit up!

so setting it manually works, however rigctld either doesn't pass the correct argument or its WSJT-X's fault in not calling the right argument.

any ideas would be helpful... we are so close to getting a universal setting that will work with all radios and not just computer controlled ones!

N0NB, 2020-01-26:
Try my fork now...rigctld didn't know about GPIO
https://github.com/mdblack98/Hamlib.git

de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

nouse4anick, 2020-02-18:
> 
> Try my fork now...rigctld didn't know about GPIO https://github.com/mdblack98/Hamlib.git de Mike W9MDB 

I finally have a bit of time, I tried to build the other library but i got this:
```
[bunch of other stuff up here]
checking for alloca... yes
./configure: line 7701: syntax error near unexpected token `2.2.6b'
./configure: line 7701: `LT_PREREQ(2.2.6b)'
```
now it did fail on the bootstrap config:
```
rotators/ts7400/Makefile.am:2: error: Libtool library used but 'LIBTOOL' is undefined
rotators/ts7400/Makefile.am:2:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
rotators/ts7400/Makefile.am:2:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
rotators/ts7400/Makefile.am:2:   If 'LT_INIT' is in 'configure.ac', make sure
rotators/ts7400/Makefile.am:2:   its definition is in aclocal's search path.
src/Makefile.am:10: error: Libtool library used but 'LIBTOOL' is undefined
src/Makefile.am:10:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/Makefile.am:10:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/Makefile.am:10:   If 'LT_INIT' is in 'configure.ac', make sure
src/Makefile.am:10:   its definition is in aclocal's search path.
autoreconf: automake failed with exit status: 1
```
according to apt I do have libtool... version 2.4.6-9 but I have no idea why it isn't picking it up.
btw, a little bit of background on me: I did go to school to be a developer, however now I'm a machine operator (training to be a machinest), I do code in my spare time however I was never really any good at the autoconfigure stuff to link multiple libraries across multiple platforms.

so, any chance someone can give me a step by step on how to fix this?
my release info:
```
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```
and the kernal version:
```
Linux KC7SDA-hampi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
```
please note this is an image from the local ham group I downloaded for one of their digital projects and I have no idea just how heavy they modified the kernel or everything they installed (I use very little as I just do fldigi right now)

link to their image if you want to see the project and what all they have done:
https://github.com/AG7GN/images

N0NB, 2020-02-18:
What does "which libtool" say?
And presumably "/usr/bin/libtool --version" ?
de Mike W9MDB

 
 

https://github.com/AG7GN/images

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

nouse4anick, 2020-02-19:
> 
> What does "which libtool" say? And presumably "/usr/bin/libtool --version" ? de Mike W9MDB 

so funny thing i get nothing with the which command and the only reference for libtool i can find is in /usr/share/libtool which is a directory and it looks like its the gnu libltdl (system independent dlopen wraper for libtool).

so right now i have no idea where to go... talking with the guy that's come up with the image i'm using (hampi), he's using the stock raspbian buster with no kernel tweaks... he's just installed a bunch of stuff and setup menus and runs a script at startup... he also reports the same error on his end, but the official hamlib 3.3 builds fine.

have the edits from the fork made it into the official package yet? any idea when that might be?

N0NB, 2020-02-19:
* On 2020 18 Feb 18:43 -0600, Art Miller wrote:
> so funny thing i get nothing with the which command and the only reference for libtool i can find is in /usr/share/libtool which is a directory and it looks like its the gnu libltdl (system [...]

What does 'apt policy libtool' report?

Here I get:

$ apt policy libtool
libtool:
  Installed: 2.4.6-9
  Candidate: 2.4.6-9
  Version table:
 *** 2.4.6-9 500
        500 http://deb.debian.org/debian buster/main amd64 Packages
        500 http://deb.debian.org/debian buster/main i386 Packages
        100 /var/lib/dpkg/status

HTH,

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

nouse4anick, 2020-02-19:
> What does 'apt policy libtool' report?

same as yours

new update: the guy that developed the hampi image for our club says that there are missing libtool related .m4 files from the fork.  what else aside from 'build-essential' is needed?

N0NB, 2020-02-19:
* On 2020 18 Feb 20:13 -0600, Art Miller wrote:
> > What does 'apt policy libtool' report?
> 
> same as yours
> 
> new update: the guy that developed the hampi image for our club says
> that there are missing libtool related .m4 files from the fork.  what
> else aside from 'build-essential' is needed?

Good question.  Perhaps he needs to back up and make sure the complete
libtool package as provided by Raspbian/Debian is provided.  As far as I
am concerned, libtool is a black box!  I've a pretty good handle on
autoconf and automake but have tended to leave libtool alone.  ;-)

If you're curious, here is the list of files included in the libtool
package:

https://packages.debian.org/buster/all/libtool/filelist

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

mdblack98, 2020-02-19:
hamlib only includes the macros it uses plus the macros that are referenced by those macros....not all m4 macros.

# macros/ax_cflags_warn_all.m4
# macros/gr_pwin32.m4
# macros/ax_pthread.m4
# macros/hl_getaddrinfo.m4
    # macros/ax_lib_readline.m4
dnl SC_PATH_PERLINC from macros/perl.m4
dnl AX_PYTHON_DEVEL from macros/ax_python_devel.m4
dnl SC_PATH_TCLCONFIG and SC_LOAD_TCLCONFIG from macros/tcl.m4
dnl AX_LUA_DEVEL from macros/ax_lua_devel.m4
    [# macros/ax_pkg_swig.m4
macros/Makefile

I think I have a Raspberry Pi in my archives that I can test this on.

Mike

same as yours

new update: the guy that developed the hampi image for our club says that there are missing libtool related .m4 files from the fork. what else aside from 'build-essential' is needed?

N0NB, 2020-02-19:
Compile and install libtool yourself and see if that fixes the problem. 

https://www.gnu.org/software/libtool/

de Mike W9MDB

so funny thing i get nothing with the which command and the only reference for libtool i can find is in /usr/share/libtool which is a directory and it looks like its the gnu libltdl (system independent dlopen wraper for libtool).

so right now i have no idea where to go... talking with the guy that's come up with the image i'm using (hampi), he's using the stock raspbian buster with no kernel tweaks... he's just installed a bunch of stuff and setup menus and runs a script at startup... he also reports the same error on his end, but the official hamlib 3.3 builds fine.

have the edits from the fork made it into the official package yet? any idea when that might be?

—

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

nouse4anick, 2020-02-19:
> hamlib only includes the macros it uses plus the macros that are referenced by those macros....not all m4 macros.

so if you pull the official repository it has a lot more .m4 files than your fork

> Compile and install libtool yourself and see if that fixes the problem. 
> 
> https://www.gnu.org/software/libtool/
> 
> de Mike W9MDB

to be honest I really shouldn't have to compile and install libtool... I really need a much simpler 'download this, do a few things, you can now get on wsjt-x with no additional interface!' type solution.

is there any chance of getting hamlib patched before the official release? or could I just copy and paste all the files from the fork into the hamlib repository directory and try compiling that?

N0NB, 2020-02-20:
On 19/02/2020 23:50, Art Miller wrote:
> hamlib only includes the macros it uses plus the macros that are
>     referenced by those macros....not all m4 macros.
>
> so if you pull the official repository it has a lot more .m4 files
> than your [...]
Hi Art,

have you tried with a clean checkout and build tree? If your sources are 
configured with older versions of automake etc. then you may have 
references to outdated m4 scripts.

73
Bill
G4WJS.

mdblack98, 2020-02-20:
There's nothing to fix in hamlib...it compiles everywhere without a problem.The only thing we can think of is your libtool package for the Pi is lacking something...so compiling libtool may be the only solution.
This is what happens when you compile programs...you sometimes find your support packages are inadequate.
de Mike W9MDB

 
 

https://www.gnu.org/software/libtool/

N0NB, 2020-02-20:
* On 2020 19 Feb 17:51 -0600, Art Miller wrote:
> to be honest I really shouldn't have to compile and install libtool...
> I really need a much simpler 'download this, do a few things, you can
> now get on wsjt-x with no additional interface!' type solution.

I agree.  Did you compare the files on your system to those in the
Debian link I sent last night?

> is there any chance of getting hamlib patched before the official
> release? or could I just copy and paste all the files from the fork
> into the hamlib repository directory and try compiling that?

As Mike stated, working from a Git tree can be troublesome as sometimes
you need to use 'git clean -dxf' to remove any and all generated files
in the working tree and completely delete everything under the build
directory (if you're using a VPATH) build.

The simplest method is to use the daily snapshots as they're already
set up so that libtool or any of the other GNU Autotools are not
required to be installed on your system.  However, if you're testing
from Mike's tree this won't help.  Better then just to do a clean 'git
clone' and working from there.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

nouse4anick, 2020-02-20:
well that really does suck... I'm bootstrapping libtool now and will see if it it works.

I really really hope the gpio issue gets fixed on the next release.

nouse4anick, 2020-02-21:
ok... so i got libtool installed:

```
pi@KC7SDA-hampi:~/Hamlib $ which libtool
/usr/local/bin/libtool
pi@KC7SDA-hampi:~/Hamlib $ /usr/local/bin/libtool --version
libtool (GNU libtool) 2.4.6.44-b9b4
Written by Gordon Matzigkeit, 1996

Copyright (C) 2014 Free Software Foundation, Inc.
```

however its still giving me errors that it can't find libtool. I've changed line 268 to point to libtool... but still getting the same error as before.

line 268 now looks like this:
```
LT_INIT("/usr/local/bin/libtool")
```
should i be pointing it at the build directory instead? is there another directory it should look at?

and yes, i did run aclocal before bootstrapping it again.

Thanks for the help guys

nouse4anick, 2020-02-23:
[snip]
> I agree. Did you compare the files on your system to those in the Debian link I sent last night?

I'm compairing them now, i might be missing a few (not sure how important they are):
/usr/share/aclocal/lt~obsolete.m4 - missing
not checking the docs, but everything else seems to be there

> As Mike stated, working from a Git tree can be troublesome as sometimes you need to use 'git clean -dxf' to remove any and all generated files in the working tree and completely delete everything [...]

I cleaned the directory, i'll try a clone here in a minute. I might for S&G grab the offical one and see if i can get that one to compile. if i do a git clone on the main hamlib is that the daily snapshot?

Thanks 73
Art

nouse4anick, 2020-02-23:
fyi i just pulled the hamlib main branch into a directory and tried to bootstrap it... got the same error with libtool... 

I've just about hit the edge of my knowledge in development on this... I have no idea what autoconfig is looking for as far as the library goes, i can keep keyboard bashing and see if i get something at some point but i'd rather not waste time and just try to find something that works.

also, I noticed something with the main branch: it will configure ar-lib, compile, config.guess .... it however fails on the amplifiers make file, bindings make file, c++ and so on... I'll look into some of those and see if i can figure out what it's looking for. I'll see if i can point to a few other directories and see if i get any further...

73
Art

ps it looks like parallel-tests worked successfully (installed test-driver)

N0NB, 2020-02-23:
The git clone might be slightly ahead (up to 1 day) than the daily.  But the master doesn't typically get updated that often.
You may as well just work with the clone as all you have to do is a "git pull" to update it.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-02-23:
Try my fork
Not that it will fix things but if there's a change needed that's where we'll test it.  I've got a bunch of patches that are almost ready to submit to the master right now.
https://github.com/mdblack98/Hamlib.git

de Mike W9MDB 

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

nouse4anick, 2020-02-23:
> 
> Try my fork Not that it will fix things but if there's a change needed that's where we'll test it.  I've got a bunch of patches that are almost ready to submit to the master right now. https://github.com/mdblack98/Hamlib.git de Mike W9MDB

So I've tried your fork mike and that got me the same thing as the main hamlib branch. I have found a few interesting things though:
- I ran aclocal --verbose and dumpped the output to a log file. I went through it and didn't like what I saw for the libtool lines that I found, I'm going to attempt to attach it to this post
- I dumped the output of the bootstrap to a log and I'm attaching that too. 
[aclocal.log](https://github.com/Hamlib/Hamlib/files/4241724/aclocal.log)
[bootstrap.log](https://github.com/Hamlib/Hamlib/files/4241725/bootstrap.log)

I'll go pull on that fork mike and i'll do the same thing with it as is and upload those logs as well

nouse4anick, 2020-02-23:
Here are the 2 logs from Mike's fork (i did a pull about 5 minutes ago on it)
[w9mdb-bootstrap.log](https://github.com/Hamlib/Hamlib/files/4241757/w9mdb-bootstrap.log)
[w9mdb-aclocal.log](https://github.com/Hamlib/Hamlib/files/4241758/w9mdb-aclocal.log)

nouse4anick, 2020-02-23:
Also I went through the readme.release and developer files and double checked to see that I had everything installed and setup correctly on it.

N0NB, 2020-02-23:
What's the date on this file?
/usr/share/aclocal/pkg.m4

As opposed to this one?
/usr/local/share/aclocal/pkg.m4

You apparently didn't uninstall the old libtool.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

nouse4anick, 2020-02-23:
i uninstalled it with
`sudo apt-get remove libtool`
and its still throwing off the errors like before.

also, silly question about git, but if i do 
`git pull`
it should grab all the files its missing and 'update' correct? i tried deleting the config file to force it to pull a fresh one but it said 'up to date'.... am i missing a flag or something?

mdblack98, 2020-02-23:
Manually remove /use/share/aclocal
And yes, git pull will 
grab any new commits.
Mike

nouse4anick, 2020-02-23:
ok so now it fails with 
```
pi@KC7SDA-hampi:~/Hamlib-mdblack98 $ ./bootstrap
Running 'autoreconf -i' to process configure.ac
and generate the configure script.
aclocal: error: couldn't open directory '/usr/share/aclocal': No such file or directory
autoreconf: aclocal failed with exit status: 1
```
when i try to run aclocal it says it can't find that directory. 

also i'm not looking for new commits, i was looking more for 'hey a file is missing go grab it again'

thanks again guys for all your help so far, I think i'm making progress but its so frustrating when this isn't telling me where stuff is defined.

Art

mdblack98, 2020-02-23:
I’m out right now but try and find source for aclocal and compile/bin stall that

nouse4anick, 2020-02-23:
well i looked into recompiling aclocal... and its a huge ass pain to do that....

i did a little investigation and found theres a aclocal-1.16 directory, so i created a symbolic link to that directory for aclocal... i got aclocal to work again, however libtool is still missing/not defined

N0NB, 2020-02-24:
Art,

Malcolm Hering suggested that you try a fresh installation of Raspbian
and try the build again.  Since what you're using you've described as a
fork of Raspbian, we need to be sure where the issue lies.  You may not
have seen his reply since I think it only went to the hamlib-developer
mailing list.

Please set this installation aside and try with a fresh installation of
Raspbian.  I beg of you to do this before you go any further down this
path of deleting system installed files and such.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

nouse4anick, 2020-02-24:
so far i haven't deleted anything, i moved the orginal aclocal to aclocal.bkup =)

also, i'm guessing that in a few months or so I'll be redoing the image anyway with a new one they come out with thats clean(er). Also I'll have to try working up a solution to get to the rest of the club for those interested in doing wspr modes w/o the interface cable to the rig that doesn't require all the moving/compiling/etc. I hope to just compile a deb package that i can have the guy upload to the site and have a script download/install it (which would be the best option for my local group until the base hamlib gets updated to work with the gpio)
i'll go ahead and grab a fresh raspbian image and toss that on a card. hopefully that'll fix everything.
Art

malcolmh, 2020-02-24:
Art, sorry - I did not spot that this thread was on Github. I suggested that something must be broken on your Rasbian system. I did the following test:

From a clean install of the latest Raspbian Buster:

sudo apt install automake libtool
git clone https://github.com/Hamlib/Hamlib.git
cd Hamlib/
mkdir build
cd build/
../bootstrap
../configure
make

All ran successfully.

N0NB, 2020-02-26:
Art, try installing the `libtool-bin` package.  I was in a hurry the other day and did not notice the file list for the `libtool` does not include the `/usr/bin/libtool` file.  I'm not sure why I've never noticed this before.  Sigh...

N0NB, 2020-02-26:
Well, Art, I don't know what is going on with your system.

I set up a Raspbian Buster virtual machine and have installed `autoconf`, `automake`, and `libtool`, but not `libtool-bin`.  I also installed `git` and cloned Mike's current master repository and built it in the usual way with no errors except for running out of space on the virtual disk image which I can't seem to enlarge.  Oh well.

nouse4anick, 2020-02-27:
> Art, try installing the `libtool-bin` package. I was in a hurry the other day and did not notice the file list for the `libtool` package does not include the `/usr/bin/libtool` file. It is in the `libtool-bin` package instead. I'm not sure why I've never noticed this before. Sigh...

so I'm not sure how to get libtool-bin installed. I'm going to just see about doing this on another setup and exporting it somehow...

if I was able to compile everything and save the entire directory and gave it to someone with the same setup they should be able to run sudo make install on it without recompiling it correct? my thought is that if they already have hamlib installed then all supporting libraries are already installed.

thanks again for the help guys, i'm going to work on this more this coming weekend and see if i can figure it out... I might talk to the guy who made the hampi image and see if he has a 'setup script' that I can run on a blank raspbian image, and if i can get the image to compile hamlib then i'll do that and run his script and make a new main image for me.

N0NB, 2020-02-27:
* On 2020 26 Feb 20:52 -0600, Art Miller wrote:
> so I'm not sure how to get libtool-bin installed.

May I suggest a Raspbian or Debian tutorial?  Updating packages is
actually quite simple using these terminal commands:

	sudo apt update
	sudo apt upgrade

These two will ensure your system is up-to-date for security fixes and
other bug fixes in various packages.  The first time you do this the
list will likely be quite large and take some time.  When done more
frequently, the updates may only involve one or a handful of packages.

To install a given package, in this case libtool-bin:

	sudo apt install libtool-bin

However, you don't need that package installed to bootstrap hamlib.  I
need to correct the README.developer file on that package.

To find a specific package:

	apt search "midnight commander"

will return a list of packages with those words including the "mc"
package which is the Swiss Army Chainsaw of file managers.  I've been
using it since first starting with Slackware Linux in the autumn of
1996.

Another packager manager I have used for at least 20 years is "aptitude".
It may not be installed by default and it is actually fairly slow but it
has a TUI that is reasonably easy to navigate.

> I'm going to just see about doing this on another setup and exporting it somehow...

Do you have a link where I can download the image you're using and test
it for myself?

> if I was able to compile everything and save the entire directory and
> gave it to someone with the same setup they should be able to run sudo
> make install on it without recompiling it correct? my [...]

Yes, if it's the exact same system and both are up-to-date.

One key thing to be aware of when compiling locally and using the
default installation directory (/usr/local) is to run 'sudo ldconfig' so
the runtime linker can find the libraries.  It doesn't need to be run
every time though it doesn't hurt.

> thanks again for the help guys, i'm going to work on this more this
> coming weekend and see if i can figure it out... I might talk to the
> guy who made the hampi image and see if he has a 'setup [...]

Again, if you can point me to an image I can download then I can try it
and see if I get the same failure.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

nouse4anick, 2020-03-06:
so the main reason I haven't updated the packages is because the last rasbian update broke the audio driver we were using for the fe-pi, they apparently finally patched it and it should be loaded soon.

here's the link to the image we are using: https://github.com/AG7GN/images

i'll try running sudo ldconfig and see what that does, I'm sorry I haven't had a chance to work on this lately I got hit with a nasty bug that took me out of commission for 3 days and I'm finally getting caught up on everything.

73
Art

mdblack98, 2020-03-11:
What's the status of this?

N0NB, 2020-03-11:
Hi Guys

Can you please unsubscribe me from this mailing list?
Tried to unsubscribe myself but,  the Unsubscribe Link at the bottom of the
email does NOT seem to work :-)

Kind regards,
Pierre

nouse4anick, 2020-03-12:
> 
> 
> What's the status of this?

well I'm going to try a few things hopefully this weekend:
- I'm going to try running that ldconfig and then try compiling the library again, if that doesn't work then....
- I'm going to download raspbian and try a 'fresh start' and see what happens
- I'll download the newest copy of the hampi image (update it) and try it if all else fails

If all of that fails then I am lost as to why this isn't working for me. At that point I'll give up and wait for the official release and wait for it to be incorporated into raspbian.

nouse4anick, 2020-03-14:
so first test failed... sudo ldconfig apparently doesn't do anything (i did sudo ldconfig -v | grep libtool and it didn't show anything for libtool)

I have to shut down for now though, we just had 3 power flickers back to back (we have up to 60 mph gusts going on right now). I'll try again later when it calms down.

N0NB, 2020-03-27:
I purchased a Pi 4 that arrived today.  I downloaded AG7GN's hampi20191214 image, cloned Mike's current repository, and despite the Autotools being installed, at least libtool is broken.  I did not update the system to mimic Art's environment as much as I could.

I've now tried it with stock Raspbian Buster and after installing autoconf, automake, and libtool, the bootstrap step runs as expected, the configure script is generated, configure runs as expected, and the build proceeds as expected.  Do note that Raspbian updates the system as part of the initial setup so all packages are up-to-date, but I recall no issues with Debian Buster (upon which Raspbian is based) through the nine months that I ran it as "testing" before it became "stable" last summer.  Since then the builds have just worked.

Somehow AG7GN's image is broken.  Stock Raspbian is fine.  Sorry, I'm of no further help.

[2 more replies]

closed 2020-06-22 by nouse4anick