Hamlib/Hamlib #189: Icom 7300 Segmentation fault

hlnurse, 2020-02-03:
Using the latest branch, Raspberry Pi 3B+ and 4B, Buster.  Rigctl or rigctld.

Thanks!

--Howard W6HN

rig_init called
initrigs4_icom: _init called
rig_register called
rig_register: rig_register (355)
rig_register called
rig_register: rig_register (309)
rig_register called
rig_register: rig_register (310)
rig_register called
rig_register: rig_register (311)
rig_register called
rig_register: rig_register (313)
rig_register called
rig_register: rig_register (314)
rig_register called
rig_register: rig_register (315)
rig_register called
rig_register: rig_register (319)
rig_register called
rig_register: rig_register (320)
rig_register called
rig_register: rig_register (321)
rig_register called
rig_register: rig_register (322)
rig_register called
rig_register: rig_register (367)
rig_register called
rig_register: rig_register (323)
rig_register called
rig_register: rig_register (346)
rig_register called
rig_register: rig_register (324)
rig_register called
rig_register: rig_register (328)
rig_register called
rig_register: rig_register (330)
rig_register called
rig_register: rig_register (326)
rig_register called
rig_register: rig_register (327)
rig_register called
rig_register: rig_register (347)
rig_register called
rig_register: rig_register (357)
rig_register called
rig_register: rig_register (363)
rig_register called
rig_register: rig_register (329)
rig_register called
rig_register: rig_register (362)
rig_register called
rig_register: rig_register (345)
rig_register called
rig_register: rig_register (356)
rig_register called
rig_register: rig_register (375)
rig_register called
rig_register: rig_register (360)
rig_register called
rig_register: rig_register (370)
rig_register called
rig_register: rig_register (361)
rig_register called
rig_register: rig_register (373)
rig_register called
rig_register: rig_register (378)
rig_register called
rig_register: rig_register (331)
rig_register called
rig_register: rig_register (312)
rig_register called
rig_register: rig_register (316)
rig_register called
rig_register: rig_register (332)
rig_register called
rig_register: rig_register (334)
rig_register called
rig_register: rig_register (344)
rig_register called
rig_register: rig_register (368)
rig_register called
rig_register: rig_register (335)
rig_register called
rig_register: rig_register (381)
rig_register called
rig_register: rig_register (369)
rig_register called
rig_register: rig_register (377)
rig_register called
rig_register: rig_register (336)
rig_register called
rig_register: rig_register (358)
rig_register called
rig_register: rig_register (380)
rig_register called
rig_register: rig_register (337)
rig_register called
rig_register: rig_register (338)
rig_register called
rig_register: rig_register (339)
rig_register called
rig_register: rig_register (340)
rig_register called
rig_register: rig_register (341)
rig_register called
rig_register: rig_register (342)
rig_register called
rig_register: rig_register (379)
rig_register called
rig_register: rig_register (343)
rig_register called
rig_register: rig_register (366)
rig_register called
rig_register: rig_register (303)
rig_register called
rig_register: rig_register (304)
rig_register called
rig_register: rig_register (306)
rig_register called
rig_register: rig_register (307)
rig_register called
rig_register: rig_register (302)
rig_register called
rig_register: rig_register (352)
rig_register called
rig_register: rig_register (353)
rig_register called
rig_register: rig_register (351)
rig_register called
rig_register: rig_register (364)
rig_register called
rig_register: rig_register (365)
rig_register called
rig_register: rig_register (354)
rig_register called
rig_register: rig_register (383)
rig_register called
rig_register: rig_register (384)
rig_register called
rig_register: rig_register (382)
rig_register called
rig_register: rig_register (371)
rig_register called
rig_register: rig_register (372)
rig_register called
rig_register: rig_register (374)
rig_register called
rig_register: rig_register (376)
icom_init called
Segmentation fault

mdblack98, 2020-02-04:
Please try my fork....needs some testing before I push it the main branch.  I think this is fixed.
https://github.com/mdblack98/Hamlib

de Mike W9MDB

hlnurse, 2020-02-04:
Thank you, Mike.  Initial testing looks good, no more seg fault with IC-7300.

--Howard W6HN

hlnurse, 2020-02-06:
Mike, the fix for the seg fault on the 7300 looks fine.  I won't close it in case you want additional feedback.

--Howard W6HN

mdblack98, 2020-02-08:
Fixed

closed 2020-02-08 by mdblack98