Hamlib/Hamlib #192: FT-991 S-meter calibration

hlnurse, 2020-02-06:
The FT-991 S-meter needs help.

S-0 = -54 (correct)
S-9 = 1280 (not correct)
S-9+25 = 1620 (not correct)

Thanks!

--Howard W6HN

N0NB, 2020-02-06:
The code seems to expect an S# response but every Yaesu rig I look at says 0-255.So has this really been broken forever?
I'll have to review all the Yaesu rigs to see if it changed behavior somewhere.
Can you measure S0 through S9+60 and tell me the rig answer using the w command?
w SM0
de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

hlnurse, 2020-02-08:
Hi Mike,

Here are the results.  They are taken from a spreadsheet, so the formatting is wonko, but I'm sure you can see what is happening.

--Howard

S Meter Reading	“w” Command Value	Hamlib Value	% Full Scale
			
S0	SM0000	-90	0
S2	SM0026	170	10
S4	SM0051	420	20
S6	SM0081	720	30
S7.5	SM0105	960	40
S9	SM0130	1270	50
+12db	SM0157	1480	60
+25db	SM0186	1730	70
+35db	SM0203	1970	80
+50db	SM0237	2280	90
+60db	sm0255	2460	100

N0NB, 2020-02-09:
Try my fork and see if it behaves better.That data looks pretty linear.
https://github.com/mdblack98/Hamlib

de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

hlnurse, 2020-02-10:
Hi Mike,

The new fork is returning -54 to 201 (255 range) for the FT-991 using rigctl STRENGTH.  Is this the correct range?  (I would expect -54 to 46, a range of 100.)

—Howard W6HN

N0NB, 2020-02-10:
Try again...helps if I multiply instead of divide.....
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

hlnurse, 2020-02-11:
Bingo!  Thanks, Mike, I've confirmed that the FT-991 S-meter is now functioning correctly.  This issue can be closed.  Divided we fall, multiplying we rise!

--Howard

closed 2020-02-11 by mdblack98