Hamlib/Hamlib #194: Warnings from running swig -java on hamlib.swg

malcolmh, 2020-02-08:
The following warnings  are output from swig:

../Hamlib/bindings/hamlib.swg:181: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/hamlib.swg:182: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/hamlib.swg:183: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/hamlib.swg:184: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/rig.swg:273: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/rotator.swg:137: Warning 518: Portability warning: File src/hamlib/rig.java will be overwritten by src/hamlib/Rig.java on case insensitive filesystems such as Windows' FAT32 and NTFS unless the class/module name is renamed
../Hamlib/bindings/rotator.swg:137: Warning 518: Portability warning: File src/hamlib/rot.java will be overwritten by src/hamlib/Rot.java on case insensitive filesystems such as Windows' FAT32 and NTFS unless the class/module name is renamed

Also, in order to compile the library, I had to comment out line 36 (#include "misc.h") from hamlib.swg. This resulted in an additional warning from swig:

cstring.i not implemented for this target

mdblack98, 2020-03-11:
What's the status of this?  Some warnings are expected though I'm unsure why you had to comment out misc.h.   What version of swig are you using and can you compile/install the latest to see if that helps?  I did update the minimum swig required.

malcolmh, 2020-03-11:
Merged current Hamlib master

SWIG version 4.0.1

$ swig -java -I../Hamlib/include/ -outcurrentdir -package hamlib -outdir src/hamlib -o jni/hamlib_wrap.c ../Hamlib/bindings/hamlib.swg
cstring.i not implemented for this target
../Hamlib/bindings/hamlib.swg:181: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/hamlib.swg:182: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/hamlib.swg:183: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/hamlib.swg:184: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/include/hamlib/rig.h:1647: Warning 451: Setting a const char * variable may leak memory.
../Hamlib/include/hamlib/rotator.h:223: Warning 451: Setting a const char * variable may leak memory.
../Hamlib/include/hamlib/rotator.h:259: Warning 451: Setting a const char * variable may leak memory.
../Hamlib/bindings/rig.swg:275: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/rotator.swg:137: Warning 518: Portability warning: File src/hamlib/rig.java will be overwritten by src/hamlib/Rig.java on case insensitive filesystems such as Windows' FAT32 and NTFS unless the class/module name is renamed
../Hamlib/bindings/rotator.swg:137: Warning 518: Portability warning: File src/hamlib/rot.java will be overwritten by src/hamlib/Rot.java on case insensitive filesystems such as Windows' FAT32 and NTFS unless the class/module name is renamed

malcolmh, 2020-03-11:
$ gcc -fPIC -I/usr/lib/jvm/java-8-openjdk-amd64/include/ -I/usr/lib/jvm/java-8-openjdk-amd64/include/linux/ -O0 -g3 -Wall -c -o jni/hamlib_wrap.o jni/hamlib_wrap.c
jni/hamlib_wrap.c:255:10: fatal error: misc.h: No such file or directory
 #include "misc.h"
          ^~~~~~~~
compilation terminated.

N0NB, 2020-03-11:
You're missing "-I hamlib/src" where the misc.h file is.Removing that include just means the functions in misc.h won't be available in the binding.
de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2020-03-11:
You're missing "-I hamlib/src" where the misc.h file is.
Removing that include just means the functions in misc.h won't be available in the binding.

de Mike W9MDB

closed 2020-03-11 by mdblack98

malcolmh, 2020-03-11:
FYI:

$ gcc -fPIC -I../Hamlib/src/ -I/usr/lib/jvm/java-8-openjdk-amd64/include/ -I/usr/lib/jvm/java-8-openjdk-amd64/include/linux/ -O0 -g3 -Wall -c -o jni/hamlib_wrap.o jni/hamlib_wrap.c
jni/hamlib_wrap.c: In function ‘Java_hamlib_HamlibJNI_rig_1caps_1macro_1name_1set’:
jni/hamlib_wrap.c:18597:13: error: ‘struct rig_caps’ has no member named ‘macro_name’; did you mean ‘model_name’?
       arg1->macro_name = (char const *) malloc(strlen((const char *)arg2)+1);
             ^~~~~~~~~~
             model_name
jni/hamlib_wrap.c:18598:28: error: ‘struct rig_caps’ has no member named ‘macro_name’; did you mean ‘model_name’?
       strcpy((char *)arg1->macro_name, (const char *)arg2);
                            ^~~~~~~~~~
                            model_name
jni/hamlib_wrap.c:18600:13: error: ‘struct rig_caps’ has no member named ‘macro_name’; did you mean ‘model_name’?
       arg1->macro_name = 0;
             ^~~~~~~~~~
             model_name
jni/hamlib_wrap.c: In function ‘Java_hamlib_HamlibJNI_rig_1caps_1macro_1name_1get’:
jni/hamlib_wrap.c:18616:30: error: ‘struct rig_caps’ has no member named ‘macro_name’; did you mean ‘model_name’?
   result = (char *) ((arg1)->macro_name);
                              ^~~~~~~~~~
                              model_name
jni/hamlib_wrap.c: In function ‘Java_hamlib_HamlibJNI_rot_1caps_1macro_1name_1set’:
jni/hamlib_wrap.c:24170:13: error: ‘struct rot_caps’ has no member named ‘macro_name’; did you mean ‘model_name’?
       arg1->macro_name = (char const *) malloc(strlen((const char *)arg2)+1);
             ^~~~~~~~~~
             model_name
jni/hamlib_wrap.c:24171:28: error: ‘struct rot_caps’ has no member named ‘macro_name’; did you mean ‘model_name’?
       strcpy((char *)arg1->macro_name, (const char *)arg2);
                            ^~~~~~~~~~
                            model_name
jni/hamlib_wrap.c:24173:13: error: ‘struct rot_caps’ has no member named ‘macro_name’; did you mean ‘model_name’?
       arg1->macro_name = 0;
             ^~~~~~~~~~
             model_name
jni/hamlib_wrap.c: In function ‘Java_hamlib_HamlibJNI_rot_1caps_1macro_1name_1get’:
jni/hamlib_wrap.c:24189:30: error: ‘struct rot_caps’ has no member named ‘macro_name’; did you mean ‘model_name’?
   result = (char *) ((arg1)->macro_name);
                              ^~~~~~~~~~
                              model_name
jni/hamlib_wrap.c: In function ‘Java_hamlib_HamlibJNI_rot_1caps_1rot_1model_1macro_1name_1set’:
jni/hamlib_wrap.c:24788:13: error: ‘struct rot_caps’ has no member named ‘rot_model_macro_name’; did you mean ‘model_name’?
       arg1->rot_model_macro_name = (char const *) malloc(strlen((const char *)arg2)+1);
             ^~~~~~~~~~~~~~~~~~~~
             model_name
jni/hamlib_wrap.c:24789:28: error: ‘struct rot_caps’ has no member named ‘rot_model_macro_name’; did you mean ‘model_name’?
       strcpy((char *)arg1->rot_model_macro_name, (const char *)arg2);
                            ^~~~~~~~~~~~~~~~~~~~
                            model_name
jni/hamlib_wrap.c:24791:13: error: ‘struct rot_caps’ has no member named ‘rot_model_macro_name’; did you mean ‘model_name’?
       arg1->rot_model_macro_name = 0;
             ^~~~~~~~~~~~~~~~~~~~
             model_name
jni/hamlib_wrap.c: In function ‘Java_hamlib_HamlibJNI_rot_1caps_1rot_1model_1macro_1name_1get’:
jni/hamlib_wrap.c:24807:30: error: ‘struct rot_caps’ has no member named ‘rot_model_macro_name’; did you mean ‘model_name’?
   result = (char *) ((arg1)->rot_model_macro_name);
                              ^~~~~~~~~~~~~~~~~~~~
                              model_name

malcolmh, 2020-03-11:
Besides, the misc.h issue, what about those warnings? Are they benign, or does something need fixing?

N0NB, 2020-03-11:
Sounds like you didn't do a "make clean" first and rebuild the bindings
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

malcolmh, 2020-03-11:
What Makefile are you referring to? The two commands I used were:

$ swig -java -I../Hamlib/include/ -outcurrentdir -package hamlib -outdir src/hamlib -o jni/hamlib_wrap.c ../Hamlib/bindings/hamlib.swg

$ gcc -fPIC -I../Hamlib/src/ -I/usr/lib/jvm/java-8-openjdk-amd64/include/ -I/usr/lib/jvm/java-8-openjdk-amd64/include/linux/ -O0 -g3 -Wall -c -o jni/hamlib_wrap.o jni/hamlib_wrap.c

> On 11 Mar 2020, at 14:33, Nate Bargmann <notifications@github.com> wrote:
> 
> Sounds like you didn't do a "make clean" first and rebuild the bindings
> de Mike W9MDB

N0NB, 2020-03-11:
Does your Hamlib/include have the latest headers?  Apparently not...macro_name is a new field.
de Mike W9MDB 

 
 

—

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

malcolmh, 2020-03-11:
According to my fork on Github: "This branch is even with Hamlib:master"

> On 11 Mar 2020, at 14:52, Nate Bargmann <notifications@github.com> wrote:
> 
> Does your Hamlib/include have the latest headers?  Apparently not...macro_name is a new field.

mdblack98, 2020-03-11:
You have a -I to a different directory though
Mike

malcolmh, 2020-03-11:
I am running those commands from a directory where I am building the Java bindings library. From that directory Hamlib is at ../Hamlib

> On 11 Mar 2020, at 15:17, Michael Black <notifications@github.com> wrote:
> 
> You have a -I to a different directory though

N0NB, 2020-03-11:
Look in Hamlib/include/rig.h -- there should be a macro_name field.
Mike

 
 

—

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

malcolmh, 2020-03-11:
Yes, there is.

malcolmh, 2020-03-11:
OK, so I built the Hamlib & ran the swig stuff again:

~/devel/HamlibJNI$ swig -java -I../Hamlib/include/ -outcurrentdir -package hamlib -outdir src/hamlib -o jni/hamlib_wrap.c ../Hamlib/bindings/hamlib.swg
cstring.i not implemented for this target
../Hamlib/bindings/hamlib.swg:181: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/hamlib.swg:182: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/hamlib.swg:183: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/hamlib.swg:184: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/include/hamlib/rig.h:1647: Warning 451: Setting a const char * variable may leak memory.
../Hamlib/include/hamlib/rotator.h:223: Warning 451: Setting a const char * variable may leak memory.
../Hamlib/include/hamlib/rotator.h:259: Warning 451: Setting a const char * variable may leak memory.
../Hamlib/bindings/rig.swg:275: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/rotator.swg:137: Warning 518: Portability warning: File src/hamlib/rig.java will be overwritten by src/hamlib/Rig.java on case insensitive filesystems such as Windows' FAT32 and NTFS unless the class/module name is renamed
../Hamlib/bindings/rotator.swg:137: Warning 518: Portability warning: File src/hamlib/rot.java will be overwritten by src/hamlib/Rot.java on case insensitive filesystems such as Windows' FAT32 and NTFS unless the class/module name is renamed

~/devel/HamlibJNI$ gcc -fPIC -I../Hamlib/src/ -I/usr/lib/jvm/java-8-openjdk-amd64/include/ -I/usr/lib/jvm/java-8-openjdk-amd64/include/linux/ -O0 -g3 -Wall -c -o jni/hamlib_wrap.o jni/hamlib_wrap.c
jni/hamlib_wrap.c: In function ‘Rig_get_level_i__SWIG_0’:
jni/hamlib_wrap.c:503:542: warning: ‘val.i’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 TER|RIG_LEVEL_NOTCHF_RAW|RIG_LEVEL_MONITOR_GAIN|RIG_LEVEL_NB))) {     self->error_status = -RIG_EINVAL;      return val.i;      }    self->error_status = rig_get_level(self->rig  ,vfo , stg, &val);    return val.i;   }
                                                                                                                     ~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_level_f__SWIG_0’:
jni/hamlib_wrap.c:504:545: warning: ‘val.f’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 TER|RIG_LEVEL_NOTCHF_RAW|RIG_LEVEL_MONITOR_GAIN|RIG_LEVEL_NB))) {     self->error_status = -RIG_EINVAL;      return val.f;      }    self->error_status = rig_get_level(self->rig  ,vfo , stg, &val);    return val.f;   }
                                                                                                                     ~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_parm_i__SWIG_0’:
jni/hamlib_wrap.c:529:206: warning: ‘val.i’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 f (((stg)&(RIG_PARM_BACKLIGHT|RIG_PARM_BAT|RIG_PARM_KEYLIGHT))) {     self->error_status = -RIG_EINVAL;      return val.i;      }    self->error_status = rig_get_parm(self->rig   , stg, &val);    return val.i;   }
                                                                                                                     ~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_parm_f__SWIG_0’:
jni/hamlib_wrap.c:530:209: warning: ‘val.f’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  (!((stg)&(RIG_PARM_BACKLIGHT|RIG_PARM_BAT|RIG_PARM_KEYLIGHT))) {     self->error_status = -RIG_EINVAL;      return val.f;      }    self->error_status = rig_get_parm(self->rig   , stg, &val);    return val.f;   }
                                                                                                                     ~~~^~

N0NB, 2020-03-14:
This has been fixed.  Should compile cleanly now with no warnings.
de Mike W9MDB

 

    On Wednesday, March 11, 2020, 11:01:22 AM CDT, Malcolm Herring <notifications@github.com> wrote:  
 
 

OK, so I built the Hamlib & ran the swig stuff again:

~/devel/HamlibJNI$ swig -java -I../Hamlib/include/ -outcurrentdir -package hamlib -outdir src/hamlib -o jni/hamlib_wrap.c ../Hamlib/bindings/hamlib.swg
cstring.i not implemented for this target
../Hamlib/bindings/hamlib.swg:181: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/hamlib.swg:182: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/hamlib.swg:183: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/hamlib.swg:184: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/include/hamlib/rig.h:1647: Warning 451: Setting a const char * variable may leak memory.
../Hamlib/include/hamlib/rotator.h:223: Warning 451: Setting a const char * variable may leak memory.
../Hamlib/include/hamlib/rotator.h:259: Warning 451: Setting a const char * variable may leak memory.
../Hamlib/bindings/rig.swg:275: Warning 453: Can't apply (char *OUTPUT). No typemaps are defined.
../Hamlib/bindings/rotator.swg:137: Warning 518: Portability warning: File src/hamlib/rig.java will be overwritten by src/hamlib/Rig.java on case insensitive filesystems such as Windows' FAT32 and NTFS unless the class/module name is renamed
../Hamlib/bindings/rotator.swg:137: Warning 518: Portability warning: File src/hamlib/rot.java will be overwritten by src/hamlib/Rot.java on case insensitive filesystems such as Windows' FAT32 and NTFS unless the class/module name is renamed

jni/hamlib_wrap.c: In function ‘Rig_get_level_f__SWIG_0’:
jni/hamlib_wrap.c:504:545: warning: ‘val.f’ may be used uninitialized in this function [-Wmaybe-uninitialized]
TER|RIG_LEVEL_NOTCHF_RAW|RIG_LEVEL_MONITOR_GAIN|RIG_LEVEL_NB))) { self->error_status = -RIG_EINVAL; return val.f; } self->error_status = rig_get_level(self->rig ,vfo , stg, &val); return val.f; }
~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_parm_i__SWIG_0’:
jni/hamlib_wrap.c:529:206: warning: ‘val.i’ may be used uninitialized in this function [-Wmaybe-uninitialized]
f (((stg)&(RIG_PARM_BACKLIGHT|RIG_PARM_BAT|RIG_PARM_KEYLIGHT))) { self->error_status = -RIG_EINVAL; return val.i; } self->error_status = rig_get_parm(self->rig , stg, &val); return val.i; }
~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_parm_f__SWIG_0’:
jni/hamlib_wrap.c:530:209: warning: ‘val.f’ may be used uninitialized in this function [-Wmaybe-uninitialized]
(!((stg)&(RIG_PARM_BACKLIGHT|RIG_PARM_BAT|RIG_PARM_KEYLIGHT))) { self->error_status = -RIG_EINVAL; return val.f; } self->error_status = rig_get_parm(self->rig , stg, &val); return val.f; }
~~~^~

—
You are receiving this because you commented.

Reply to this email directly, view it on GitHub, or unsubscribe._______________________________________________
Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

malcolmh, 2020-03-14:
Apart from those "-Wmaybe-uninitialized" warnings from gcc, it now builds successfully.
Thanks!

N0NB, 2020-03-14:
What warnings are those?

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

malcolmh, 2020-03-14:
$ gcc -fPIC -I/usr/lib/jvm/java-8-openjdk-amd64/include/ -I/usr/lib/jvm/java-8-openjdk-amd64/include/linux/ -O0 -g3 -Wall -c -o jni/hamlib_wrap.o jni/hamlib_wrap.c
jni/hamlib_wrap.c: In function ‘Rig_get_level_i__SWIG_0’:
jni/hamlib_wrap.c:506:542: warning: ‘val.i’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 TER|RIG_LEVEL_NOTCHF_RAW|RIG_LEVEL_MONITOR_GAIN|RIG_LEVEL_NB))) {     self->error_status = -RIG_EINVAL;      return val.i;      }    self->error_status = rig_get_level(self->rig  ,vfo , stg, &val);    return val.i;   }
                                                                                                                     ~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_level_f__SWIG_0’:
jni/hamlib_wrap.c:507:545: warning: ‘val.f’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 TER|RIG_LEVEL_NOTCHF_RAW|RIG_LEVEL_MONITOR_GAIN|RIG_LEVEL_NB))) {     self->error_status = -RIG_EINVAL;      return val.f;      }    self->error_status = rig_get_level(self->rig  ,vfo , stg, &val);    return val.f;   }
                                                                                                                     ~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_parm_i__SWIG_0’:
jni/hamlib_wrap.c:532:206: warning: ‘val.i’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 f (((stg)&(RIG_PARM_BACKLIGHT|RIG_PARM_BAT|RIG_PARM_KEYLIGHT))) {     self->error_status = -RIG_EINVAL;      return val.i;      }    self->error_status = rig_get_parm(self->rig   , stg, &val);    return val.i;   }
                                                                                                                     ~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_parm_f__SWIG_0’:
jni/hamlib_wrap.c:533:209: warning: ‘val.f’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  (!((stg)&(RIG_PARM_BACKLIGHT|RIG_PARM_BAT|RIG_PARM_KEYLIGHT))) {     self->error_status = -RIG_EINVAL;      return val.f;      }    self->error_status = rig_get_parm(self->rig   , stg, &val);    return val.f;   }
                                                                                                                     ~~~^~

mherring@mini5:~/devel/HamlibJNI$ ./build_linux.sh 
jni/hamlib_wrap.c: In function ‘Rig_get_level_i__SWIG_0’:
jni/hamlib_wrap.c:506:542: warning: ‘val.i’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 TER|RIG_LEVEL_NOTCHF_RAW|RIG_LEVEL_MONITOR_GAIN|RIG_LEVEL_NB))) {     self->error_status = -RIG_EINVAL;      return val.i;      }    self->error_status = rig_get_level(self->rig  ,vfo , stg, &val);    return val.i;   }
                                                                                                                     ~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_level_f__SWIG_0’:
jni/hamlib_wrap.c:507:545: warning: ‘val.f’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 TER|RIG_LEVEL_NOTCHF_RAW|RIG_LEVEL_MONITOR_GAIN|RIG_LEVEL_NB))) {     self->error_status = -RIG_EINVAL;      return val.f;      }    self->error_status = rig_get_level(self->rig  ,vfo , stg, &val);    return val.f;   }
                                                                                                                     ~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_parm_i__SWIG_0’:
jni/hamlib_wrap.c:532:206: warning: ‘val.i’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 f (((stg)&(RIG_PARM_BACKLIGHT|RIG_PARM_BAT|RIG_PARM_KEYLIGHT))) {     self->error_status = -RIG_EINVAL;      return val.i;      }    self->error_status = rig_get_parm(self->rig   , stg, &val);    return val.i;   }
                                                                                                                     ~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_parm_f__SWIG_0’:
jni/hamlib_wrap.c:533:209: warning: ‘val.f’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  (!((stg)&(RIG_PARM_BACKLIGHT|RIG_PARM_BAT|RIG_PARM_KEYLIGHT))) {     self->error_status = -RIG_EINVAL;      return val.f;      }    self->error_status = rig_get_parm(self->rig   , stg, &val);    return val.f;   }
                                                                                                                     ~~~^~

N0NB, 2020-03-14:
Try again...should be fixed....
de Mike W9MDB

 

    On Saturday, March 14, 2020, 12:01:53 PM CDT, Malcolm Herring <notifications@github.com> wrote:  
 
 

$ gcc -fPIC -I/usr/lib/jvm/java-8-openjdk-amd64/include/ -I/usr/lib/jvm/java-8-openjdk-amd64/include/linux/ -O0 -g3 -Wall -c -o jni/hamlib_wrap.o jni/hamlib_wrap.c
jni/hamlib_wrap.c: In function ‘Rig_get_level_i__SWIG_0’:
jni/hamlib_wrap.c:506:542: warning: ‘val.i’ may be used uninitialized in this function [-Wmaybe-uninitialized]
TER|RIG_LEVEL_NOTCHF_RAW|RIG_LEVEL_MONITOR_GAIN|RIG_LEVEL_NB))) { self->error_status = -RIG_EINVAL; return val.i; } self->error_status = rig_get_level(self->rig ,vfo , stg, &val); return val.i; }
~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_level_f__SWIG_0’:
jni/hamlib_wrap.c:507:545: warning: ‘val.f’ may be used uninitialized in this function [-Wmaybe-uninitialized]
TER|RIG_LEVEL_NOTCHF_RAW|RIG_LEVEL_MONITOR_GAIN|RIG_LEVEL_NB))) { self->error_status = -RIG_EINVAL; return val.f; } self->error_status = rig_get_level(self->rig ,vfo , stg, &val); return val.f; }
~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_parm_i__SWIG_0’:
jni/hamlib_wrap.c:532:206: warning: ‘val.i’ may be used uninitialized in this function [-Wmaybe-uninitialized]
f (((stg)&(RIG_PARM_BACKLIGHT|RIG_PARM_BAT|RIG_PARM_KEYLIGHT))) { self->error_status = -RIG_EINVAL; return val.i; } self->error_status = rig_get_parm(self->rig , stg, &val); return val.i; }
~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_parm_f__SWIG_0’:
jni/hamlib_wrap.c:533:209: warning: ‘val.f’ may be used uninitialized in this function [-Wmaybe-uninitialized]
(!((stg)&(RIG_PARM_BACKLIGHT|RIG_PARM_BAT|RIG_PARM_KEYLIGHT))) { self->error_status = -RIG_EINVAL; return val.f; } self->error_status = rig_get_parm(self->rig , stg, &val); return val.f; }
~~~^~

mherring@mini5:

jni/hamlib_wrap.c: In function ‘Rig_get_level_f__SWIG_0’:
jni/hamlib_wrap.c:507:545: warning: ‘val.f’ may be used uninitialized in this function [-Wmaybe-uninitialized]
TER|RIG_LEVEL_NOTCHF_RAW|RIG_LEVEL_MONITOR_GAIN|RIG_LEVEL_NB))) { self->error_status = -RIG_EINVAL; return val.f; } self->error_status = rig_get_level(self->rig ,vfo , stg, &val); return val.f; }
~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_parm_i__SWIG_0’:
jni/hamlib_wrap.c:532:206: warning: ‘val.i’ may be used uninitialized in this function [-Wmaybe-uninitialized]
f (((stg)&(RIG_PARM_BACKLIGHT|RIG_PARM_BAT|RIG_PARM_KEYLIGHT))) { self->error_status = -RIG_EINVAL; return val.i; } self->error_status = rig_get_parm(self->rig , stg, &val); return val.i; }
~~~^~

jni/hamlib_wrap.c: In function ‘Rig_get_parm_f__SWIG_0’:
jni/hamlib_wrap.c:533:209: warning: ‘val.f’ may be used uninitialized in this function [-Wmaybe-uninitialized]
(!((stg)&(RIG_PARM_BACKLIGHT|RIG_PARM_BAT|RIG_PARM_KEYLIGHT))) { self->error_status = -RIG_EINVAL; return val.f; } self->error_status = rig_get_parm(self->rig , stg, &val); return val.f; }
~~~^~

—
You are receiving this because you commented.

Reply to this email directly, view it on GitHub, or unsubscribe._______________________________________________
Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

malcolmh, 2020-03-14:
Hole in one!