Hamlib/Hamlib #200: Kenwood TS-2000 Timeout

dod1450, 2020-02-14:
I have a Kenwood TS-2000 that are having timeout issues. The connection is via a USB to RS232(9 pins) cable.  I am using the hamlib file from this site.

N0NB, 2020-02-14:
We'll need some debug info.  I have a TS-2000 emulation on my Apache ANAN 100 and it works fi
Can you run rigctl and provide some debug info showing the problem?
rigctl -m 214 -r com3 -s 115200 -vvvvv -Z
Adjust com port and baud to suit.
de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

dod1450, 2020-02-15:
.\rigctl.exe -m 214 -r com5 -s 57600 -vvvvv -Z
2020-02-14:22:49:50.845713: rigctl, Hamlib 4.0~git
2020-02-14:22:49:50.846215: Report bugs to <hamlib-developer@lists.sourceforge.net>

2020-02-14:22:49:50.846714: rig_init called
2020-02-14:22:49:50.847712: initrigs4_kenwood called
2020-02-14:22:49:50.848212: rig_register called
2020-02-14:22:49:50.849209: rig_register: rig_register (213)
2020-02-14:22:49:50.849709: rig_register called
2020-02-14:22:49:50.850208: rig_register: rig_register (201)
2020-02-14:22:49:50.850707: rig_register called
2020-02-14:22:49:50.851705: rig_register: rig_register (225)
2020-02-14:22:49:50.852204: rig_register called
2020-02-14:22:49:50.852703: rig_register: rig_register (203)
2020-02-14:22:49:50.853202: rig_register called
2020-02-14:22:49:50.853700: rig_register: rig_register (204)
2020-02-14:22:49:50.854700: rig_register called
2020-02-14:22:49:50.857195: rig_register: rig_register (216)
2020-02-14:22:49:50.858690: rig_register called
2020-02-14:22:49:50.859191: rig_register: rig_register (224)
2020-02-14:22:49:50.859690: rig_register called
2020-02-14:22:49:50.860688: rig_register: rig_register (205)
2020-02-14:22:49:50.861187: rig_register called
2020-02-14:22:49:50.861687: rig_register: rig_register (207)
2020-02-14:22:49:50.862186: rig_register called
2020-02-14:22:49:50.862684: rig_register: rig_register (209)
2020-02-14:22:49:50.863683: rig_register called
2020-02-14:22:49:50.864182: rig_register: rig_register (210)
2020-02-14:22:49:50.864681: rig_register called
2020-02-14:22:49:50.866676: rig_register: rig_register (222)
2020-02-14:22:49:50.867176: rig_register called
2020-02-14:22:49:50.869173: rig_register: rig_register (214)
2020-02-14:22:49:50.869672: rig_register called
2020-02-14:22:49:50.870170: rig_register: rig_register (230)
2020-02-14:22:49:50.871168: rig_register called
2020-02-14:22:49:50.871669: rig_register: rig_register (221)
2020-02-14:22:49:50.872167: rig_register called
2020-02-14:22:49:50.873164: rig_register: rig_register (229)
2020-02-14:22:49:50.873663: rig_register called
2020-02-14:22:49:50.874161: rig_register: rig_register (238)
2020-02-14:22:49:50.875159: rig_register called
2020-02-14:22:49:50.877656: rig_register: rig_register (202)
2020-02-14:22:49:50.878156: rig_register called
2020-02-14:22:49:50.880152: rig_register: rig_register (211)
2020-02-14:22:49:50.880650: rig_register called
2020-02-14:22:49:50.881151: rig_register: rig_register (206)
2020-02-14:22:49:50.881648: rig_register called
2020-02-14:22:49:50.882150: rig_register: rig_register (208)
2020-02-14:22:49:50.882645: rig_register called
2020-02-14:22:49:50.883150: rig_register: rig_register (215)
2020-02-14:22:49:50.883150: rig_register called
2020-02-14:22:49:50.883644: rig_register: rig_register (226)
2020-02-14:22:49:50.884143: rig_register called
2020-02-14:22:49:50.884642: rig_register: rig_register (217)
2020-02-14:22:49:50.884642: rig_register called
2020-02-14:22:49:50.885141: rig_register: rig_register (233)
2020-02-14:22:49:50.885640: rig_register called
2020-02-14:22:49:50.885640: rig_register: rig_register (220)
2020-02-14:22:49:50.887137: rig_register called
2020-02-14:22:49:50.887137: rig_register: rig_register (223)
2020-02-14:22:49:50.888634: rig_register called
2020-02-14:22:49:50.888634: rig_register: rig_register (227)
2020-02-14:22:49:50.889133: rig_register called
2020-02-14:22:49:50.889632: rig_register: rig_register (234)
2020-02-14:22:49:50.889632: rig_register called
2020-02-14:22:49:50.890132: rig_register: rig_register (231)
2020-02-14:22:49:50.890631: rig_register called
2020-02-14:22:49:50.890631: rig_register: rig_register (239)
2020-02-14:22:49:50.891130: rig_register called
2020-02-14:22:49:50.891130: rig_register: rig_register (237)
[... 151 lines cut ...]
2020-02-14:22:49:57.544264: read_string called
2020-02-14:22:49:57.850457: read_string(): Timed out 0.303943 seconds after 0 chars
2020-02-14:22:49:57.851454: kenwood_transaction: cmdstr = FA
2020-02-14:22:49:57.852953: serial_flush called
2020-02-14:22:49:57.854949: write_block called
2020-02-14:22:49:57.907456: write_block(): TX 3 bytes
2020-02-14:22:49:57.908389: 0000    46 41 3b                                            FA;
2020-02-14:22:49:57.909897: read_string called
2020-02-14:22:49:58.216959: read_string(): Timed out 0.305017 seconds after 0 chars
2020-02-14:22:49:58.217844: kenwood_transaction: cmdstr = FA
2020-02-14:22:49:58.219821: serial_flush called
2020-02-14:22:49:58.221332: write_block called
2020-02-14:22:49:58.273568: write_block(): TX 3 bytes
2020-02-14:22:49:58.274434: 0000    46 41 3b                                            FA;
2020-02-14:22:49:58.276449: read_string called
2020-02-14:22:49:58.582108: read_string(): Timed out 0.303570 seconds after 0 chars
2020-02-14:22:49:58.583146: kenwood_transaction: cmdstr = FA
2020-02-14:22:49:58.584875: serial_flush called
2020-02-14:22:49:58.587883: write_block called
2020-02-14:22:49:58.640859: write_block(): TX 3 bytes
2020-02-14:22:49:58.641705: 0000    46 41 3b                                            FA;
2020-02-14:22:49:58.643662: read_string called
2020-02-14:22:49:58.950199: read_string(): Timed out 0.304989 seconds after 0 chars
2020-02-14:22:49:58.950938: kenwood_open: no response from rig
rig_open: error = Communication timed out

N0NB, 2020-02-15:
#1 Wrong com port#2 Wrong baud rate (can't be bits, stop bits or parity as the defaults are 8N1 which is the TS-2000 defaults)#3 Cable (not plugged in or bad)

de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

dod1450, 2020-02-15:
Right com and baud. Cable works for all other amateur applications, IE 
HRD, FLDIGI, WSJT and others.

mdblack98, 2020-02-15:
What baud rate are you running at?
There are 3 parameters you can try adding to the the rigctl command line....preferably in this order
timeoutwrite_delay
post_write_delay

timeout defaults to 200ms -- so try increasing it by 50ms increments up to 500ms to see if that works.--set-conf=timeout=250
write_delay is a delay after each byte written and defaults to 0.  Increase in 10ms increments up to 50ms.  Anything more than that would seem to be overly excessive.--set-conf=write_delay=10
post_write_delay is a delay after writing a command which isn't really any different in this case from timeout.  It also defaults to 0 so could be incremented by 50ms too.
de Mike W9MDB

N0NB, 2020-02-15:
On 14/02/2020 21:56, dod1450 wrote:
> I have a Kenwood TS-2000 that are having timeout issues. The 
> connection is via a USB to RS232(9 pins) cable. I am using the hamlib 
> file from this site.

Hi Daniel,

try adding -Cserial_handshake=Hardware to your rigctl command line.

73
Bill
G4WJS.

dod1450, 2020-02-16:
Bill and Nate,

    I have tried on the recommended parameters and even changed the 
rs-232 to rs-232 cable

with NO luck. Anyone know what I can try next. I do not have the source 
code and a compiler on my windows 10 system.

N0NB, 2020-02-17:
Can you please do a screenshot of your WSJT-X and FLDigi settings?

Could be your serial cables get their power from the control lines.

In WSJT-X do you have either "Force Control Lines" options set?

If you have DTR selected than add this switch to rigctl
--set-conf=dtr_state=ON

If you have RTS selected than add this switch
--set-conf=rts_state=ON

de Mike W9MDB

dod1450, 2020-02-17:
Here is the screenshot for both applications setting

mdblack98, 2020-02-17:
User's serial interface required --set-conf=rts_state=ON

closed 2020-02-17 by mdblack98