Hamlib/Hamlib #201: [FeatureRequest] Please provide pyhton bindings in Windows releases

Wuestenschiff, 2020-02-16:
I am struggling to get the bindings to work under Windows (I am a LInux developper). Is there a possibility to include the bindings in the prebuilt releases? Building hamlib under Windows is not that easy.

Thanks for the hard work!

mdblack98, 2020-02-16:
I'll see about doing this.
de Mike W9MDB

Wuestenschiff, 2020-02-25:
@mdblack98 Did you find any time to have a look at this? (Sorry for being impatient but I have a finished Software which I can't install in the shack of our club).

N0NB, 2020-02-25:
Not yet.
Trying to finish up some other bugs and will eventually get to it.
de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

BigNerd95, 2020-06-08:
I'm also interested in this enhancement.
What about making a pip package?

BigNerd95, 2020-06-09:
I tried with this command
```
./configure --host=i686-w64-mingw32 --with-python-binding PYTHON="/usr/bin/python3" PYTHON_CPPFLAGS="-I/usr/include/python3.6m" PYTHON_LIBS="-L/usr/lib -lpython3.6m" --prefix=/tmp/HLWIN CPPFLAGS="-I/home/lory/builds/libusb-1.0.22/include" LDFLAGS="-L/home/lory/builds/libusb-1.0.22/MinGW32/dll"
```

But it always fail
```
checking whether to build python binding... yes
checking for python... /usr/bin/python3
checking for a version of Python >= '2.1.0'... yes
checking for the distutils Python package... yes
checking for Python include path... -I/usr/include/python3.6m
checking for Python library path... -L/usr/lib -lpython3.6m
checking for Python site-packages path... /usr/lib/python3/dist-packages
checking python extra libraries... -lpthread -ldl  -lutil -lm
checking python extra linking flags... -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
checking consistency of all components of python development environment... no
configure: error: in `/home/lory/builds/hamlib-3.3':
configure: error: 
  Could not link test program to Python. Maybe the main Python library has been
  installed in some non-standard library path. If so, pass it to configure,
  via the LIBS environment variable.
  Example: ./configure LIBS="-L/usr/non-standard-path/python/lib"
  ============================================================================
   ERROR!
   You probably have to install the development version of the Python package
   for your distribution.  The exact name of this package varies among them.
  ============================================================================

See `config.log' for more details
```

I already tried a lot of combination of `PYTHON_LIBS="-L/usr/lib -lpython3.6m"` but it never works

Any suggestion?

gvanem, 2020-09-24:
Pytjon 2/3 bindings works in my fork at: https://github.com/gvanem/Hamlib/

mdblack98, 2021-03-24:
Notes
Hamlib Python bindings on MinGW

As of 2021-03-21 using JTSDK 3.0.1

Edit /usr/include/pythonX.X/pyconfig.h and #undef HAVE_SYS_SELECT_H and HAVE_SYS_TERMIO_H
Install libdl https://github.com/dlfcn-win32/dlfcn-win32
https://stackoverflow.com/questions/12455160/using-libdl-so-in-mingw

mdblack98, 2023-08-03:
After numerous attempts it doesn't appear possible to build the python bindings in our release build system.

closed 2023-08-03 by mdblack98

kyleboyle, 2024-05-23:
FYI I can build python bindings in cygwin with mingw32 on the 4.5.5 release with this:

`./configure --prefix=$HOME/local --build=x86_64-pc-cygwin --host=x86_64-pc-mingw32 --target=x86_64-pc-mingw32 CC='x86_64-pc-cygwin-gcc' CXX='x86_64-pc-cygwin-g++' --with-python-binding`

cygwin has to have a bunch of modules installed, libusb, swig, etc 
[image]

I haven't tried i686 yet.