Hamlib/Hamlib #202: Rigctld/WSJTX IC-7300 issue in latest Master

hlnurse, 2020-02-20:
In one of the last few merges the rigctld-WSJTX connection for the IC-7300 appears to be broken.  WSJTX does not see a sub VFO and issues Hamlib error: Invalid parameter while exchanging VFOs.  

WSJT-X version 2.1.2.  In WSJT-X, radio is set to Hamlib NET rigctl, localhost:4532, PTT CAT, Mode USB, Split Operation None.

I start rigctld from Terminal with this command:

sudo rigctld -m 3073 -r /dev/ttyUSB0 -s 115200 -P GPIO -p 17 -vvvvv

Log is below.

I also note that most rig numbers have been changed (offset).  Is there a reason for this?

Thanks!

--Howard

pi@rigpi4:~/Downloads/Hamlib-master $ sudo rigctld -m 3073 -r /dev/ttyUSB0 -s 115200 -P GPIO -p 17 -vvvvv
Recommend using --vfo switch for rigctld
rigctl and netrigctl will automatically detect vfo mode
rigctld, Hamlib 4.0~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
initrigs4_icom: _init called
rig_register called
rig_register: rig_register (3055)
rig_register called
rig_register: rig_register (3009)
rig_register called
rig_register: rig_register (3010)
rig_register called
rig_register: rig_register (3011)
rig_register called
rig_register: rig_register (3013)
rig_register called
rig_register: rig_register (3014)
rig_register called
rig_register: rig_register (3015)
rig_register called
rig_register: rig_register (3019)
rig_register called
rig_register: rig_register (3020)
rig_register called
rig_register: rig_register (3021)
rig_register called
rig_register: rig_register (3022)
rig_register called
rig_register: rig_register (3067)
rig_register called
rig_register: rig_register (3023)
rig_register called
rig_register: rig_register (3046)
rig_register called
rig_register: rig_register (3024)
rig_register called
rig_register: rig_register (3028)
rig_register called
rig_register: rig_register (3030)
rig_register called
rig_register: rig_register (3026)
rig_register called
rig_register: rig_register (3027)
rig_register called
rig_register: rig_register (3047)
rig_register called
rig_register: rig_register (3057)
rig_register called
rig_register: rig_register (3063)
rig_register called
rig_register: rig_register (3029)
[... 436 lines cut ...]
rig_parse_vfo called
rig_set_vfo called
icom_set_vfo called
icom_set_vfo: Rig does not have VFO Main/Sub?
sync_callback: client lock disengaged
rigctl_parse: debug1
rigctl_parse: debug3
rigctl_parse: debug4
rigctl_parse: debug5
rigctl_parse: debug10
sync_callback: client lock engaged
rigctl(d): T 'currVFO' '0' '' ''
rig_set_ptt called
sync_callback: client lock disengaged
rig_open called
sync_callback: client lock engaged
rig_close called
icom_rig_close: called
port_close called
ser_close called
ser_close: restoring options
Closed rig model 3073, 'IC-7300 - no clients, will reopen for new clients'
sync_callback: client lock disengaged
Connection closed from localhost:35930
handle_socket: close(handle_data_arg->sock) Unknown error -1

N0NB, 2020-02-20:
The rig numbers have changed to make more headroom for new models for each manufacturer.  Icom is starting to run close to the 100 max that was the old system.  Now it's 1000 models per manufacturer so should last quite a while.
I'll have to find out why WSJT-X ends up asking for Main/Sub when the 7300 has no such VFO defined.  The WSJT-X code indicates is should be calling for VFOA/B since that is defined for the 7300.  I am able to reproduce this problem here so can chase this down.
de Mike W9MDB

 
 

Hamlib-developer mailing list
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-02-21:
On 2020-02-20 21:27, hlnurse wrote:
> 
> I also note that most rig numbers have been changed (offset). Is there
> a reason for this?

yes

Author: Michael Black <mdblack98@yahoo.com>
Date:   Tue Feb 18 06:53:56 2020 -0600

     Change RIG_MODEL_NUMBER macro to use 1000 multiplier instead of 100
     Icom is getting close to running out of numbers
     Change rig_model_t to uint32_t to ensure it's 32-bit

ICOM makes too many god and succesful rigs that we're running out of
space ..

mdblack98, 2020-02-21:
Please try my fork again....should be OK now.
https://github.com/mdblack98/Hamlib.git

de Mike W9MDB

 

 

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

hlnurse, 2020-02-21:
Thanks, Mike, looks good here.  Was the fix specific to the IC-7300 or would it apply to other radios as well?

--Howard

N0NB, 2020-02-22:
Other rigs too....
This should all get checked into the master probably by tomorrow...chasing down one more bug with the IC9700 that hopefully will get fixed today or tomorrow.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-02-22:
* On 2020 21 Feb 18:09 -0600, Black Michael wrote:
> Other rigs too....
> This should all get checked into the master probably by
> tomorrow...chasing down one more bug with the IC9700 that hopefully
> will get fixed today or tomorrow.

Will that mean I need to get ready to branch 4.0 and release an rc1?

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

mdblack98, 2020-02-22:
We're getting close.  I'd like to clear up all the issues on the list so I'll review them and see what's left after this rather large update I'm getting ready to push.It will clear up a couple of issues and depending on the severity of what's left we may be ready....I'll let you know.
Mike

 

Will that mean I need to get ready to branch 4.0 and release an rc1?

73, Nate

mdblack98, 2020-03-03:
Fixed

closed 2020-03-03 by mdblack98