Hamlib/Hamlib #203: Support transceiver polling on Windows

tmiw, 2020-02-24:
It looks like polling needs sigaction() to exist on Windows to function properly if [this](https://github.com/Hamlib/Hamlib/blob/master/src/event.c#L225) is any indication. Unfortunately, it does not, hence why the feature disables itself. Any possibility of porting that functionality over from Unix/Linux?

(This issue came up during the submission of [this PR](https://github.com/drowe67/freedv-gui/pull/47) for another project. The feature seemed to work on Mac and Linux but not Windows.)

mdblack98, 2020-02-24:
All the PR shows is that Windows wasn't tested...
All you have to do is ask for the mode via hamlib.  sigaction is for unsolicited radio info which most rigs don't provide and that PR doesn't reference sigaction at all.
de Mike W9MDB

srsampson, 2020-02-24:
```All the PR shows is that Windows wasn't tested...```
Look towards the bottom

```...that PR doesn't reference sigaction at all.```
You must be reading something different

In any case, your response is not very useful.

N0NB, 2020-02-24:
FreeDV simply needs to do a mode request via hamlib to get mode at anytime, even if PTT is active.You can ask for current Mode/Freq on any rig,  just can't change freq on some rigs when transmitting.
Using sigaction is not necessary to get the mode.  It can be asked for any time freq is asked for.
I see FreeDV asks for VFO (v) and PTT (T)...they just need to add an "m" command to get the mode.
2020-02-24:20:20:57.659563: rigctl(d): v 'currVFO' '' '' ''2020-02-24:20:20:57.660538: rig_get_vfo called2020-02-24:20:20:57.666399: tt588_get_vfo: vfo=VFOA2020-02-24:20:20:57.667374: sync_callback: client lock disengaged2020-02-24:20:21:00.438652: rigctl_parse: debug12020-02-24:20:21:00.439625: rigctl_parse: debug32020-02-24:20:21:00.440601: rigctl_parse: debug42020-02-24:20:21:00.441580: rigctl_parse: debug52020-02-24:20:21:00.443533: rigctl_parse: debug102020-02-24:20:21:00.447438: sync_callback: client lock engaged2020-02-24:20:21:00.449388: rigctl(d): T 'currVFO' '0' '' ''2020-02-24:20:21:00.450365: rig_set_ptt called2020-02-24:20:21:00.451342: tt588_set_ptt: ptt=02020-02-24:20:21:00.452318: serial_flush called2020-02-24:20:21:00.454271: serial_flush: tcflush2020-02-24:20:21:00.455247: write_block called2020-02-24:20:21:00.457199: write_block(): TX 5 bytes2020-02-24:20:21:00.457199: 0000    2a 54 00 00 0d                                      *T...2020-02-24:20:21:00.461108: sync_callback: client lock disengaged

de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-02-24:
Arrghhh...blasted HTML formatting.....

On Monday, February 24, 2020, 02:34:15 PM CST, Black Michael <mdblack98@yahoo.com> wrote: 

FreeDV simply needs to do a mode request via hamlib to get mode at anytime, even if PTT is active.
You can ask for current Mode/Freq on any rig,  just can't change freq on some rigs when transmitting.

Using sigaction is not necessary to get the mode.  It can be asked for any time freq is asked for.

I see FreeDV asks for VFO (v) and PTT (T)...they just need to add an "m" command to get the mode.

2020-02-24:20:20:57.659563: rigctl(d): v 'currVFO' '' '' ''
2020-02-24:20:20:57.660538: rig_get_vfo called
2020-02-24:20:20:57.666399: tt588_get_vfo: vfo=VFOA
2020-02-24:20:20:57.667374: sync_callback: client lock disengaged
2020-02-24:20:21:00.438652: rigctl_parse: debug1
2020-02-24:20:21:00.439625: rigctl_parse: debug3
2020-02-24:20:21:00.440601: rigctl_parse: debug4
2020-02-24:20:21:00.441580: rigctl_parse: debug5
2020-02-24:20:21:00.443533: rigctl_parse: debug10
2020-02-24:20:21:00.447438: sync_callback: client lock engaged
2020-02-24:20:21:00.449388: rigctl(d): T 'currVFO' '0' '' ''
2020-02-24:20:21:00.450365: rig_set_ptt called
2020-02-24:20:21:00.451342: tt588_set_ptt: ptt=0
2020-02-24:20:21:00.452318: serial_flush called
2020-02-24:20:21:00.454271: serial_flush: tcflush
2020-02-24:20:21:00.455247: write_block called
2020-02-24:20:21:00.457199: write_block(): TX 5 bytes
2020-02-24:20:21:00.457199: 0000    2a 54 00 00 0d                                      *T...
2020-02-24:20:21:00.461108: sync_callback: client lock disengaged

de Mike W9MDB

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2020-02-24:
I just looked at the FreeDV code.

You just need this somewhere...maybe after you connect or more often...

mode_t mode;
int retcode = rig_get_mode(m_rig, RIG_VFO_CURR, &mode);
if (retcode != RIG_OK).....do something....
if (mode == RIG_MODE_USB || mode == RIG_MODE_PKTUSB)...
else if (mode == RIG_MODE_LSB || mode = RIG_MODE_PKTLSB)....
else bad mode

de Mike W9MDB

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-02-24:
Re:-
You can ask for current Mode/Freq on any rig,

Sadly not entirely so.  The still popular FT-736r (and possibly others)
just do not have that functionality.  For "client" software such as
Gpredict, you need to fake it either by hacking Hamlib, or an extra layer
between application and Hamlib.

Sorry about the font, doing this on a phone.

73:  Dave G0WBX

mdblack98, 2020-02-24:
Yes...there are a few rigs that don't provide that capability...nothing we can do about that.

In that case the rig_get_mode call returns RIG_ENAVAIL;

de Mike W9MDB

tmiw, 2020-02-24:
My apologies, I should probably clarify further as to what's going on.

The PR linked in OP is for a proposed GUI element in FreeDV that displays the radio's current mode and turns red if it's not what's expected for the radio's current band (LSB <= 40m, USB >= 20m). The basic functionality of this works on all tested platforms (Mac, Linux and Windows 10) in that when the Start button's pressed, the current code in the PR is able to make the needed rig_* calls to retrieve the current frequency and mode (src/hamlib.cpp [lines 192-214](https://github.com/drowe67/freedv-gui/pull/47/files#diff-192b42b90f0378588953e0aa18ddf023R192)).

A nice to have feature, however, is the ability for FreeDV to track changes to frequency and mode and update the aforementioned GUI element as that occurs. Based on what I've been able to find in the Hamlib documentation thus far, this seems to be accomplished through the use of code similar to the following:

`rig_set_freq_callback(m_rig, &hamlib_freq_cb, this);`
`rig_set_mode_callback(m_rig, &hamlib_mode_cb, this);`
`rig_set_trn(m_rig, RIG_TRN_POLL);`

followed by specific implementations of the registered callback functions to do whatever desired actions the developer wants.

In my testing with the above code, the callbacks seem to properly be getting executed within a second or so after changing band and mode on the tested radio itself (Elecraft KX3)--but only on Mac (macOS Catalina) and Linux (Ubuntu 18.04 LTS). When the same code is run on Windows 10, only the initial state update mentioned above seems to occur and not the expected state tracking.

Anyway, the reason why I concluded that sigaction() seems to be necessary for rig_set_trn() is that rig_set_trn() calls add_trn_poll_rig(), which is wrapped such that the implementation becomes simply a stub that returns -RIG_ENIMPL if HAVE_SIGACTION isn't defined. In addition, there's a second #ifdef wrapper in rig_set_trn() [that doesn't even bother calling add_trn_poll_rig()](https://github.com/Hamlib/Hamlib/blob/7c7ecef678c62eaed10bacab548f99f64ce65811/src/event.c#L763) if HAVE_SETITIMER isn't defined. 

All that said, as the PR is now, the basic feature works as intended. It just would be nice to be able to do the above on Windows as well, even if it means Hamlib has to have Windows-specific code to simulate what it's doing on other platforms. Maybe this should be tagged to indicate that it's a feature request rather than an actual issue?

BTW, @N0NB, good call on not being able to read frequency/mode on some rigs. I'll have to see how to handle that on the FreeDV side (maybe by not enabling that control for them?)

N0NB, 2020-02-24:
Why don't you ask directly for freq & mode during your own polling cycle rather than relying on the rig to send it to you async?

As far as I know more rigs will answer rig_get_freq and rig_get_mode than will give async changes.

de Mike W9MDB

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-02-25:
Just for reference...in hamlib there are 292 rig models available....but only 132 show CAT transceive capability...most all the Icom rigs have it.
So you will be  compatible with may more rigs if you poll freq & mode yourself rather than relying on sigaction.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

tmiw, 2020-02-25:
Actually, speaking of rig support, I did try calling rig_set_trn() with RIG_TRN_RIG instead of RIG_TRN_POLL and found that it didn't work with the radio I was using it with (in retrospect, probably due to lack of support as you mention). However, RIG_TRN_POLL did work with the radio I was using. 

In fact, it appears that Hamlib only does capability checking if using RIG_TRN_RIG (see [here](https://github.com/Hamlib/Hamlib/blob/master/src/event.c#L746)). Using polling seems to result in [search_rig_and_poll() getting called](https://github.com/Hamlib/Hamlib/blob/master/src/event.c#L368) regardless of radio support; handling for rigs that don't support e.g. frequency queries happens in there.

Anyway, it looks like there is some sort of handling in Hamlib already that could be tapped into for any Windows specific implementation if the library itself is extended in the future. Alternatively, it could just be used as a basis for an implementation on the FreeDV side.

N0NB, 2020-02-25:
I've never really looked at that before but I believe the meaning is intended to be this:
RIG_TRN_OFF -- can't get any info from rig at allRIG_TRN_POLL -- get info by polling the rig(s)RIG_TRN_RIG -- rig(s) can provide async data
This hasn't been documented very well so needs to be improved.A lot of this is based on Icom's CI-V protocol which is different than everybody else's as you can have multiple rigs on the bus.
What rig were you testing?   
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2020-02-25:
Also...RIG_TRN_POLL will work with more rigs then RIG_TRN_RIG since I do believe any rig that has RIG_TRN_RIG can be polled too but not vice versa.
de Mike W9MDB

 

 

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

tmiw, 2020-02-25:
I was using a Elecraft KX3 for testing. 

Also, if RIG_TRN_OFF is set, wouldn't that mean that manual rig_* calls to get frequency/mode wouldn't work? I have the set_trn stuff #ifdef'd out on the FreeDV side so I would have assumed it'd be set to that by default.

N0NB, 2020-02-25:
The rig_* calls will still work.  RIG_TRN_OFF just means they have to be polled.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

tmiw, 2020-02-26:
That matches the behavior I was seeing. For documentation purposes, though, I'd say that RIG_TRN_POLL causes Hamlib to do the polling (on supported platforms and/or radios).

N0NB, 2020-02-26:
Yes...after a bit more thought RIG_TRN_OFF simply disables the RIG_TRN_POLL or RIG_TRN_RIG that has been turned on.
These are values for enabling/disabling the function.
Mike 

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2020-12-07:
Possible Windows Async code to borrow for hamlib
https://stackoverflow.com/questions/59867506/asynchronous-serial-i-o-on-win32-how-to-create-an-alert-event

mdblack98, 2021-11-29:
This issue is being replaced by a future implementation of async operations.

closed 2021-11-29 by mdblack98