Hamlib/Hamlib #204: Update wiki/Supported Rotators, add macro name

kerel-fs, 2020-03-01:
When using Hamlib via the Python bindings (e.g. in [satnogsclient:satnogsclient/observer/worker.py#L134)](https://gitlab.com/librespacefoundation/satnogs/satnogs-client/-/blob/master/satnogsclient/observer/worker.py#L134)) generated by swig, the rotator model must be provided using the rotator model macros. To allow users to get the model name without looking into the source code of Hamlib (macro defines at [rotlist.h](https://github.com/Hamlib/Hamlib/blob/4c47ff4f9315788566a26a5996f34d45406ccd1a/include/hamlib/rotlist.h)), I added the macro names to a list of supported rotators, based on the list from the wiki. Additionally I added missing models, fixed a typo in the heading and introduced the use of markdown-syntax tables.

Please consider to update the wiki page [Supported Rotators](https://github.com/Hamlib/Hamlib/wiki/Supported-Rotators) with the following table (this should be able via copy-paste. If github doesn't provide a way to show the source of this post, please let me know and I will post an escaped version): 

Rot #  | Macro Name | Mfg                    | Model                   | Version         | Status
-------|-|------------------------|-------------------------|-----------------|---------
1 | ROT_MODEL_DUMMY | Hamlib                 | Dummy                   | 0.2             | Beta
2 | ROT_MODEL_NETROTCTL| Hamlib                 | NET rotctl              | 0.3             | Beta
201 | ROT_MODEL_EASYCOMM1 | Hamlib                 | EasycommI               | 0.4             | Beta
202 | ROT_MODEL_EASYCOMM2 | Hamlib                 | EasycommII              | 0.4             | Beta
204 |  | Hamlib                 | EasycommIII             | 0.4             | Alpha
301 | ROT_MODEL_FODTRACK | XQ2FOD                 | Fodtrack                | 0.2             | Stable
401 | ROT_MODEL_ROTOREZ | Idiom Press            | Rotor-EZ                | 2010-02-14      | Beta
402 | ROT_MODEL_ROTORCARD | Idiom Press            | RotorCard               | 2010-02-14      | Untested
403 | ROT_MODEL_DCU | Hy-Gain                | DCU-1/DCU-1X            | 2010-08-23      | Untested
404 | ROT_MODEL_ERC | DF9GR                  | ERC                     | 2010-08-23b     | Alpha
405 | ROT_MODEL_RT21 | Green Heron            | RT-21                   | 2014-09-14      | Alpha
501 | ROT_MODEL_SARTEK1 | SARtek                 | SARtek-1                | 0.2             | Untested
601 | ROT_MODEL_GS232A | Yaesu                  | GS-232A                 | 0.3             | Beta
602 | ROT_MODEL_GS232_GENERIC | Yaesu/Kenpro           | GS-232                  | 0.1             | Beta
603 | ROT_MODEL_GS232B | Yaesu                  | GS-232B                 | 0.2             | Beta
604 | ROT_MODEL_F1TETRACKER | F1TE                   | GS232/F1TE Tracker      | 0.1             | Beta
701 | ROT_MODEL_PCROTOR | WA6UFQ                 | PcRotor                 | 0.1             | Untested
801 | ROT_MODEL_HD1780 | Heathkit               | HD 1780 Intellirotor    | 0.1             | Beta
901 | ROT_MODEL_SPID_ROT2PROG | SPID                   | Rot2Prog                | 1.0             | Stable
902 | ROT_MODEL_SPID_ROT1PROG | SPID                   | Rot1Prog                | 1.0             | Stable
1001 | ROT_MODEL_RC2800 | M2                     | RC2800                  | 0.1.1           | Beta
1101 | ROT_MODEL_RCI_AZEL | EA4TX                  | ARS RCI AZ&EL           | 0.1             | Beta
1102 | ROT_MODEL_RCI_AZ | EA4TX                  | ARS RCI AZ              | 0.1             | Beta
1201 | ROT_MODEL_IF100 | AMSAT                  | IF-100                  | 0.1             | Untested
1301 | ROT_MODEL_TS7400 | LA7LKA                 | ts7400                  | 0.1             | Beta
1401 | ROT_MODEL_NEXSTAR | Celestron              | NexStar                 | 0.1             | Untested
1501 | ROT_MODEL_ETHER6 | DG9OAA                 | Ether6 (via ethernet)   | 0.1             | Beta
1601 | ROT_MODEL_CNCTRK | CNCTRK                 | CNCTRK                  | 0.1             | Untested
-| ROT_MODEL_PROSISTEL |  |  |  | 
-| ROT_MODEL_MEADE | Meade | DS-2000 |  | 
-| ROT_MODEL_IOPTRON | IOPTRON |  |  |

N0NB, 2020-03-01:
Hi Fabian.

Typically, the list is updated via a script in the main directory upon a
release.  The list has been hosted at the SourceForge Web space as it is
generated to HTML and far predates Markdown or the GitHub dialect.  This
list is not well structured and could use a lot of improvement.

Of course, if someone wants to take on the effort to generate Markdown
that could be imported directly into the GitHub Wiki, that would be
welcome.

:-)

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

DL4PD, 2020-03-02:
Not sure I understand what you are talking about - "the list" where Fabian's suggestion points at is located here:
 * https://github.com/Hamlib/Hamlib/wiki/Supported-Rotators
That sould be the wiki you are talking about, no!?

kerel-fs, 2020-03-02:
> Typically, the list is updated via a script in the main directory upon a
release.

Auto-generating this list upon a release is great and I'd like to continue that!  (Given that we can also update the list prior to the next release now, otherwise we'll have to host a current list ourself somewhere else)

The list seems to be be generated with
```
rotctl --list
```

The code for this output might be the following from `tests/rotctl_parse.c` (although I wonder why this is located in the directory `tests`): https://github.com/Hamlib/Hamlib/blob/be922f800fcbf4284ecc8372f48e173c4d4a7da6/tests/rotctl_parse.c#L1613-L1631

This code for `rotctl --list` is rather complicated due to the peculiar method to sort entries, and its last update was performed in 2012 (see https://github.com/Hamlib/Hamlib/commit/417cb98d2eed6d756ac3683cb9f09b9ed1053d4e, neglecting a typo fix in https://github.com/Hamlib/Hamlib/commit/f9b2f27373ab4c04a75be59edfc826db32296884 / 2018). Thus I propose to write a simple Python script instead.

kerel-fs, 2020-03-02:
I've written such a Python script now, see [this gist](https://gist.github.com/kerel-fs/2354ff9c082cd3c1594afc38c67bc32a).

Please let me know how to integrate this into the Hamlib tree & release process.

N0NB, 2020-03-02:
Unfortunately your solution introduces a dependency on the hamlib python bindings to generate documentation.  Not very desirable IMHO.
Why do you use macro names?  They were not intended for user consumption.  Everybody else uses the model name or model number.
de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

kerel-fs, 2020-03-02:
The macro name is used to allow users to enter an expressive string instead of an arbitrary number for their model id in a user variable `settings.ROT_MODEL`. This string is then used to instantiate the rotator using the Hamlib python binding:
```python
rotator = Rotator(settings.ROT_MODEL, settings.ROT_BAUD, settings.ROT_PORT)
rotator.open()
# ...
```

Is it possible to reliably get the model id from a user-provided model name using the Python bindings? If yes, I'd fully agree to drop the macro names from user-exposure again and keep them as an internal implementation detail only.

kerel-fs, 2020-03-02:
> Unfortunately your solution introduces a dependency on the hamlib python bindings to generate documentation.  Not very desirable IMHO.

Agreed. Especially  since there is exiting non-Python code for the docs generation for rigs at [rigmatrix.c](https://github.com/Hamlib/Hamlib/blob/da86568edbeab2f7adf3adccffb614a5b18c5f2a/tests/rigmatrix.c). On the other hand I'd prefer to use a modern script language instead of C when writing new documentation generation code (easy templating, filtering, sorting).

mdblack98, 2020-03-02:
Why don't you provide a GUI/interface for selecting the model?
de Mike W9MDB

N0NB, 2020-03-02:
And why are we updating the formatting?  We need to maintain both Radio and Rotator pages looking the same.
Are you referring people to the Wiki page to figure out what string to use?  And this is only for rotators and not rigs?
Mike

 
 

Yes, this will probably done as well. :)
This issue was written to update the formatting of https://github.com/Hamlib/Hamlib/wiki/Supported-Rotators though.

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

kerel-fs, 2020-03-02:
> And why are we updating the formatting?

I though markdown-syntax table might improve the documentation.

> And this is only for rotators and not rigs?

Indeed, just as a starter.

Best wishes,
Fabian

mdblack98, 2020-03-02:
If you look at the documentation on the Radios it's much more extensive and would be hard to duplicate in a python script.
Thought, at the moment, rigmatrx seems to be seg faulting on my machine....gotta' track that down.
I'd rather see the more extensive docs on the rotators like the radios have.
Mike

kerel-fs, 2020-03-02:
Thanks for maintaining this great library and your feedback on this issue. I'll not follow this issue further for now though.

The new Python script for supported rotor models is working, format-checked and easily extendable, and I'll not work on any `printf` code for documentation myself. Nevertheless please let me know in case you're opinion changes and I'll happily extend it to generate more extensive documentation.

I can understand your opinion that adding a Python-based script only for the list of supported rotator models might be undesirable based on already existing extensive documentation generated using C / `printf`.

Sincerely,
kerel

OT: @N0NB , @mdblack98 Note that the original message is included verbatim in your answers ([Top-Posting style](https://en.wikipedia.org/wiki/Posting_style#Top-posting)). Thus the github issue is hard to read online (see [Hamlib#204](https://github.com/Hamlib/Hamlib/issues/204)). Please consider to remove the original message when answering to prevent this.

mdblack98, 2020-03-06:
### Macro names now show for rigs, amps, and rotators.
Wiki docs will be updated when 4.0 release candidate comes out
rotctl -l
 Rot #  Mfg                    Model                   Version         Status        Macro
     1  Hamlib                 Dummy                   0.2             Beta          ROT_MODEL_DUMMY
     2  Hamlib                 NET rotctl              0.3             Beta          ROT_MODEL_NETROTCTL
   201  Hamlib                 EasycommI               0.4             Beta          ROT_MODEL_EASYCOMM1
   202  Hamlib                 EasycommII              0.4             Beta          ROT_MODEL_EASYCOMM2
   204  Hamlib                 EasycommIII             0.4             Alpha         ROT_MODEL_EASYCOMM3
   301  XQ2FOD                 Fodtrack                0.2             Stable        ROT_MODEL_FODTRACK
   401  Idiom Press            Rotor-EZ                2010-02-14      Beta          ROT_MODEL_ROTOREZ
   402  Idiom Press            RotorCard               2010-02-14      Untested      ROT_MODEL_ROTORCARD
   403  Hy-Gain                DCU-1/DCU-1X            2010-08-23      Untested      ROT_MODEL_DCU
   404  DF9GR                  ERC                     2010-08-23b     Alpha         ROT_MODEL_ERC
   405  Green Heron            RT-21                   2014-09-14      Alpha         ROT_MODEL_RT21
   501  SARtek                 SARtek-1                0.2             Untested      ROT_MODEL_SARTEK1
   601  Yaesu                  GS-232A                 0.5             Beta          ROT_MODEL_GS232A
   602  Various                GS-232 Generic          0.3             Beta          ROT_MODEL_GS232_GENERIC
   603  Yaesu                  GS-232B                 0.5             Stable        ROT_MODEL_GS232B
   604  F1TE                   GS232/F1TE Tracker      0.1             Beta          ROT_MODEL_F1TETRACKER
   605  Yaesu/Kenpro           GS-23                   0.2             Alpha         ROT_MODEL_GS23
   606  Yaesu/Kenpro           GS-232                  0.1             Beta          ROT_MODEL_GS232
   607  AMSAT                  LVB Tracker             0.1             Alpha         ROT_MODEL_LVB
   608  FoxDelta               GS232/ST2               0.1             Alpha         ROT_MODEL_ST2
   701  WA6UFQ                 PcRotor                 0.1             Untested      ROT_MODEL_PCROTOR
   801  Heathkit               HD 1780 Intellirotor    0.1             Beta          ROT_MODEL_HD1780
   901  SPID                   Rot2Prog                1.0             Stable        ROT_MODEL_SPID_ROT2PROG
   902  SPID                   Rot1Prog                1.0             Stable        ROT_MODEL_SPID_ROT1PROG
   903  SPID                   MD-01/02 (ROT2 mode)    1.0             Stable        ROT_MODEL_SPID_MD01_ROT2PROG
  1001  M2                     RC2800                  0.1.1           Beta          ROT_MODEL_RC2800
  1101  EA4TX                  ARS RCI AZ&EL           0.2             Beta          ROT_MODEL_RCI_AZEL
  1102  EA4TX                  ARS RCI AZ              0.1             Beta          ROT_MODEL_RCI_AZ
  1201  AMSAT                  IF-100                  0.1             Untested      ROT_MODEL_IF100
  1301  LA7LKA                 ts7400                  0.1             Beta          ROT_MODEL_TS7400
  1401  Celestron              NexStar                 0.1             Untested      ROT_MODEL_NEXSTAR
  1501  DG9OAA                 Ether6 (via ethernet)   0.1             Beta          ROT_MODEL_ETHER6
  1601  CNCTRK                 CNCTRK                  0.1             Untested      ROT_MODEL_CNCTRK
  1701  Prosistel              Prosistel D             0.5             Stable        ROT_MODEL_PROSISTEL
  1801  Meade                  LX200/Autostar          0.2             Stable        ROT_MODEL_MEADE
  1901  iOptron                iOptron                 0.1             Alpha         ROT_MODEL_IOPTRON

closed 2020-03-06 by mdblack98

kerel-fs, 2020-03-06:
Wow, many thanks @mdblack98 !