Hamlib/Hamlib #205: TS-480 set_powerstat

hlnurse, 2020-03-03:
Power on using set_powerstat for the TS-480 appears to be broken.  The TS-480 has two power off modes, one (PS3;) where the cpu sleeps and the other (PS0;) where the cpu is active.   Shouldn't the TS-480 attempt power on as part of the init?  

Log below.

Thanks!

--Howard W6HN

pi@rigpi:~/Downloads/Hamlib-master $ rigctl -m 2028 -r /dev/ttyUSB0 -vvvvv
rigctl, Hamlib 4.0~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
initrigs4_kenwood called
rig_register called
rig_register: rig_register (2013)
rig_register called
rig_register: rig_register (2001)
rig_register called
rig_register: rig_register (2025)
rig_register called
rig_register: rig_register (2003)
rig_register called
rig_register: rig_register (2004)
rig_register called
rig_register: rig_register (2016)
rig_register called
rig_register: rig_register (2024)
rig_register called
rig_register: rig_register (2005)
rig_register called
rig_register: rig_register (2007)
rig_register called
rig_register: rig_register (2009)
rig_register called
rig_register: rig_register (2010)
rig_register called
rig_register: rig_register (2022)
rig_register called
rig_register: rig_register (2014)
rig_register called
rig_register: rig_register (2030)
rig_register called
rig_register: rig_register (2021)
rig_register called
rig_register: rig_register (2029)
rig_register called
rig_register: rig_register (2043)
rig_register called
rig_register: rig_register (2044)
rig_register called
rig_register: rig_register (2045)
rig_register called
rig_register: rig_register (2038)
rig_register called
rig_register: rig_register (2002)
rig_register called
rig_register: rig_register (2011)
rig_register called
rig_register: rig_register (2006)
rig_register called
rig_register: rig_register (2008)
rig_register called
rig_register: rig_register (2015)
rig_register called
rig_register: rig_register (2026)
rig_register called
rig_register: rig_register (2017)
rig_register called
rig_register: rig_register (2033)
[... 244 lines cut ...]
kenwood_transaction: cmdstr = PS1
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 4 bytes
0000    50 53 31 3b                                         PS1;            
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string called
read_string(): Timed out 0.200271 seconds after 0 chars
kenwood_transaction: cmdstr = PS1
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 4 bytes
0000    50 53 31 3b                                         PS1;            
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string called
read_string(): Timed out 0.200274 seconds after 0 chars
kenwood_set_powerstat called status=1
kenwood_set_powerstat: timeout waiting for powerup, try 1
set_powerstat: error = Communication timed out

N0NB, 2020-03-03:
According to the TS-480 document I have PS9; is the CPU sleep.
So if you send ";;;;PS1;" does it wake up?
Using the "w" command from rigctl?
w ;;;;PS1;

 
 

Hamlib-developer mailing list
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

mdblack98, 2020-03-03:
Try my fork....I just added the kenwood_open to the TS480 and change the PS1 command to wake up a sleeping CPU too (hopefully).
https://github.com/mdblack98/Hamlib

de Mike W9MDB

 

 

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

hlnurse, 2020-03-03:
No, w ;;;;PS1; does not wake it up.

Rig command: w ;;;;PS1;
rigctl_parse: debug1
rigctl_parse: debug2
rigctl_parse: debug2a
rigctl_parse: debug5
rigctl_parse: debug10
rigctl_send_cmd: called
rigctl_send_cmd: backend_num=2
rigctl_send_cmd: KENWOOD
rigctl_send_cmd: arg1=;;;;PS1;
rigctl_send_cmd: bufcmd=;;;;PS1;
rigctl_send_cmd: text protocol, send_cmd_term=0x00 0x0d 0x00
serial_flush called
serial_flush: tcflush
rigctl_send_cmd: rigport=3, bufcmd=;;;;PS1;, cmd_len=8
write_block called
write_block(): TX 8 bytes
0000    3b 3b 3b 3b 50 53 31 3b                             ;;;;PS1;        
rigctl_send_cmd: debug Cmd
read_string called
read_string(): RX 1 characters
0000    00                                                  .               
Reply:

hlnurse, 2020-03-03:
No improvement with the new fork in \set_powerstat, init or w command. Debug logs look the same.

--Howard

mdblack98, 2020-03-04:
You said your manual shows PS3 ?  Can you send me your CAT manual?
The one I have gave the ";;;;PS1;" to wake up but I guess the manual is wrong or your rig doesn't behave as the manual says it should.
So unless your manual has some other way to take up the rig I don't know what else to do.
The debug logs should not look quite the same since it's sending the new ;;;;PS1 command now several times.  I'm concerned though that it might cause a timeout in WSJT-X or such. 
Mike

mdblack98, 2020-03-04:
Try my for again please...I had a typo in there....
Mike

hlnurse, 2020-03-04:
Hi Mike,

I was wrong about PS3;, you are correct that PS9; is used for the CPU sleep mode. I will test again in a bit with the new fork.  I’m thinking that the only way PS1; works is if you turn off the radio by PS0; or PS9;, not by means of the power switch.  I’m testing using a radio owned by a club at a location remote to the members, so it may take time to confirm my suspicion.  (I see nothing about this in the manual.)

—Howard

hlnurse, 2020-03-04:
Good news, in the latest fork the power on/off works perfectly with the TS-480.

Thank you!

Howard W6HN

mdblack98, 2020-03-06:
Fixed

closed 2020-03-06 by mdblack98