Hamlib/Hamlib #207: FT-991A Frequency/mode setting

hlnurse, 2020-03-07:
Setting frequency and mode on the FT-991A appears to be broken. Getting frequency and mode are OK.  The debug log, below, shows init, f, F<freq> and w FA<freq>;

--Howard

-----------------------
pi@rigpi4:~ $ rigctl -m 1035 -r /dev/ttyUSB0 -vvvvv
rigctl, Hamlib 4.0~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
yaesu: initrigs4_yaesu called
rig_register called
rig_register: rig_register (1021)
rig_register called
rig_register: rig_register (1027)
rig_register called
rig_register: rig_register (1010)
rig_register called
rig_register: rig_register (1005)
rig_register called
rig_register: rig_register (1006)
rig_register called
rig_register: rig_register (1007)
rig_register called
rig_register: rig_register (1039)
rig_register called
rig_register: rig_register (1009)
rig_register called
rig_register: rig_register (1020)
rig_register called
rig_register: rig_register (1001)
rig_register called
rig_register: rig_register (1022)
rig_register called
rig_register: rig_register (1023)
rig_register called
rig_register: rig_register (1011)
rig_register called
rig_register: rig_register (1015)
rig_register called
rig_register: rig_register (1013)
rig_register called
rig_register: rig_register (1014)
rig_register called
rig_register: rig_register (1028)
rig_register called
rig_register: rig_register (1031)
rig_register called
rig_register: rig_register (1016)
rig_register called
rig_register: rig_register (1003)
rig_register called
rig_register: rig_register (1024)
rig_register called
rig_register: rig_register (1004)
rig_register called
rig_register: rig_register (1025)
rig_register called
rig_register: rig_register (1029)
rig_register called
rig_register: rig_register (1037)
rig_register called
rig_register: rig_register (1032)
rig_register called
rig_register: rig_register (1030)
rig_register called
rig_register: rig_register (1017)
rig_register called
rig_register: rig_register (1019)
rig_register called
rig_register: rig_register (1018)
rig_register called
rig_register: rig_register (1026)
rig_register called
rig_register: rig_register (1033)
rig_register called
rig_register: rig_register (1034)
rig_register called
rig_register: rig_register (1035)
rig_register called
rig_register: rig_register (1036)
rig_register called
rig_register: rig_register (1038)
rig_register called
rig_register: rig_register (1040)
ft991_init called, version 0.28.9
newcat_init called
rig_open called
port_open called
serial_open called
serial_setup called
serial_flush called
serial_flush: tcflush
newcat_open called
newcat_open: write_delay = 0 msec
newcat_open: post_write_delay = 50 msec
rig_set_powerstat called
newcat_set_powerstat called
newcat_valid_command called
newcat_valid_command PS
write_block called
write_block(): TX 1 bytes
0000    0a                                                  .               
write_block called
write_block(): TX 4 bytes
0000    50 53 31 3b                                         PS1;            
newcat_get_trn called
newcat_valid_command called
newcat_valid_command AI
cmd_str = AI;
write_block called
write_block(): TX 3 bytes
0000    41 49 3b                                            AI;             
read_string called
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
newcat_get_cmd: read count = 2, ret_data = ?;
newcat_get_cmd: Rig busy
read_string called
read_string(): RX 2 characters
0000    3f 3b                                               ?;              
newcat_get_cmd: read count = 2, ret_data = ?;
newcat_get_cmd: Rig busy
read_string called
read_string(): RX 4 characters
0000    41 49 30 3b                                         AI0;            
newcat_get_cmd: read count = 4, ret_data = AI0;
newcat_set_trn called
newcat_valid_command called
newcat_valid_command AI
cmd_str = AI0;
serial_flush called
serial_flush: tcflush
cmd_str = AI0;
write_block called
write_block(): TX 4 bytes
0000    41 49 30 3b                                         AI0;            
cmd_str = ID;
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string called
read_string(): RX 7 characters
0000    49 44 30 36 37 30 3b                                ID0670;         
newcat_set_cmd: read count = 7, ret_data = ID0670;
newcat_get_rigid called
newcat_get_info called
newcat_get_info: cmd_str = ID;
cmd_str = ID;
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string called
read_string(): RX 7 characters
0000    49 44 30 36 37 30 3b                                ID0670;         
newcat_get_cmd: read count = 7, ret_data = ID0670;
rig_id = 670, *s = 0670
rig_get_vfo called
rig_set_parm called
Opened rig model 1035, 'FT-991'
Backend version: 0.28.9, Status: Stable

Rig command: f
rigctl_parse: debug1
rigctl_parse: debug5
rigctl_parse: debug10
rig_get_freq called
newcat_get_freq called
newcat_get_freq: passed vfo = currVFO
newcat_valid_command called
newcat_valid_command FA
newcat_set_vfo_from_alias: alias vfo = currVFO
newcat_valid_command called
newcat_valid_command FA
cmd_str = FA;
cmd_str = FA;
write_block called
write_block(): TX 3 bytes
0000    46 41 3b                                            FA;             
read_string called
read_string(): RX 12 characters
0000    46 41 30 30 33 35 35 34 34 30 30 3b                 FA003554400;    
newcat_get_cmd: read count = 12, ret_data = FA003554400;
newcat_get_freq: freq = 3554400.000000 Hz for vfo VFOA
Frequency: 3554400

Rig command: F 3600000
rigctl_parse: debug1
rigctl_parse: debug3
rigctl_parse: debug4
rigctl_parse: debug5
rigctl_parse: debug10
rig_set_freq called
newcat_set_freq called
newcat_valid_command called
newcat_valid_command FA
newcat_set_freq: passed vfo = currVFO
newcat_set_freq: passed freq = 3600000.000000 Hz
newcat_set_freq: R2 minimum freq = 0.000000 Hz
newcat_set_freq: R2 maximum freq = 0.000000 Hz
set_freq: error = Invalid parameter

Rig command: w FA003600000;
rigctl_parse: debug1
rigctl_parse: debug2
rigctl_parse: debug2a
rigctl_parse: debug5
rigctl_parse: debug10
rigctl_send_cmd: called
rigctl_send_cmd: backend_num=1
rigctl_send_cmd: arg1=FA003600000;
rigctl_send_cmd: send_cmd_term==-1, arg1=FA003600000;
rigctl_send_cmd: expecting binary hex string here
send_cmd: error = Invalid parameter

mdblack98, 2020-03-07:
Fixed

closed 2020-03-07 by mdblack98

N0NB, 2020-03-07:
Try again Howard....should be fixed now.
de Mike W9MDB

Hamlib-developer mailing list
https://lists.sourceforge.net/lists/listinfo/hamlib-developer