Hamlib/Hamlib #208: Icom IC-7300 power-on during init

hlnurse, 2020-03-08:
Starting with power off, IC-7300, rigctld successfully powers up the radio but terminates with a "Feature not available" error. It also appears that power on is not being sent (although debug says it is) before the log string fe attempt.

Log below.

--Howard

---------------------------------------------------
pi@rigpi4:~ $ rigctld -m 3073 -r /dev/ttyUSB0 -s 115200 -vvvvv
Recommend using --vfo switch for rigctld
rigctl and netrigctl will automatically detect vfo mode
rigctld, Hamlib 4.0~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
initrigs4_icom: _init called
rig_register called
rig_register: rig_register (3055)
rig_register called
rig_register: rig_register (3009)
rig_register called
rig_register: rig_register (3010)
rig_register called
rig_register: rig_register (3011)
rig_register called
rig_register: rig_register (3013)
rig_register called
rig_register: rig_register (3014)
rig_register called
rig_register: rig_register (3015)
rig_register called
rig_register: rig_register (3019)
rig_register called
rig_register: rig_register (3020)
rig_register called
rig_register: rig_register (3021)
rig_register called
rig_register: rig_register (3022)
rig_register called
rig_register: rig_register (3067)
rig_register called
rig_register: rig_register (3023)
rig_register called
rig_register: rig_register (3046)
rig_register called
rig_register: rig_register (3024)
rig_register called
rig_register: rig_register (3028)
rig_register called
rig_register: rig_register (3030)
rig_register called
rig_register: rig_register (3026)
rig_register called
rig_register: rig_register (3027)
rig_register called
rig_register: rig_register (3047)
rig_register called
rig_register: rig_register (3057)
rig_register called
rig_register: rig_register (3063)
rig_register called
rig_register: rig_register (3029)
rig_register called
rig_register: rig_register (3062)
rig_register called
rig_register: rig_register (3045)
rig_register called
rig_register: rig_register (3056)
rig_register called
rig_register: rig_register (3075)
rig_register called
[... 203 lines cut ...]
read_string(): RX 6 characters
0000    fe fe 94 e0 03 fd                                   ......          
read_string called
read_string(): RX 11 characters
0000    fe fe e0 94 03 00 40 27 14 00 fd                    ......@'...     
from_bcd called
icom_get_usb_echo_off called
icom_get_usb_echo_off: retry temp set to 1
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 6 bytes
0000    fe fe 94 e0 03 fd                                   ......          
read_string called
read_string(): RX 6 characters
0000    fe fe 94 e0 03 fd                                   ......          
read_string called
read_string(): RX 11 characters
0000    fe fe e0 94 03 00 40 27 14 00 fd                    ......@'...     
icom_get_usb_echo_off: USB echo on detected
rig_get_func called
rig_has_get_func called
icom_rig_open: satmode=0
rig_open: error = Feature not available 
pi@rigpi4:

mdblack98, 2020-03-08:
Fixed

closed 2020-03-08 by mdblack98

N0NB, 2020-03-08:
Just fixed it Howard.
de Mike W9MDB

 
 

Hamlib-developer mailing list
https://lists.sourceforge.net/lists/listinfo/hamlib-developer