Hamlib/Hamlib #209: IC-7300 Init broken

hlnurse, 2020-03-10:
I suspect this issue may be related to the ongoing split changes.  The write/read blocks occur more than 20 times.  I've included the last block from the log obtained with:

rigctl -m 1073 -r /dev/ttyUSB0 -s 115200 -vvvvv

--Howard

-------------------
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 87 bytes
0000    fe fe 94 e0 1e 01 00 00 00 00 ff ff ff ff 74 1a     ..............t.
0010    8b b6 ec ea 8f b6 e8 f4 fc b6 d8 14 7b b6 44 ec     ............{.D.
0020    a0 b6 e0 35 f4 01 24 77 96 b6 24 07 a1 b6 44 ec     ...5..$w..$...D.
0030    a0 b6 70 33 f6 b6 e8 f4 fc b6 fc ef 8f b6 20 dc     ..p3.......... .
0040    fc b6 00 00 00 00 01 00 00 00 a0 99 d0 b6 00 00     ................
0050    00 00 54 0a fb b6 fd                                ..T....         
read_string called
read_string(): RX 79 characters
0000    fe fe 94 e0 1e 01 00 00 00 00 ff ff ff ff 74 1a     ..............t.
0010    8b b6 ec ea 8f b6 e8 f4 b6 d8 14 7b b6 44 ec a0     ...........{.D..
0020    b6 e0 35 f4 01 24 77 96 b6 24 07 a1 b6 44 ec a0     ..5..$w..$...D..
0030    b6 70 33 f6 b6 e8 f4 b6 ef 8f b6 20 dc b6 00 00     .p3........ ....
0040    00 00 01 00 00 00 a0 99 d0 b6 00 00 00 00 54        ..............T 
read_string called
read_string(): RX 0 characters
read_string called
read_string(): RX 0 characters
read_string called
read_string(): RX 0 characters
read_string called
read_string(): RX 0 characters
read_string called
read_string(): RX 0 characters
read_string called
read_string(): RX 0 characters
read_string called
read_string(): RX 0 characters
read_string called
read_string(): RX 0 characters
read_string called
read_string(): RX 0 characters
read_string called
read_string(): RX 0 characters
icom_transaction: failed: Unknown error -13
icom_get_freq_range: error from C_CTL_EDGE?  err=Communication bus error
icom_get_freq_range: Hamlib ranges
icom_get_freq_range: rig chan 0, low=30000, high=74800000
rig_get_vfo called
rig_set_parm called
rig_has_set_parm called
Opened rig model 3073, 'IC-7300'
Backend version: 0.36.5, Status: Stable

Rig command: ^C

N0NB, 2020-03-10:
Try again...I'll be testing this more with an IC-7300 user tomorrow.Trying to do some automatic frequency range detection.
de Mike W9MDB

 
 

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

hlnurse, 2020-03-11:
Looks fine, Mike, thanks!

—Howard

mdblack98, 2020-03-11:
Fixed.

closed 2020-03-11 by mdblack98