Hamlib/Hamlib #210: QRPver Minion Mini (TS-440) - CAT Commands accepted only when entered as complete string

DK4FT, 2020-03-21:
I am using a QRPver Minion Mini, which has implemented a CAT interface using the TS-440 command set and having issues with hamlib.

rigctrl receiving at any send cat command a ?; as response from the radio.
Is it possible to send always complete strings/commands to the radio instead of sending character by character?

After testing the cat interface with a terminal programm I think the radio wants to see always complete strings:

**Manual, by keyboard entered characters produced -> Command syntax was incorrect:**
  _FA;?;
  FB;?;
  FA00007154000;?;_

**copied complete string into terminal -> did work fine and produced correct responses:**
  _FA;FA00007200000;
  FB;FB00028060000;
  FA00007200000;IF00007200000     +000000 0002000000 ;_

I assume there is a difference to a real TS-440, which probably could also handle one character after the other send on the cat interface until the End symbol ";" is received by the radio.

I did some testing with Hamlib on my Raspi and during testing with rigctl any response were _?;_
It its the same result compared to sending charcter by character to the rig.

The initiated cat commands where
  - v - read VFO
  - f - read Frequency
  - t - get TX status

> i@HamPi:~ $ rigctl -m 202 -r /dev/ttyACM0 -vvvv
> rigctl, Hamlib 3.3
> Report bugs to <hamlib-developer@lists.sourceforge.net>
> 
> rig_init called
> initrigs3_kenwood called
> rig_register called
> rig_register: rig_register (213)
> rig_register called
> rig_register: rig_register (201)
> rig_register called
> rig_register: rig_register (225)
> rig_register called
> rig_register: rig_register (203)
> rig_register called
> rig_register: rig_register (204)
> rig_register called
> rig_register: rig_register (216)
> rig_register called
> rig_register: rig_register (224)
> rig_register called
> rig_register: rig_register (205)
> rig_register called
> rig_register: rig_register (207)
> rig_register called
> rig_register: rig_register (209)
> rig_register called
> rig_register: rig_register (210)
> rig_register called
> rig_register: rig_register (222)
> rig_register called
> rig_register: rig_register (214)
> rig_register called
> rig_register: rig_register (230)
> rig_register called
> rig_register: rig_register (221)
> rig_register called
> rig_register: rig_register (229)
> rig_register called
> rig_register: rig_register (238)
> rig_register called
> rig_register: rig_register (202)
> rig_register called
> rig_register: rig_register (211)
> rig_register called
> rig_register: rig_register (206)
> rig_register called
> rig_register: rig_register (208)
> rig_register called
> rig_register: rig_register (215)
> rig_register called
> rig_register: rig_register (226)
> rig_register called
> rig_register: rig_register (217)
> rig_register called
> rig_register: rig_register (233)
> rig_register called
> rig_register: rig_register (220)
> rig_register called
> rig_register: rig_register (223)
> rig_register called
> rig_register: rig_register (227)
> rig_register called
> rig_register: rig_register (234)
> rig_register called
> rig_register: rig_register (231)
> rig_register called
> rig_register: rig_register (239)
> rig_register called
> rig_register: rig_register (237)
> rig_register called
> rig_register: rig_register (228)
> rig_register called
> rig_register: rig_register (219)
> rig_register called
> rig_register: rig_register (232)
> rig_register called
> rig_register: rig_register (236)
> rig_register called
> rig_register: rig_register (240)
> kenwood_init called
> rig_open called
> port_open called
> serial_open called
> serial_setup called
> rig_get_vfo called
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> Opened rig model 202, 'TS-440'
> rig_strstatus called
> Backend version: 1.0.0.6.1, Status: Alpha
> 
> Rig command: v
> rig_get_vfo called
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> get_vfo: error = Command rejected by the rig
> 
> Rig command: f
> rig_get_freq called
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> get_freq: error = Command rejected by the rig
> 
> Rig command: t
> rig_get_ptt called
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> serial_flush called
> write_block called
> read_string called
> get_ic10_if: unexpected answer ?;, len=2
> get_ptt: error = Command rejected by the rig

mdblack98, 2020-03-21:
Also it may be an easy fix.  Sounds like the Minion is timing out.
The default inter-char delay for the TS440 is 20ms.
write_delay: "Delay in ms between each byte sent out" Default: 0, Value: 20 Range: 0.0..1000.0, step 1.0

So try changing it to 0
 rigctl -m 202 -r /dev/ttyACM0 -vvvvv --set-conf=write_delay=0

de Mike W9MDB

DK4FT, 2020-03-21:
mni tnx mike - that did the magic.
The minion is reacting as expected now, many thanks for the speedy answer and solution.

73 de
Alex, DK4FT

closed 2020-03-21 by DK4FT

N0NB, 2020-03-21:
You may want to contact the Minion people and let them know they need to fix that bug.
It should be waiting for the semicolon to terminate a command regardless how long it takes....you should be able to type in a command like you were doing and have it work.
de Mike W9MDB

 
 

Hamlib-developer mailing list
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

DK4FT, 2020-03-22:
My expectation was the rig has to wait for the semicolon as well, but was not sure.
I will forward this bug/wrong implementation to the QRPver people and thanks for the confirmation.

Now I am troubleshooting WSJT-x <-> rigctld connection, but that is another story.
(WSJT starting up fine, but showing an error at any change Frequency, PTT, ...)
For the moment I try to find known issues, as I could not be the only one with such problems...

73 de
Alex, DK4FT

N0NB, 2020-03-22:
Send me a debug log and we'll see what's going on.  Remember using 5 v's and the -Z switch for timing too.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

DK4FT, 2020-03-22:
thanks for your offer..

Here is what I did and recorded in the attached file:

1. started rigctld 
> rigctld -m 202 -r /dev/ttyACM0 -s 57600 -p /dev/ttyACM0 -P RIG --set-conf=write_delay=0 -vvvvv -Z
2. started WSJT-x
3. changed frewuency in wsjt-x to 7074 kHz

[wsjtx-rictld_error.txt](https://github.com/Hamlib/Hamlib/files/4365345/wsjtx-rictld_error.txt)

It seems the received string is not correctly terminated by the radio, here two examples:

> 2020-03-22:12:39:05.070293: 0000    49 44 3b                                            ID;             
> 2020-03-22:12:39:05.070309: read_string called
> 2020-03-22:12:39:05.070447: read_string(): RX 15 characters
> 2020-03-22:12:39:05.070466: 0000    49 46 30 30 30 30 37 30 37 34 30 30 30 20 20        IF00007074000   
> 2020-03-22:12:39:05.070480: kenwood_transaction: Command is not correctly terminated 'IF00007074000  '
> 2020-03-22:12:39:05.070495: kenwood_transaction: cmdstr = FA00007074000
> 

> 2020-03-22:12:39:05.708331: write_block(): TX 3 bytes
> 2020-03-22:12:39:05.708400: 0000    52 58 3b                                            RX;             
> 2020-03-22:12:39:05.708420: write_block called
> 2020-03-22:12:39:05.738536: write_block(): TX 3 bytes
> 2020-03-22:12:39:05.738586: 0000    49 44 3b                                            ID;             
> 2020-03-22:12:39:05.738606: read_string called
> 2020-03-22:12:39:05.738663: read_string(): RX 3 characters
> 2020-03-22:12:39:05.738681: 0000    52 58 3b                                            RX;             
> 2020-03-22:12:39:05.738697: kenwood_transaction: wrong reply RX for command verification ID
> 2020-03-22:12:39:05.738715: kenwood_transaction: cmdstr = RX

N0NB, 2020-03-22:
I see that the manufacturer says to use the TS-450...not the TS-440.  They also say FLRig works.The TS-450 also defaults to write_delay=0Can you try the TS-450 to see if it behaves better?
Can you compile hamlib?  Would like to know how this behaves with the latest version.
I can make some changes to allow a poor emulation to work....but they don't even respond to the commands correctly at all.If it doesn't support a command it should return "?;" and not some random answer.
For examples, the response to "ID;" is a number...not an IF string.And there is no response to the "RX" command at all.  All set commands have no response.   Only queries have a response.
Their emulation needs some improvement.
de Mike W9MDB

 
 

   

   
   - started WSJT-x

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

DK4FT, 2020-03-22:
They are writing they implemented an TS-440, but in FLRig TS-450 should be used, as TS-440 is not available. (flrig is another story - it is freezing from time to time (windows and linux) , but is able to handle the responses of this radio. Omni Rig is even more fault tolerant and working fine under windows).

If I choose TS-450S (type 203) in rigctld it returns directly an error.

>  rigctld -m 203 -r /dev/ttyACM0 -s 57600 -P RIG --set-conf=write_delay=0 -vvvvv -Z

**Result:**

> 2020-03-22:14:40:07.935151: write_block called
> 2020-03-22:14:40:07.946798: write_block(): TX 3 bytes
> 2020-03-22:14:40:07.946984: 0000    49 44 3b                                            ID;             
> 2020-03-22:14:40:07.947070: read_string called
> 2020-03-22:14:40:07.947944: read_string(): RX 6 characters
> 2020-03-22:14:40:07.948147: 0000    49 44 30 30 34 3b                                   ID004;          
> 2020-03-22:14:40:07.948237: kenwood_open: found match 004
> 2020-03-22:14:40:07.948320: kenwood_open: wrong driver selected (203 instead of 202)
> rig_open: error = Invalid parameter

mdblack98, 2020-03-22:
That's interesting it gives the correct response to the ID command this time.
Mike

DK4FT, 2020-03-22:
Here the list of all supported commands, which is only a subset of the whole TS-440 comand set.
(part of the omnrig.ini for this radio)

Some example responses (entered in a terminal):
**BOLD = command** _italic = response_
ID; giving me always a ID004; back.

  **RX;**_RX;_
  **TX;**_TX;_
  **IF;**_IF00007154000     +000000 0006000000 ;_
  **ID;**_ID004;_
  **MD2;**
  **FA;**_FA00007200000;_
  **FB;**_FB00028060000;_
  **FA00007200000;**_IF00007200000     +000000 0002000000 ;_

Probably wsjt-x is calling commands, which are not available by the radio.
[MINION-SDR_supported_commands.txt](https://github.com/Hamlib/Hamlib/files/4365471/MINION-SDR_supported_commands.txt)

N0NB, 2020-03-22:
It would be nice if they targeted something a bit more recent than a TS-440.
I can't even find a CAT protocol document for it.  At least I have the TS-450 document.

I ran into something similar with the TS-480.

When TX is on RX does return RX0.When not in transmit RX does NOT return anything

I'll bet the TS-440 behaves the same way which means responding to the RX command when not in transmit mode is incorrect.
If you could please send me an entire debug log running with WSJT-X and rigctld I can see better what's going on and if it's fixable.
And you didn't answer if you can compile hamlib?
de Mike W9MDB 

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-03-22:
* On 2020 22 Mar 12:01 -0500, Black Michael wrote:
> 
> It would be nice if they targeted something a bit more recent than a TS-440.
> I can't even find a CAT protocol document for it.  At least I have the TS-450 document.

Mike, look for the IC-10 manual.  I have a PDF copy here I can send
along if that is easier.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

mdblack98, 2020-03-22:
I found a copy.
I see the 440 is "alpha" status and it looks like the PTT function is using kenwood instead of IC10 like it would appear that it should.
I'll need to find a TS-440 owner though to test it out on a real TS-440.
Mike

 
 

Mike, look for the IC-10 manual. I have a PDF copy here I can send
along if that is easier.

73, Nate

N0NB, 2020-03-22:
* On 2020 22 Mar 12:40 -0500, Michael Black wrote:
> I found a copy.
> I see the 440 is "alpha" status and it looks like the PTT function is
> using kenwood instead of IC10 like it would appear that it should.
> I'll need to find a TS-440 owner though to test it out on a real TS-440.

Fairly easy to find a TS-440 owner and a bit more difficult to find one
with the IC-10 interface.  A lot of those early Kenwood models required
such an interface until the market turned toward integrating the RS-232
chips in the radio.  I built my own for a TS-850 I had for a few years
in the late '90s.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

DK4FT, 2020-03-22:
Hi Nate,

before I foget to answer, I am able to build from source, will do tomorrow.
(the Raspian is build by using an installation scrip from Jason, KM4ACK - build from source)

Attached the complete logs traced the following:
1. start rigctld
2. start wsjtx - wait until standard polling cycle started
3. frequency change
4. wsjtx stopped with a Rig Controll Error.

[wsjtx-rictld_error_complete.txt](https://github.com/Hamlib/Hamlib/files/4365962/wsjtx-rictld_error_complete.txt)

mdblack98, 2020-03-22:
When you compile use my fork.
https://github.com/mdblack98/Hamlib

I just checked in a change that probably fixes your problem.
de Mike W9MDB

 
 

   
   - start rigctld
   - frequency change

DK4FT, 2020-03-23:
About your workaround - I tried to use the fork above, but did get an issue during make:

```
/bin/ld: ../src/.libs/libhamlib.so: undefined reference to `ic10_set_ptt'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:952: rigctl] Fehler 1
make[1]: Verzeichnis »/home/pi/Downloads/Hamlib/tests« wird verlassen
make: *** [Makefile:605: all-recursive] Fehler 1
```

Did you also adress the other odd behaviour of the minion?

Collected issues during tests (**bold = command**/_italic = response_):

  **RX;**_RX;_ -> NOK no response expected
  **TX;**_TX;_ -> NOK no response expected
  **FA00007210000;**_IF00007210000 +000000 0006001000 ;_ NOK, expected is: FA00007210000;
  **SP1;**_IF00007074055 +000000 0006001000 ;_ -> NOK no response expected
  **SP2;**_IF00007074055 +000000 0006002000 ;_ -> NOK no response expected

73 de Alex, DK4FT

N0NB, 2020-03-23:
I fixed it.
I think the change I put in should fix all these problems...it basically just ignores the response.
Mike

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

DK4FT, 2020-03-23:
ok, was able to build and install - WSJT-x still crashes at frequency change and PTT. I attached the new logs...

[WSJT-rigctld_error_200323.txt](https://github.com/Hamlib/Hamlib/files/4370155/WSJT-rigctld_error_200323.txt)

mdblack98, 2020-03-23:
Try my fork again....
Mike

DK4FT, 2020-03-23:
You are my hero - main issues are worked around.

Startup of WSJT-x - ok
Frequency change - ok
PTT - ok

Just if I try to use Split from radio rigctld crashes after calling this command: SP1; or SP2;
Log is attached.
[WSJT-rigctld_error_200323_SP1_crash.txt](https://github.com/Hamlib/Hamlib/files/4370332/WSJT-rigctld_error_200323_SP1_crash.txt)

w4jkl, 2020-03-23:
Mike,

Did you find anyone with a TS-440 to help you test?

I have one that I keep as a backup to the 7300.

73,

John W4JKL

N0NB, 2020-03-23:
Try again...you would get a segfault award is this was a release version....:-)Though one could argue this bug has been around for a long time.

de Mike W9MDB

 

   ._______________________________________________
Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer

DK4FT, 2020-03-23:
hihi - but we are getting closer...

Startup of WSJT-x - ok
Frequency change - ok
PTT - ok
Split activated in WSJT-x - ok
PTT at split -> WSJT-x crash

[WSJT-rigctld_error_200323_TX_Split.txt](https://github.com/Hamlib/Hamlib/files/4370752/WSJT-rigctld_error_200323_TX_Split.txt)

mdblack98, 2020-03-23:
Try again...I thought WJST-X knew how to handle rigs with only VFOA and VFOB but it seems to be asking for Main/Sub according to your debug log.I wonder if that's a bug with WSJT-X/rigctld somehow?
de Mike W9MDB

DK4FT, 2020-03-23:
I have no idea if it is a WSJT bug or something different, but
Split is somehow special - getting still the same crash with PTT at split -> WSJT-x crash

` Hamlib error: Invalid parameter while setting split TX frequency and mode`

[WSJT-rigctld_error_200323_TX_Split_2.txt](https://github.com/Hamlib/Hamlib/files/4371111/WSJT-rigctld_error_200323_TX_Split_2.txt)

N0NB, 2020-03-23:
If „split“ is the problem, choose „fake it“ split mode in WSJTX,
then wsjtx does it the pedestrian way.

> 
> 
> I have no idea if it is a WSJT bug or something different, but
> Split is somehow special - getting still the same crash with PTT at split -> WSJT-x crash
> 
> Hamlib error: Invalid parameter while setting split TX frequency and mode
> 
> WSJT-rigctld_error_200323_TX_Split_2.txt <https://github.com/Hamlib/Hamlib/files/4371111/WSJT-rigctld_error_200323_TX_Split_2.txt>
> 
> Hamlib-developer mailing list
> Hamlib-developer@lists.sourceforge.net
> https://lists.sourceforge.net/lists/listinfo/hamlib-developer

+---------------------------------+----------------------------------+
| Prof. Christoph van Wüllen      | Tele-Phone (+49) (0)631 205 2749 |
| TU Kaiserslautern, FB Chemie    | Tele-Fax   (+49) (0)631 205 2750 |
| Erwin-Schrödinger-Str.          |                                  |
| D-67663 Kaiserslautern, Germany | vanWullen@Chemie.Uni-KL.De <mailto:vanWullen@chemie.uni-kl.de>       |
|                                                                    |
| HomePage:  http://www.chemie.uni-kl.de/vanwullen <http://www.chemie.uni-kl.de/vanwullen>                   |
+---------------------------------+----------------------------------+

DK4FT, 2020-03-23:
Fake It = Fine for me.

How is the future of this "hack"?
Will these changes are ...
- merged back into the master?
- added as new device for Minion Mini?
- left in the fork?

N0NB, 2020-03-23:
You picked Data/Pkt in WSJT-X and the Minion doesn't have packet mode.
Just stick with USB.
de Mike W9MDB

 
 

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

N0NB, 2020-03-23:
It's will be pushed to the master when it is working completely....And it's not a hack....'twas a bug...
de Mike W9MDB

 
 

   

https://lists.sourceforge.net/lists/listinfo/hamlib-developer

DK4FT, 2020-03-23:
Same issue with USB and the Minion has a DATA mode:  MD6;

mdblack98, 2020-03-23:
The Minion may have MD6 but the TS-440 does not.  And unless there's some specific reason we have to usg MD6 on the Minion I'm not inclined to generate a new model for it.
If USB isn't working please send a debug log for that mode too.
And you are using my fork?
https://github.com/mdblack98/Hamlib

de Mike W9MDB

DK4FT, 2020-03-23:
Yes, I am using your fork and it is up to date.

The only difference from USB to Digi is the max Bandwidth in TX,
but the difference from 32, to 3,5 kHz does not justify another model in hamlib.
But even in USB WSJT crashes...

[WSJT-rigctld_error_200323_TX_Split_USB.txt](https://github.com/Hamlib/Hamlib/files/4371662/WSJT-rigctld_error_200323_TX_Split_USB.txt)

```

Transmitter radiation band:

- LSB/USB Mode: 3200 Hz Max. (It is set in the menu. Settings of the lower and upper cutoffs are 50-300, 2500-3200 Hz.).

- РDIGI Mode: 3500Hz.

```

w4jkl, 2020-03-23:
Split mode on the TS-140S (it has a Z-80 CPU) is too slow to satisfy 
WSJT-X, and WSJT-X crashes. I have to use Fake It. I do the same with 
the TS-440S (it has an 8080 CPU, so may do better at the split). What 
baud rate does your QRP rig run? Kenwoods only ran 4800 baud back then 
and performed accordingly. Hamlib should expect nothing more.

John W4JKL

N0NB, 2020-03-23:
His problem is a bug in the ic10 code which I just fixed.
de Mike W9MDB

 
 

—

Hamlib-developer mailing list
Hamlib-developer@lists.sourceforge.net
https://lists.sourceforge.net/lists/listinfo/hamlib-developer